omerfaruk-aran/esphome_samsung_hvac_bus #1: Help with Non-NASA Ducted system

buzurk26, 2023-06-18:
Hi, 

Great project! very keen to get this working with my AC

I have a non-Nasa ducted system that previously had the wireless controller on the f1/f2 terminals, but it has stopped working of late and Samsung don't sell replacements

I got some esp32s and the rs485 ttls, I'm a bit of a noob but wondering how i could get started with your code on my non-nasa setup?

Only functions ill need are

- AC ON/OFF
- MODE (Auto, Heat, Cool, Fan, Dry)
- Fan Speed
- Target Temperature
- Current Temperature

buzurk26, 2023-06-19:
Hi @lanwin I would be happy to do the testing in hope to get a working solution

buzurk26, 2023-06-24:
Hi @lanwin I have a ESP32 and TTL adapter all setup with ESPhome on it ready to go, let me know what i need to do to help test

thanks a bunch

lanwin, 2023-07-03:
Give me some time. Currently I am a little bit overloaded. 

In the mean time you could solder everything together. If everything works, the LED on the RS485 board should blink every few seconds.

Then you can bring up ESPHome and try to setup the ESP with my example script https://github.com/lanwin/esphome_samsung_ac/blob/main/example.yaml . Everything below device should be removed for now and the baud_rate should be changed to 2400.

If everything should work, you sould be able to access the webserver on port 80.

buzurk26, 2023-07-04:
Hi @lanwin 

Thank you so much, greatly appreciate your help and apologies for bugging you

Working in the background to control the zone motors and got that sorted so just pending the AC control, will spin up an esp with your code and capture the data?

buzurk26, 2023-07-07:
Hi, Its all connected now with your code, I left out everything under "devices" as you suggested

In the esp logs i see this over and over
[image]

buzurk26, 2023-07-08:
Switched to an ESP8266, still not recieving data on the ESP

I have this UART converter
[image]

Wired like this
[image]

and here is my code https://pastebin.com/ujyNVnqt

lanwin, 2023-07-09:
> Switched to an ESP8266, still not recieving data on the ESP [image]
> 
> I have this UART converter [image]
> 
> Wired like this [image]
> 
> and here is my code https://pastebin.com/ujyNVnqt

> Switched to an ESP8266, still not recieving data on the ESP [image]
> 
> I have this UART converter [image]
> 
> Wired like this [image]
> 
> and here is my code https://pastebin.com/ujyNVnqt

Thats correct so far and the output is what I would expect. As I said. The actual data handling of the old protocol is not yet implemented. This is why there is not output.

lanwin, 2023-07-09:
Please install again. I added some basic logging witch should log the received data messages. You could attach some samples I can use.

buzurk26, 2023-07-09:
@lanwin  - Ah makes sense, i hadnt realised.

I just re-upload the code to the esp i assume and it will pull your changes??

Also, does your code support esp32? i plan to use esp32 in future

lanwin, 2023-07-09:
Reinstalling should pull the changes. 

ESPHome supports ESP32 so the code should work there too.

buzurk26, 2023-07-09:
Thank you, Just reinstalled now. Not getting any output unfortunately

[image]

lanwin, 2023-07-09:
Can you add this to your config under UART

```yaml
  debug:
    direction: BOTH
    dummy_receiver: false
      after:
        delimiter: "\n"
        sequence:
        - lambda: UARTDebug::log_hex(direction, bytes, ' ');
```

This sould print everything whats received via UART.

And did your RS485 convert is blinking?

buzurk26, 2023-07-09:
Sure, will add that now and try

The RX led on the converter is blinking rapidly yes, nothing on the tx led

buzurk26, 2023-07-09:
Sorry just having some issues adding that bit to esphome
[image]

lanwin, 2023-07-09:
You are at the wrong place. Under UART not under logger.

buzurk26, 2023-07-09:
Yeah sorry my bad, tired eyes

uploading now

buzurk26, 2023-07-09:
Heaps out output now

[image]

buzurk26, 2023-07-09:
capture here https://pastebin.com/RfvmDbXr

lanwin, 2023-07-09:
Ok it seems that its receiving data. Please remove that lines again and pull and install again. I added another logging code.

buzurk26, 2023-07-09:
Awesome, Will do that now.

buzurk26, 2023-07-09:
Back to no data unfortunately
[image]

lanwin, 2023-07-09:
Ok just put back the debug code from above. That data did not seems to match from what is specified in the protocol. 

There should at least be some \x32 and \x34 in there.

Can you try to swap F1 and F2?

buzurk26, 2023-07-09:
Oki, will swap f1/f2 and add that debug code back in

buzurk26, 2023-07-09:
Here is the log now, https://pastebin.com/1jiz9uPE

lanwin, 2023-07-09:
Hmm still no \x32 and \x34. Can you swap back F1/F2 and then try the following baud rates? Please check then of you find multiple \x32 and \x34 in log.

1200, 2400, 4800, 9600, 19200, 38400

buzurk26, 2023-07-09:
Sure can try those tomorrow and let you know.

I do remember when trying a different baud rate the other day I got error
like below

buzurk26, 2023-07-09:
[image]

lanwin, 2023-07-09:
You can ignore this.

buzurk26, 2023-07-09:
Awesome, will do some testing tomorrow and let you know.

buzurk26, 2023-07-10:
Got to 9600 baud rate and things changed
[image]

RAW Code for 9600 baud -https://pastebin.com/Jb7xbR34

lanwin, 2023-07-10:
Interesting. I thought the older systems use 2400 baud. 

So anyway. We have now data packets I can work with (the yellow parts).

You can now delete the URT debug stuff.

buzurk26, 2023-07-10:
Oh that is awesome to hear :) Glad were on the right track 

I'll take that out now and reboot, Anything else you need let me know

lanwin, 2023-07-10:
Can you please reinstall and run it some time and see of you get some log messages.

buzurk26, 2023-07-11:
Yep will do, did you want me to capture the logs?

I can certainly see a difference, the yellow logs are coming in at a much slower rate now and have extra details in front of them
[image]

buzurk26, 2023-07-11:
Here is the log, running for about 1.5 - 2 hrs
[logs_esp32-hvac-test_run (2).txt](https://github.com/lanwin/esphome_samsung_ac/files/12010442/logs_esp32-hvac-test_run.2.txt)

buzurk26, 2023-07-11:
Awesome :) will do that now

buzurk26, 2023-07-11:
[image]

buzurk26, 2023-07-11:
Sorry not quite sure w hat format to add that known devices into your example code

`samsung_ac:
  devices:
  - address: "20.00.00"
    room_temperature:
      name: Test1_temp
    target_temperature:
      name: Test1_target
    power:
      name: Test1_power`

lanwin, 2023-07-11:
This should work.
```
samsung_ac:
  devices:
   - address: "00"
    room_temperature:
      name: Test1_temp
    target_temperature:
      name: Test1_target
    power:
      name: Test1_power
```

buzurk26, 2023-07-11:
OMG!!!!!!! i think were getting there - AC is on now and target is set to 20 deg on the panel

on/off not sure thats working, also is there option for fan speed and mode? (sorry if im jumping too far ahead, dont mean to just excited,)

Thank you

buzurk26, 2023-07-11:
I really, really appreciate your help on this, truely, let me know if you need anything further from me

buzurk26, 2023-07-20:
Hi @lanwin 
Just wanted to touch base to see if there was anything else i can do to assist on this one?

Do you know if (now that we have that information) that it likely we can get the rest? Just trying to plan ahead and what to buy but before i go down that path hoping to get this working

lanwin, 2023-07-20:
I am sorry. Currently I am ahead of a week off. So don't expect anything the next two weeks.

buzurk26, 2023-08-07:
Hi, Hope you had a good break

Wondering if you had time to revisit this?>

thank you

lanwin, 2023-08-21:
Hi, I am back. Maybe I get some time the next days to work on that.

buzurk26, 2023-08-21:
That would be absolutely awesome! Thank you :) 

Can i donate to you or anything by way of saying thank you?

Foxhill67, 2023-08-31:
Highly interested in this project, thanks for the work done so far! I have the AE090RXEDGG/EU + AE260RNWSGG/EU heat pump combination from Samsung. According to the manual, the optional wifi module kit MIM H04EN can be installed. This kit is NASA compliant. But the kit is expensive and only accessible via Samsung Smartthings app. I am very much interested in this ESPHOME project to directly connect my heat pump to Home Assistant. Do you think it will be possible to at least read all interesting information (temperatures, power usage, error codes, etc)?

lanwin, 2023-08-31:
> Good evening, I have this heat pump from Samsung. I don't know which category it falls into, NASA or NO-NASA. Or if it is likely that this integrity could work when it is finished. Thank you Zoul AE160JNYDGH

It seems Samsung uses only this two protocols for everything. So when its a new devices I would bet it uses NASA internally.

buzurk26, 2023-09-01:
Hi Mate, Hope your well just wanted to see if you had time to work on this any further? thanks

lanwin, 2023-09-05:
Ah sorry. What I need for each value is this messages from the log:

[11:48:41][W][command:042]: 00 C8 40 3200c84020000000408900472634

buzurk26, 2023-09-05:
No worries can get that, just wanted to let you know what was already working in between your code and mattias code base, im sure you can integrate the working parts in to save rework

So you want me to perform each function on the panel and capture the code that is logged after that???

lanwin, 2023-09-05:
Not on the panel. I mean directly control your AC via remote

buzurk26, 2023-09-05:
I dont have a remote sorry, Only have the touch panel that does all the control, this is what i have
[image]

ITs a ducted system not a wall mount one with remote

buzurk26, 2023-09-05:
This of the panel as my remote in the context of this system :)

lanwin, 2023-09-05:
> Think of the panel as my remote in the context of this system :)

Right :D

buzurk26, 2023-09-05:
So if i perform an action on the panel (remote) then the esp should show a log of code within 5 seconds of that action and you want me to note that?

lanwin, 2023-09-05:
Right and I need the exact values there where present wenn these messages appear so I can compare the values I get out wenn decoding the message.

buzurk26, 2023-09-06:
Still working through this sorry its really hard to lay it out as there are so many combinations

Can i exclude what is already working in mattias code as posted in my last posts (with the screenshots above) 

Do you have a good way to map this out so its readable?

lanwin, 2023-09-06:
> Still working through this sorry its really hard to lay it out as there are so many combinations

I dont need all combinations. A few examples with different values are ok.

> Can i exclude what is already working in mattias code as posted in my last posts (with the screenshots above)

No. He uses a different programming language. So I still need test data to verify it works.

buzurk26, 2023-09-07:
Hopefully this is what you need, thanks again :)

[AC-Output.pdf](https://github.com/lanwin/esphome_samsung_ac/files/12547665/AC-Output.pdf)

[159 more replies]

closed 2024-02-07 by lanwin