omerfaruk-aran/esphome_samsung_hvac_bus #2: Help to show humidity value from NASA DVM

kolsaito, 2023-07-20:
Hi, 
Thanks for your time.
I've tried as you guide, and it worked.
I can get below functions:
- On/off
- Monitor temperature
- Set target
- Select mode
And as the log shows that I can get humidity value, I tried to add humidity sensor but not success.
Could you give me some guideline to show both humidity and temperature in ESPhome.
Thanks!
@lanwin 
[image]

lanwin, 2023-07-20:
@kolsaito currently humidity is not implemented on the ESPHome side. I will add it. But give me some time. I am right before a week off.

kolsaito, 2023-07-21:
@lanwin very excited to hear that.

bzumik1, 2023-08-24:
Some progress here?

lanwin, 2023-08-25:
Not yet. I was off for vacation and now I need to work off some things before I can start working on it again.

lanwin, 2023-10-05:
Humidity is now available. You find an example in the example.yaml.

closed 2023-10-05 by lanwin

bzumik1, 2023-10-05:
Hi @lanwin, as this is working, we should probably update the [current features](https://github.com/lanwin/esphome_samsung_ac/issues/2#current-features) section in README.md file

bzumik1, 2023-10-05:
BTW isn't there a way how to put this humidity under the climate component, I think it should be possible. But this is lower priority of course. It would just make it cleaner in HA because now it mixes all the data from indoor units under sensor section.

[image]

lanwin, 2023-10-05:
> Hi @lanwin, as this is working (I have tested it), we should probably update the [current features](https://github.com/lanwin/esphome_samsung_ac/issues/2#current-features) section in README.md file
> [image]

Thanks for the hint. I updated the readme.

lanwin, 2023-10-05:
> BTW isn't there a way how to put this humidity under the climate component, I think it should be possible. But this is lower priority of course. It would just make it cleaner in HA because now it mixes all the data from indoor units under sensor section.
> [image]

Why do you think this should be possible? I can not see a way to set the humidity within the climate component api.

bzumik1, 2023-10-05:
Based on brief check of the documentation for ESPHome it looked to me that you can create custom climate component which can basically contain what ever you want, it just needs to implement climate interface, but I might be wrong.

What I would like to achieve is to separate data from each indoor unit. My first idea was that it would be the best if we could create device per each indoor unit but based on what I read this does not seem to be possible with ESPHome. Next option was to put everything under custom climate component.

lanwin, 2023-10-05:
Hmm ok I know what you mean. The Climate part is like a profile https://esphome.io/components/climate/index.html . From what I can see is that it did not support humidity or anything custom. But maybe there is another way to group the components in ESPHome.

bzumik1, 2023-10-05:
I have also opened [this](https://github.com/esphome/feature-requests/issues/2420) issue on ESPHome github. In my eyes it would be best if they would allow to create multiple devices for one ESP board. Let's see what they will respond. I would say this would be useful for other projects as well. Anyway as I said this has low priority so we can solve it later. I will probably open separate issue for that just we can track it.