omerfaruk-aran/esphome_samsung_hvac_bus #3: WindFree mode

bzumik1, 2023-08-24:
Would it be possible to add WindFree mode?

lanwin, 2023-08-25:
I think that would be possible. But the Term WindFree did not exists on protocol level. Maybe it hides between the other control option are its called differently there. 

So basically I did not figured it out yet und I am also not sure whats the best way to integrate it in ESPHome.

bzumik1, 2023-08-25:
I have just ordered the RS485 adapter so when it arrives I can help you to find out the name for this option. I guess what when I turn the WindFree option using wi-fi kit on the same bus we should be able to figure out what command it sent.

lucassant95, 2023-09-26:
This project is AWESOME! I'm buying the adapter and step down modules to start using this ASAP! I have 3 units of Samsung wind free ACs and was wondering how to make them connected to my smart home. I'll try this repo that looks like the best alternative than using IR remotes. Do you have a telegram group or something?

bzumik1, 2023-09-26:
There seem to be nice HW from MStack if you like nice packaging. https://github.com/lanwin/esphome_samsung_ac/issues/4

lanwin, 2023-10-02:
Its pretty funny. Windfree seems to be called alternative_mode on the protocol stack side. Can not spend much time currently in it. But it shouldnt be hard.

bzumik1, 2023-10-02:
That sounds great! Please keep us posted on progress and let us now if any help from our side is needed.

bzumik1, 2023-10-03:
Hi @lanwin, I got the HW and it is now working for me. Please let me know how can I help you to get the **WindFree** mode to the code base. I am also happy to help with testing etc.

bzumik1, 2023-10-03:
Hi @lanwin, I have tried to record some logs when I turn on the WindFree mode using IR remote controller. I did so at around **11:25:00 (+-10s)** on indoor unit with address **20.00.00**. Hopefully it will help

[esp-samsung-ac-windfree.log](https://github.com/lanwin/esphome_samsung_ac/files/12790897/esp-samsung-ac-windfree.log)

P.S. for someone else who would like to record logs here is command I have used `esphome run esp-samsung-ac.yaml --device OTA > esp-samsung-ac-windfree.log`

bzumik1, 2023-10-23:
Hi @lanwin sorry to bother you, is there any update for this feature? In case you would need my help please just let me know.

bzumik1, 2023-10-24:
Ok no problem 🙂

mantiscr, 2023-12-02:
Hi,
I developed an alternative solution on the F3, F4 side, just saw your repo and wanted to share this.
Extra Functions (40 60 _xx_)
_00_ All Off
_03_ Turbo (Refer to user manual for usage)
_06_ Long Reach (Refer to user manual for usage)
_09_ Windfree (Refer to user manual for usage)

(PS. i m just looking for turning off the indoor unit sound, as far as i see it can be changed only from wireless controller.)

PR #80 (2024-02-07): WindFree and other presets
Added support for setting different presets via `NASA_ALTERNATIVE_MODE`.

On my AR09TXFCAWKNEU (3 indoor units), I have three, Quiet (2), Fast (3) and WindFree (9). In #3 another one, Long Reach (6) was mentioned, so there could be more in the wild.
Additional config to enable/disable different modes might be required in the future.

I went with the presets/custom_presets approach, but on the other hand, all three that I tested with are mutually exclusive with fan modes, so they could also be considered as custom_fan_modes instead.

Fixes #3 
Fixes #41
closed 2024-02-07 by lanwin