omerfaruk-aran/esphome_samsung_hvac_bus #4: HW compatibility

bzumik1, 2023-08-24:
Hi do you have idea if this would be compatible:

- [M5STACK ATOM Lite](https://a.aliexpress.com/_mO88aeK)
- [M5STACK RS485](https://a.aliexpress.com/_mLhOZQA)

The compute module is based on esp32 S3

lanwin, 2023-08-25:
I see no reason why this shouldn't work. You should validate if there any experiences with ESPHome and Atom lite in the web. Technically any ESP32 should work.

bzumik1, 2023-08-25:
I have check it and ATOM Lite should be supported, it is one of the devices on the main page of bluetooth proxy. I have ordered one, when it arrives I will confirm the functionality. It is quite expensive, but I like the package overall and if that should be the final solution for Samsung AC it is 10x less expensive than the official WiFi kit.

bzumik1, 2023-10-01:
Hi @lanwin tomorrow I should receive the HW mentioned above. Could you please double check that the I am connecting it correctly. I unfortunately do not have multimeter to test the voltages but based on your description in README.md file the V1 and V2 should be 12V voltage for power and F1 and F2 are communication lines so if I am not mistaken the connection should be like this:

[image]

If you can confirm this to me it would really help, I would like to start with the testing as soon as possible.

lanwin, 2023-10-02:
> Hi @lanwin tomorrow I should receive the HW mentioned above. Could you please double check that the I am connecting it correctly. I unfortunately do not have multimeter to test the voltages but [...]

I am not an hardware guy so I can give you no no guarantees. But from what I know this should be correct. Its cool that it has an 12v in. Mine could handle only 3-5v.

bzumik1, 2023-10-02:
> > Hi @lanwin tomorrow I should receive the HW mentioned above. Could you please double check that the I am connecting it correctly. I unfortunately do not have multimeter to test the voltages but based on your description in README.md file the **V1** and **V2** should be 12V voltage for power and **F1** and **F2** are communication lines so if I am not mistaken the connection should be like this:
> 
> > 
> 
> > [image]
> 
> > 
> 
> > If you can confirm this to me it would really help, I would like to start with the testing as soon as possible.
> 
> 
> 
> I am not an hardware guy so I can give you no guarantees. But from what I know this should be correct. Its cool that it has an 12v in. Mine could handle only 3-5v.

Thank you for checking. Yes I think this could be perfect solution to replace the expensive, cloud base SAMSUNG WiFi kit. Let's hope it will work as expected.

bzumik1, 2023-10-03:
I have got it working, I am able to control both AC units it seems that every implemented thing is working so far.

[image]

Just for completeness, this is my config I use with this HW:

```yaml
substitutions:
  name: esp-samsung-ac

# BASIC CONFIGURATION  
esphome:
  name: ${name}

esp32:
  board: m5stack-atom

web_server:
  port: 80

# Enable logging
logger:

api:
  encryption:
    key: !secret esp_home_api_password

ota:
  password: !secret esp_home_ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${name}.o"
    password: !secret esp_home_fallback_wifi_password

captive_portal:

# AC CONFIGURATION  
uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: [samsung_ac]

samsung_ac:
  devices:
    - address: "20.00.00" # unit address

      # You could use the ESPHome/Homeassist climate control
      climate:
        name: "Ložnice"

      # And/or use seperate controls/sensors
      room_temperature:
        name: Test1_temp
      target_temperature:
        name: Test1_target
      power:
        name: Test1_power
      mode:
        name: Test1_mode
    
        # LIVING ROOM
    - address: "20.00.01" # unit address

      # You could use the ESPHome/Homeassist climate control
      climate:
        name: "Obývák"

      # And/or use seperate controls/sensors
      room_temperature:
        name: Test1_temp
      target_temperature:
        name: Test1_target
      power:
        name: Test1_power
      mode:
        name: Test1_mode

# DEVICE INFO
text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${name} IP adresa"
      icon: mdi:ip-network

sensor:
  - platform: wifi_signal
    name: "${name} signál"
    update_interval: 10s
  
  - platform: uptime
    name: "${name} doba běhu"
```

I had to change the RX, TX pins for the UART, but besides that it was pretty much straight forward.

lanwin, 2023-10-03:
> I have got it working, I am able to control both AC units it seems that every implemented thing is working so far.
> [image]
> 
> Just for completeness, this is my config I use with this HW:
> 
> ```yaml
> substitutions:
>   name: esp-samsung-ac
> 
> 
> 
> # BASIC CONFIGURATION  
> esphome:
>   name: ${name}
> 
> esp32:
>   board: m5stack-atom
> 
> web_server:
>   port: 80
> 
> # Enable logging
> logger:
> 
> api:
>   encryption:
>     key: !secret esp_home_api_password
> 
> ota:
>   password: !secret esp_home_ota_password
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: "${name}.o"
>     password: !secret esp_home_fallback_wifi_password
> 
> captive_portal:
> 
> 
> 
> # AC CONFIGURATION  
> uart:
>   tx_pin: GPIO19
>   rx_pin: GPIO22
>   baud_rate: 9600
>   parity: EVEN
> 
> external_components:
>   - source: github://lanwin/esphome_samsung_ac@main
>     components: [samsung_ac]
> 
> samsung_ac:
>   devices:
>     # BEDROOM
>     - address: "20.00.00" # unit address
> 
>       # You could use the ESPHome/Homeassist climate control
>       climate:
>         name: "Ložnice"
> 
>       # And/or use seperate controls/sensors
>       room_temperature:
>         name: Test1_temp
>       target_temperature:
>         name: Test1_target
>       power:
>         name: Test1_power
>       mode:
>         name: Test1_mode
>     
>      # LIVING ROOM
>     - address: "20.00.01" # unit address
> 
>       # You could use the ESPHome/Homeassist climate control
>       climate:
>         name: "Obývák"
> 
>       # And/or use seperate controls/sensors
>       room_temperature:
>         name: Test1_temp
>       target_temperature:
>         name: Test1_target
>       power:
>         name: Test1_power
>       mode:
>         name: Test1_mode
> 
> 
> 
> # DEVICE INFO
> text_sensor:
>   - platform: wifi_info
>     ip_address:
>       name: "${name} IP adresa"
>       icon: mdi:ip-network
> 
> sensor:
>   - platform: wifi_signal
>     name: "${name} signál"
>     update_interval: 10s
>   
>   - platform: uptime
>     name: "${name} doba běhu"
> ```
> 
> I had to change the RX, TX pins for the UART, but besides that it was pretty much straight forward.

👍🏻

bzumik1, 2023-10-03:
@lanwin what do you think about putting this HW to README.md on main page next to the DIY solution. This basically does not require any soldering you just connect it on bus using the same cables as you would use for WiFi kit and upload there the  ESPHome config. It is pretty final solution and easy for noobs like me who do not like soldering 😄 We could also add there links for documentation and buy so it would be even easier for new joiners. We could get more attention to this project. I think it is awesome and it deserves more publicity.