omerfaruk-aran/esphome_samsung_hvac_bus #5: Another controller on network

bzumik1, 2023-08-24:
Can this be used with another controller connected on the communication bus? I have wifi kit and I would like to use that one till we migrate all features to ESPHome.

lanwin, 2023-08-25:
Technically I see no reason against it. The protocol basically works like an event bus where every message can be received by every consumer.

closed 2023-08-25 by lanwin

bzumik1, 2023-08-25:
Thanks for confirmation.

bzumik1, 2023-10-03:
I have just installed both SAMSUNG WiFi Kit as well as ESPHome based controller and both works just fine, I even can see the address of the SAMSUNG WiFi Kit displayed under indoor units.

[image]

In my case the address is 62.00.8f

lanwin, 2023-10-07:
> I have just installed both SAMSUNG WiFi Kit as well as ESPHome based controller and both works just fine, I even can see the address of the SAMSUNG WiFi Kit displayed under indoor units.
> [image]
> 
> In my case the address is 62.00.8f

That is now fixed