omerfaruk-aran/esphome_samsung_hvac_bus #6: Power Consumption NASA

bzumik1, 2023-08-24:
Will it be possible to add power consumption for NASA devices? Also is there some way how to support you? You are doing great job!

lanwin, 2023-08-25:
Not sure. I think I have seen some power values. So when they exists it should be simple to add them.

Give me some time. My desk is currently pretty full.

bzumik1, 2023-08-25:
Fully understand that, I really love this project. If there would be anything I could help with please let me know.

lanwin, 2023-08-25:
Yes you can. You can help tests this and also help improve the documentation. I terms like. Did its understandable, is there anything missing. Adding working devices...etc.

bzumik1, 2023-08-25:
Ok, as soon as I will receive the necessary hardware I will start with testing.

Krzysiek1976, 2023-10-02:
Hi
good job :)
Greetings from Poland .
I installed it and it works on the model: AC052RNMDKG/AC0RXADKG.
Kris

bzumik1, 2023-10-02:
Hi, so you are able to see the power consumption? That is great! I should have my HW today, looking forward to test it as well 🙂

Krzysiek1976, 2023-10-02:
no, monitors via counter and Modbus. 
[image]

bzumik1, 2023-10-02:
Ok, I am little bit confused because you wrote 

>> I installed it and it works on the model: AC052RNMDKG/AC0RXADKG.

To issue regarding power consumption 🙂.

BTW how do you read it using ModBus? You have convertor from NASA to ModBus?

Krzysiek1976, 2023-10-02:
Sorry ,
operate basic functions:

Monitor room temperature
Change target temperature
Change AC mode

power consumption doesn't read

lanwin, 2023-10-03:
Hu it seems he deleted his posts.

A friend of mine uses the WIFI Kit from Samsung and there is a power consumption chart available. It must be possible the read that data. But I did not find the right message for that yet.

bzumik1, 2023-10-03:
I have WiFi Kit also available, right now I have replaced it with ESPHome with your code, but I could connect both if that would help somehow ...

lanwin, 2023-10-03:
Yes that would be a good idea. I did not know If the wifikit simply receives power messages or if it ask for it when the chart is accessed in the app.

bzumik1, 2023-10-03:
Ok, I will try to connect it today or tomorrow. I have quite complicated access to communication bus so it will take me some time 😄

bzumik1, 2023-10-03:
Hi @lanwin, I connected the WiFi Kit back again and checked the power consumption in App, I did so at **16:40:00**, here are the logs:

[esp-samsung-ac-power-consumption.log](https://github.com/lanwin/esphome_samsung_ac/files/12793730/esp-samsung-ac-power-consumption.log)

hopefully it will be helpful.

bzumik1, 2023-10-04:
Hi @lanwin did you have time to check the the logs? Was there any useful information? If not let me know if I should test something else.

lanwin, 2023-10-05:
@bzumik1 currently the log contains only stuff I didnt filter out. There are a lot more messages you can not see. I will add an config property to enable logging soon.

bzumik1, 2023-10-05:
Oh I see, let me know what the more detailed logging would be available and I will record the logs once again.

bzumik1, 2023-10-08:
Hi @lanwin, how does it look with the advanced logging?

sredfern, 2023-10-16:
Hey @bzumik1 - I'm actually planning on picking this up. Had a NASA 3 head unit installed in my house last week and stumbled across this project. I have a seperate current clamp measuring the AC codenser for real power usage and have found that the Samsung app is under reporting usage by around 60%. 🤞 for PR's coming in the next few weeks. 

Happy to help this side of the project as I'm pretty excited about exploring this.

bzumik1, 2023-10-16:
That sounds great. Thank you @sredfern

bzumik1, 2023-10-23:
Hi @lanwin is there any update on this?

bzumik1, 2023-10-23:
Hi @sredfern, how do you progress? Do you need some help with testing?

sredfern, 2023-10-24:
[image]

Still working on getting my hardware up. Have built a seperate rs485 listener with a PiZeroW that I want to connect at the same time so I can dump the data into a database for analysis.

bzumik1, 2023-10-25:
It looks awesome!

bzumik1, 2023-11-01:
Hi @sredfern were you able to read the data and load them to DB?

sredfern, 2023-11-02:
Hey @bzumik1 This is a hobby project for me and I've got a whole life outside of this (kids, job, house, life). It's pretty demotivating for someone to just endlessly comment on stuff asking to follow up. If you're so excited about working on it (and have the time to learn) can I suggest you attempt to fix it? I won't speak for the creator of this project but trying to solve the problem yourself is the greatest contribution you can make to this project, not following up every week.

bzumik1, 2023-11-02:
Hi @sredfern I understand your perspective and apologize if I've bothered you. Unfortunately, due to my limited knowledge, I can't assist with the code, but I've tried to improve the documentation and keep the developed features up to date. My interest was based on a positive approach and supporting the project, not on demotivation. Thank you for your work on the project, and I want to express my gratitude for what you've accomplished so far.

sredfern, 2023-11-04:
It's really great that you're excited but the way to keep other people excited isn't to bug them about getting something done. It's not hard to work out how it all works. Reminding people who can do this stuff how little time they have is a sure fire way of people losing interest. 

I managed to not need my raspberry pi method. Have managed to used the UARTDebug to spam the log with all the data from the RS485 bus.

`
logger:
  tx_buffer_size: 1024

uart:
  tx_pin: GPIO26
  rx_pin: GPIO18
  rx_buffer_size: 1024
  baud_rate: 9600
  parity: EVEN
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, '.');`

sredfern, 2023-11-06:
OK, managed to get all of the commands into my local Loki instance. Going to start digging around and seeing what I can make now.

If interested, make sure your syslog supports udp packets :
```
external_components:
  - source: github://TheStaticTurtle/esphome_syslog
    components: [syslog]
  - source: github://lanwin/esphome_samsung_ac@main
    components: [samsung_ac]

syslog:
  ip_address: "XX.XX.XX.XX"
  port: 514
  min_level: VERY_VERBOSE
  enable_logger: true
```

[image]

sredfern, 2023-11-12:
Had some better success understanding the protocol using this uart config if others are keen. 

```
uart:
  tx_pin: GPIO26
  rx_pin: GPIO18
  rx_buffer_size: 1024
  baud_rate: 9600
  parity: EVEN
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: [0x34]
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ' ');
```

bzumik1, 2023-11-12:
Thanks for updates!

bzumik1, 2023-11-12:
@sredfern may I ask what program you are using to analyse those logs, it seems quite handy.

lanwin, 2023-11-13:
@sredfern thank you for your effort!

But I am not sure what you do there. If you want to know how to decode the protocol you simply have to look at https://github.com/lanwin/esphome_samsung_ac/blob/main/components/samsung_ac/nasa.cpp . There is everything you need to know on how to decode the protocol.

The point is that the NASA protocol is pretty generic. Its not like the non NASA protocol witch is really specific. The NASA protocol is simply like a way to send generic variables as messages out over the wire. 

The difficult part is to find witch variable does what. If I wouldn't be able to decompile the source of SNET, it would have take much much longer to come up with this project. But even SNET does not references all variables I receive.

sredfern, 2023-11-13:
Thanks @lanwin

My goal here is to build a little application that can take the variable stream from the NASA protocol with the timestamp, message number and variable into a time series database I can then start exploring and comparing to my HA instance with the smart things integration enabled.

From there I'm hoping to start trying to reverse engineer some of the variables:
1. Which are dependent on the heat pump to be turned on?
2. What message numbers are clustered together into message sets?
3. What is the usual time between transmission between ?
4. Which are constant environmental variables from the condenser and if they're worth extracting? (ie: pretty sure there are temperature sensors in the condenser, would be interesting to explore that)

> There is everything you need to know on how to decode the protocol.
As mentioned before, I'm fluent in Javascript, SQL and Python. I can hack together C code but I struggle to fully understand it. By day I'm a data scientist, embedded systems are a hobby for me. What I'm hoping to bring to your project is my deductive skills in working out what some of these messages mean. To do that I need to get the data in a format I understand.

sredfern, 2023-11-13:
> what program you are using to analyse those logs

I'm using a log analysis stack... It's likely overkill for this use case but I use it for a few other projects. Here are the applications + data flow. 

esphome - using TheStaticTurtle/esphome_syslog to send log to server
syslog-ng - Syslog server, accepts packets from esphome instance 
Docker Stack (
promtail - grafana log ingestion pipeline 
loki - log query server
grafana - web front end
)
While overkill it does give me a really good long view of all the messages from this Samsung NASA protocol. I should be able to keep all the data indefinitely. I don't expect Samsung to be able to reprogram the heads remotely but if they do we should be able to grab the firmware. What I find amazing is Samsung is clearly using these wifi units to capture usage data about their installed unit base. 

The amount of data streaming from these units is INSANE. The reason I think this project is so impactful is it provides a clear path for you as a user to take control of this data and maintain functionality of using your phone to turn on your AC unit.

lanwin, 2023-11-13:
There is now a better way. If you pull and configure an MQTT client in ESPHome the its dumps all messages variables it receives to the topic test/nasa/{id}.

This way you can simply write code to dump the mqtt messages to grafana.

Currently this is enabled by default. I need to think about whats the best way to enable this via configuration.

sredfern, 2023-11-14:
That would work great. Thank you!

I'm having a problem installing it on my unit. I had to add the [matt](https://esphome.io/components/mqtt) component. Getting this message: 

```
INFO ESPHome 2023.11.0b5
INFO Reading configuration /config/esphome/samsung_ac_control.yaml...
INFO Generating C++ source...
ERROR ID mqtt_mqttselectcomponent is already registered
``` 

When you remove the mqtt component: 

```
INFO ESPHome 2023.11.0b5
INFO Reading configuration /config/esphome/samsung_ac_control.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing samsung-ac-control (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/samsung-ac-control/src/esphome/components/samsung_ac/nasa.cpp.o
Compiling .pioenvs/samsung-ac-control/src/esphome/components/samsung_ac/samsung_ac_device.cpp.o
Compiling .pioenvs/samsung-ac-control/src/esphome/components/samsung_ac/util.cpp.o
src/esphome/components/samsung_ac/nasa.cpp:7:10: fatal error: esphome/components/mqtt/mqtt_client.h: No such file or directory
 #include "esphome/components/mqtt/mqtt_client.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/samsung-ac-control/src/esphome/components/samsung_ac/nasa.cpp.o] Error 1
========================== [FAILED] Took 3.78 seconds ==========================
```

Below is my configuration file: 
```
esphome:
  name: samsung-ac-control

esp32:
  board: nodemcu-32s

substitutions:
  unit00: main_bedroom
  unit01: kids_bedroom
  unit02: livingroom

# Enable logging
logger:

external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: [samsung_ac]

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
....

 mqtt:
   broker: ...

captive_portal:
    
web_server:
  port: 80

uart:
  tx_pin: GPIO26
  rx_pin: GPIO18
  rx_buffer_size: 1024
  baud_rate: 9600
  parity: EVEN

samsung_ac:
  devices:
    - address: "20.00.00"
      climate:
        name: "${unit00} Heat Pump"
      room_temperature:
        name: ${unit00}_room_temp
      target_temperature:
        name: ${unit00}_target_temp
      power:
        name: ${unit00}_power
      mode:
        name: ${unit00}_mode
      room_humidity:
        name: ${unit00}_room_humidity
...

```

Is this very different to your configuration file @lanwin ?

lanwin, 2023-11-14:
You simple need to add this:

```
mqtt:
  broker: IP
  username: !secret mqtt_user
  password: !secret mqtt_password
  discovery: false
```

lanwin, 2023-11-14:
I would suggest to use "Clean build files" in ESPHome. MQTT seems to be a little bit special there.

sredfern, 2023-11-15:
Hey Steve,

Thanks for your help. I've tried the following:
1. Clean rebuild
2. I had originally used the [mqtt HA auto configuration](https://esphome.io/components/mqtt#using-with-home-assistant) but tried using a manual mqtt (with username and password) with the same error very early in the compile step:

I've extracted out the verbose compile step: 

```
DEBUG Running to_code in esphome.components.mqtt (num 10)
DEBUG Adding: // mqtt:
DEBUG Adding: //   broker: 192.168.*****
//   username: samsung_ac
//   password: ******
//   discovery: false
//   id: mqtt_mqttclientcomponent
//   port: 1883
//   discovery_retain: true
//   discovery_prefix: homeassistant
//   discovery_unique_id_generator: legacy
//   discovery_object_id_generator: none
//   use_abbreviations: true
//   topic_prefix: samsung-ac-control
//   keepalive: 15s
//   reboot_timeout: 15min
//   birth_message:
//     topic: samsung-ac-control/status
//     payload: online
//     qos: 0
//     retain: true
//   will_message:
//     topic: samsung-ac-control/status
//     payload: offline
//     qos: 0
//     retain: true
//   shutdown_message:
//     topic: samsung-ac-control/status
//     payload: offline
//     qos: 0
//     retain: true
//   log_topic:
//     topic: samsung-ac-control/debug
//     qos: 0
//     retain: true
DEBUG Adding global: mqtt::MQTTClientComponent *mqtt_mqttclientcomponent;
DEBUG Adding: mqtt_mqttclientcomponent = new mqtt::MQTTClientComponent();
DEBUG Registered variable mqtt_mqttclientcomponent of type mqtt::MQTTClientComponent
DEBUG Adding: mqtt_mqttclientcomponent->set_component_source("mqtt");
DEBUG Adding: App.register_component(mqtt_mqttclientcomponent);
DEBUG Adding library: ottowinter/AsyncMqttClient-esphome@0.8.6
DEBUG Adding define: <esphome.core.Define object at 0x10b141610>
DEBUG Adding global: using namespace mqtt;
DEBUG Adding: mqtt_mqttclientcomponent->set_broker_address("192.168.*****");
DEBUG Adding: mqtt_mqttclientcomponent->set_broker_port(1883);
DEBUG Adding: mqtt_mqttclientcomponent->set_username("samsung_ac");
DEBUG Adding: mqtt_mqttclientcomponent->set_password("******");
DEBUG Adding: mqtt_mqttclientcomponent->disable_discovery();
DEBUG Adding: mqtt_mqttclientcomponent->set_topic_prefix("samsung-ac-control");
DEBUG Adding define: <esphome.core.Define object at 0x10b141a50>
DEBUG Adding: mqtt_mqttclientcomponent->set_birth_message(mqtt::MQTTMessage{
    .topic = "samsung-ac-control/status",
    .payload = "online",
    .qos = 0,
    .retain = true,
});
DEBUG Adding: mqtt_mqttclientcomponent->set_last_will(mqtt::MQTTMessage{
    .topic = "samsung-ac-control/status",
    .payload = "offline",
    .qos = 0,
    .retain = true,
[... 119 lines cut ...]
    CORE.flush_tasks()
  File "/usr/local/Cellar/esphome/2022.12.5/libexec/lib/python3.11/site-packages/esphome/core/__init__.py", line 616, in flush_tasks
    self.event_loop.flush_tasks()
  File "/usr/local/Cellar/esphome/2022.12.5/libexec/lib/python3.11/site-packages/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/usr/local/Cellar/esphome/2022.12.5/libexec/lib/python3.11/site-packages/esphome/__main__.py", line 168, in wrapped
    await coro(conf)
  File "/Users/sredfern/projects/esp_samsung/.esphome/external_components/11b80f5a/components/samsung_ac/__init__.py", line 135, in to_code
    await select.register_select(sel, conf, options=values)
  File "/usr/local/Cellar/esphome/2022.12.5/libexec/lib/python3.11/site-packages/esphome/components/select/__init__.py", line 82, in register_select
    await setup_select_core_(var, config, options=options)
  File "/usr/local/Cellar/esphome/2022.12.5/libexec/lib/python3.11/site-packages/esphome/components/select/__init__.py", line 74, in setup_select_core_
    mqtt_ = cg.new_Pvariable(config[CONF_MQTT_ID], var)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/esphome/2022.12.5/libexec/lib/python3.11/site-packages/esphome/cpp_generator.py", line 578, in new_Pvariable
    return Pvariable(id_, rhs)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/esphome/2022.12.5/libexec/lib/python3.11/site-packages/esphome/cpp_generator.py", line 560, in Pvariable
    CORE.register_variable(id_, obj)
  File "/usr/local/Cellar/esphome/2022.12.5/libexec/lib/python3.11/site-packages/esphome/core/__init__.py", line 746, in register_variable
    raise EsphomeError(f"ID {id} is already registered")
esphome.core.EsphomeError: ID mqtt_mqttselectcomponent is already registered
```

@bzumik1 - Could you try on your instance and see if it gets an early error?

lanwin, 2023-11-15:
Can you post your full yaml. I want to try if I get the same error there.

sredfern, 2023-11-16:
Hi Steve,

By 3 head units I mean I have 3 indoor devices.

Below is my configuration. Thank you for your help.

```
esphome:
  name: samsung-ac-control

esp32:
  board: nodemcu-32s

substitutions:
# Change the disp_name to something you want  
  unit00: main_bedroom
  unit01: kids_bedroom
  unit02: livingroom

# Enable logging
logger:

external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: [samsung_ac]

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "aasdfasdfasfd"
  password: "asdfasfdasfe"

mqtt:
  broker: 192.168.
  username: samsung_ac
  password: passwd
  discovery: false

captive_portal:
    
web_server:
  port: 80

uart:
  tx_pin: GPIO26
  rx_pin: GPIO18
  rx_buffer_size: 1024
  baud_rate: 9600
  parity: EVEN

samsung_ac:
  devices:
    - address: "20.00.00"
      climate:
        name: "${unit00} Heat Pump"
      room_temperature:
        name: ${unit00}_room_temp
      target_temperature:
        name: ${unit00}_target_temp
      power:
        name: ${unit00}_power
      mode:
        name: ${unit00}_mode
      room_humidity:
        name: ${unit00}_room_humidity
    - address: "20.00.01"
      climate:
        name: "${unit01} Heat Pump"
      room_temperature:
        name: ${unit01}_room_temp
      target_temperature:
        name: ${unit01}_target_temp
      power:
        name: ${unit01}_power
      mode:
        name: ${unit01}_mode
      room_humidity:
        name: ${unit01}_room_humidity
    - address: "20.00.02"
      climate:
        name: "${unit02} Heat Pump"
      room_temperature:
        name: ${unit02}_room_temp
      target_temperature:
        name: ${unit02}_target_temp
      power:
        name: ${unit02}_power
      mode:
        name: ${unit02}_mode
      room_humidity:
        name: ${unit02}_room_humidity
```

lanwin, 2023-11-16:
When you remove the mode: parts it works. I am not sure why this is. Seems to be some special magic for the MQTT parts behind the scenes.

sredfern, 2023-11-17:
Hi Steve, thanks for debugging this with me.

I can see some data in MQTT but nothing relating to the test/nasa structure you've made here [here](https://github.com/lanwin/esphome_samsung_ac/blob/main/components/samsung_ac/nasa.cpp#L459-L475)

[image]

I would be expecting to see a test topic. Searching within the structure shows nothing as well.

lanwin, 2023-11-19:
> Hi Steve, thanks for debugging this with me.
>
> I can see some data in MQTT but nothing relating to the test/nasa structure you've made here [...]

But you see incoming data within the log?

sredfern, 2023-11-19:
I’m not sure I can. This is the two parts of the logs I think is relevant: 
[image]
[image]

I can see all the HA presented values being transmitted to mqtt. I’m unable to see any of the codes and values. 

FYI, I’m travelling for work so this post isn’t as detailed as I would have liked. Thank you for all your help here Steve.

sredfern, 2023-12-10:
Hi Steve, back from travel. Great to see your progress with debugging tools. Thanks so much for all the work you have put in here, I really appreciate it. 

I have been playing around with it on my end and the problem I'm coming up with is the client never connects. What's more strange is if I configure ESPHome's MQTT interface I can see it connect: 

**From yaml**
```
mqtt:
  broker: 192.168.86.141
  username: samsung_ac
  password: passwd
  discovery: false
```

from Mosquitto logs
```
2023-12-09 15:12:23: New connection from 192.168.86.188:63147 on port 1883.
2023-12-09 15:12:23: New client connected from 192.168.86.188:63147 as samsung-ac-control-d4d4da83e554 (p2, c0, k15, u'samsung_ac').
```

From there, I can see the usual output you expect from connecting a esphome client to an MQTT instance.
[image]

However, when I remove the esphome native MQTT configuration and put the configuration into the samsung_ac part of the config:

```
samsung_ac:
  debug_mqtt_host: 192.168.86.141
  debug_mqtt_port: 1883
  debug_mqtt_username: samsung_ac
  debug_mqtt_password: passwd
  debug_log_messages: false # logs all messages - only for debugging
  debug_log_messages_raw: false # logs the raw message bytes from all messages - only for debugging
```

I get nothing in the log from the Mosquitto logs.
 
I have a question - It seems like in [debug_mqtt.cpp](https://github.com/lanwin/esphome_samsung_ac/blob/46d5c6e7cfeda0cc434232a706850d2366107960/components/samsung_ac/debug_mqtt.cpp#L4) you're using USE_ESP8266 that only runs the code if you're running an 8266. Would this execute with an esp32? 

I'm using a ESP32 so that could be why it won't connect to my MQTT instance. 

Let me know your thoughts. Thanks

Foxhill67, 2023-12-10:
Hi Steve, thanks for the great work done so far! I am setting up an ESP32 / TTL to RS485 system to monitor my Samsung EHS heatpump (inside AE260RNWSGG/EU outside AE090RXEDGG/EU). 

I too do not see a connection made to my MQTT instance, probably because in debug_mqtt.cpp a connection is only made if USE_ESP8266 is defined (which clearly is not the case when using an ESP32)?

lanwin, 2023-12-11:
Yes it is true. Currently the code only works with ESP8266 cause for ESP32 I need to use another client. Did not tried that yet.

lanwin, 2024-01-02:
I added support for ESP32. But I am not able to test it, since I have no ESP32 here. So please try.

Foxhill67, 2024-01-02:
Tested with ESP32 but get compile errors:
[logs_esp-samsung-ehs_run.txt](https://github.com/lanwin/esphome_samsung_ac/files/13809382/logs_esp-samsung-ehs_run.txt)

May be #include "mqtt_client.h" is missing? See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html

lanwin, 2024-01-02:
Ah sorry I compiled against the wrong code and did not noticed the build errors. Now it should compile.

Foxhill67, 2024-01-02:
Yes, now works like a charm! tnxs.

sredfern, 2024-01-05:
@lanwin - Can confirm working like a charm. Have started dumping the data for analysis.
[image]

FYI I have a fork I'm exploring to also add in the device from and to as part of the topic in MQTT. You can check it out [here](https://github.com/sredfern/esphome_samsung_ac). If I find it useful I'll create a PR to this project.

lanwin, 2024-01-05:
Sounds good! 👍🏻

mbo18, 2024-02-02:
Just updated all my ESP and I now have this in my ESP logs:

```
[17:21:49][W][samsung_ac:560]: s:10.00.00 d:b0.00.ff NASA_OUTDOOR_CONTROL_WATTMETER_1UNIT  30.000000
[17:21:49][W][samsung_ac:554]: s:10.00.00 d:b0.00.ff LVAR_OUT_CONTROL_WATTMETER_1W_1MIN_SUM 30.000000
[17:21:49][W][samsung_ac:547]: s:10.00.00 d:b0.00.ff LVAR_OUT_CONTROL_WATTMETER_ALL_UNIT_ACCUM 1589.247000kwh
```
Does that means that you will be able to add power consumption sensors??
Would be really nice.

lanwin, 2024-02-02:
Yes I added test code to check if there are coming valid data since @Foxhill67 has provided the corresponding here https://github.com/lanwin/esphome_samsung_ac/issues/28

atanasenko, 2024-02-17:
This is currently my configuration for electricity/power related sensors for outdoor unit AJ068TXJ3KG/EU:
```
  devices:
    - address: "10.00.00"
      outdoor_temperature:
        name: AC Outdoor temperature
      custom_sensor:
        - name: AC Voltage
          message: 0x24fc
          device_class: voltage
          state_class: measurement
          unit_of_measurement: V
        - name: AC Current
          message: 0x8217
          device_class: current
          state_class: measurement
          unit_of_measurement: A
          accuracy_decimals: 1
          filters:
            - multiply: 0.1
        - name: AC Power
          message: 0x8413
          device_class: energy
          state_class: measurement
          unit_of_measurement: W
        - name: AC Energy
          message: 0x8414
          device_class: energy
          state_class: total_increasing
          unit_of_measurement: KWh
          accuracy_decimals: 1
          filters:
            - multiply: 0.001
```

starsalmon, 2024-04-26:
> This is currently my configuration for electricity/power related sensors for outdoor unit AJ068TXJ3KG/EU:
>
> ```
>   devices:
>     - address: "10.00.00"
>       outdoor_temperature:
>         name: AC [...]

Thank you so much, I could see the readings in the logs but had no idea how to use them as entities. 2 small issues for anyone copy pasting this:
AC Power should be device_class: power
AC energy unit_of_measurement needs to be kWh instead of KWh

Tycorc, 2024-05-25:
Can also confirm the custom sensor config from @mbo18  works perfectly fine with my AJ050TXJ2KG

[4 more replies]

closed 2024-10-29 by omerfaruk-aran