omerfaruk-aran/esphome_samsung_hvac_bus #7: AC configuration using ESPHome

bzumik1, 2023-09-26:
Hi, it should be possible to configure indoor/outdoor unit over the communication bus. This would allow to read/modify some AC settings such as disable sound, disable/enable pump for condensate etc. Do you think it would be possible? This would really help because the configuration of AC using IR controller is terrible. Just reading the set values would be great, allowing modifications would be awesome.

eddystonestan, 2023-09-26:
I'm also interested in attempting to configure the unit over the F1/F2 comms bus (for EHS heatpump). What documentation do you have @bzumik1? @lanwin where did you find the NASA message protocol within the S-NET application? Trying to find it when decompiling but no luck so far...

lanwin, 2023-09-26:
In theory that should be possible. The tricky part is to find the right messages to change the values. But I am also not sure if the indoor units really allow all configuration over the bus. 

BTW. Why someone should disable the condensate pump?

lanwin, 2023-09-26:
> I'm also interested in attempting to configure the unit over the F1/F2 comms bus (for EHS heatpump). What documentation do you have @bzumik1? @lanwin where did you find the NASA message protocol within the S-NET application? Trying to find it when decompiling but no luck so far...

The basic stuff (parsing etc) should be within the shared.dll. But the message definitions are stored as xml in "C:\Program Files (x86)\S-NET pro2\Projects\RuleScript\NASA.ptc".

bzumik1, 2023-09-26:
> In theory that should be possible. The tricky part is to find the right messages to change the values. But I am also not sure if the indoor units really allow all configuration over the bus. 
> 
> 
> 
> BTW. Why someone should disable the condensate pump?
> 
> 
> 
> 

I am pretty sure they allow the confirmation over the bus, when I was setting up my unit I watched a lot of SAMSUNG documentation videos where they were setting the indoor units using PC with appropriate converter as well as the samsung thermostat.

lanwin, 2023-09-27:
> Have found the NASA file, taking me a while to work out what's going on in there! Did you create a list of known commands and their structure somewhere for the NASA protocol (just so I don't do this myself if it's already been done and listed somewhere!). I have the installation document for the heat pump, but it wasn't particularly useful in terms of communications other than saying which FSVs could be set and what their permitted ranges were.

Yes but only for the messages I need to use. https://github.com/lanwin/esphome_samsung_ac/blob/9004cedcf8530a09addb35636af5f0c56323a283/components/samsung_ac/nasa.cpp#L395