omerfaruk-aran/esphome_samsung_hvac_bus #8: Support for **09/12AVH** Model

lucassant95, 2023-09-30:
Hey folks! Unfortunately upon installation I encountered a problem where the version of my AC doesn't have a F1/F2 port. Only a C port! I don't quite know if It is possible to start working on a integration for this model since It doesn't look to follow the standards of this project. Let me know if I can help in anything! Tks :D

bzumik1, 2023-09-30:
Hi, what is product number of your AC? I can not find anything under "samsung 09/12AVH"

lucassant95, 2023-09-30:
Sorry, It was the reference in the Manual. The model is AR12AVHABWKNAZ

lanwin, 2023-10-01:
Its pretty hard to say. From Google it seems your device is sold in Brasil right? Maybe Samsung simply needs to use different naming in some countries. 

But I guess it still uses two wires? One would make no sens. 

What you could do is to buy a simple RS485 USB devices (they should be expensive) and check if you could get any data with Samsungs SNET 1 or 2 software (you could find it via google). If that works, this component should also work.

lucassant95, 2023-10-01:
Yes! I'm from Brasil haha
But indeed it's only 1 Cable for communication and two for power. The communication only has a C as label. The fourth cabe seems to be ground[image]

lanwin, 2023-10-01:
I would guess that the second wire is the ground connector on the left side. Did you have a multimeter? Then you need to make sure that there is no hight voltage on the connection between ground and C. 

But be careful! On 1(N) and 2(L) should have 110/223 volt.

lanwin, 2023-11-19:
For now I close this issue. If you have further questions. Please open a new discussion.

closed 2023-11-19 by lanwin