omerfaruk-aran/esphome_samsung_hvac_bus #10: Instalation of firmware does not work

bzumik1, 2023-10-02:
Hi @lanwin, 

I have finally received the HW and I have tried to put the firmware on the ESP, but I am getting this error:

[image]

Any idea how to fix it?

bzumik1, 2023-10-02:
I had to replace the `https://github.com/lanwin/esphome_samsung_ac@main` from example.yaml with `github://lanwin/esphome_samsung_ac@main` now it seems to be working. Not sure if this is issue on my side only or if we should update the example.yaml

bzumik1, 2023-10-02:
Based on the [documentation](https://esphome.io/components/external_components.html) it looks like there is an issue in your example.yaml file. It either needs to be specified like this

``` yaml
external_components:
  - source:
      type: git
      url: https://github.com/lanwin/esphome_samsung_ac
      ref: main
    components: [samsung_ac ]
```

or like this

``` yaml
external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: [samsung_ac]
```

the second one looks better to me so I would correct the example.yaml to that version but it is up to you.

lanwin, 2023-10-03:
Thank you. I did not noticed this cause I use a local source for dev. I fixt the example file.

closed 2023-10-03 by lanwin