omerfaruk-aran/esphome_samsung_hvac_bus #9: PR to ESPHome Code Base

bzumik1, 2023-10-02:
Hi, is there plan to merge this component to ESPHome code base? I think it would be nice touch. I fully understand that we need to test it properly first but I think that with this functionality we have right now it would be nice 1.0 version.

lanwin, 2023-10-03:
I am absolutely not sure about that. I did not know what requirements they have to do that. Maybe later. But for now I would like to focus on adding some features and fixing issues.

bzumik1, 2023-10-03:
I totally agree, I just wanted to open this so at point when we will feel comfortable it si working we would create PR to ESPHome. I have seen some other AC brands are already there. But this would be just nice thing to have, I agree it has lowest priority.

TCB13, 2023-12-04:
+1 on this. Thank you.