omerfaruk-aran/esphome_samsung_hvac_bus #12: [NASA] blades direction

bzumik1, 2023-10-03:
Hi, I am opening this to add support for setting the blades direction on indoor units. This would probably also cover special modes for blades such as "Long Wind" etc.

bzumik1, 2023-10-23:
Hi @lanwin, I can imagine you have a lot of other work besides this project, but would it be possible to create some tracker on what you are features are you currently working on?

mbo18, 2023-12-11:
Hi!

Here are debug logs from my HVAC when changing swing mode:
- At 17:35:00: vertical 
- At 17:35:29: horizontal
- At 17:36:02: fix
- At 17:36:21: all

[Swing mode.txt](https://github.com/lanwin/esphome_samsung_ac/files/13638428/Swing.mode.txt)

Thank you

lanwin, 2023-12-12:
@mbo18 thank you! That will help. But don't expect a fast change. My time is pretty limited.

PR #81 (2024-02-07): Swing mode
Support for swing mode (wind direction/blades direction).
This one is a tiny less straightforward since the climate value for the mode is single enum, while nasa has two messages for horizontal and vertical separately.

Tested out on AR09TXFCAWKNEU.

Again, non_nasa only gets a placeholder.

Fixes #12
closed 2024-02-07 by lanwin