omerfaruk-aran/esphome_samsung_hvac_bus #15: [COMMON] Group Data for Indoor Units

bzumik1, 2023-10-05:
It would be nice to be able to group the data based on the indoor unit such as temperature, power consumption etc. This would be very useful especially when you have multiple indoor units. Currently all the data we read are placed under common sensor section in Home Assistant.

There are two possible solution which needs more investigation. One is to create separate device per indoor unit, this doesn't look to be possible right now, I have opened [this issue](https://github.com/esphome/feature-requests/issues/2420) in ESPHome repository so maybe this feature would be implemented. Another option would be to group the data under custom climate component (or another component if necessary), I was inspired by custom [implementation of Haier AC](https://esphome.io/components/climate/haier) where it seems to be possible to add additional information under climate component.

lanwin, 2023-10-05:
I am not sure if the custom climate component really would make any difference. In fact the climate property from my configuration creates an custom climate component under the hood.

bzumik1, 2023-10-05:
Have you checked the [Haier implementation on GitHub](https://github.com/paveldn/haier-esphome)? They are adding some additional fields there. Unfortunately, I am not skilled enough to go through the code, so maybe you are right, and that is not the way.

lanwin, 2023-10-05:
Yes but I see nothing which could help in cause of grouping.

lanwin, 2024-02-06:
Did you found anything useful in the mean time?

bzumik1, 2024-02-07:
Currently, it appears that the feature in question is not available. However, as pointed out by [mbo18](https://github.com/mbo18) in [this discussion](https://github.com/esphome/feature-requests/issues/2420), there seems to be ongoing progress towards its implementation. It is definitely worth keeping an eye on. Notably, [kpfleming](https://github.com/kpfleming) is actively working on developing this feature, and his proposed solution appears to be quite promising.

closed 2024-10-29 by omerfaruk-aran