omerfaruk-aran/esphome_samsung_hvac_bus #17: [NASA] Logging Devices on Bus

bzumik1, 2023-10-05:
Currently when the devices connected devices are logged the output looks like this:

```log
[13:50:47][C][samsung_ac:107]: registered devices: 20.00.00, 20.00.01
[13:50:47][C][samsung_ac:122]: known indoor devices: 20.00.00, 20.00.01, 62.00.8f
[13:50:47][C][samsung_ac:123]: known outdoor devices: 10.00.00
```

I would rename the "known devices" to "Discovered devices" also it would be nice to start the log with capital letter on beginning to keep it consistent with other log messages. Also please check that my WiFi controller was put to indoor devices category, it would be nice to put it separately in "Other devices" or you can even put it to correct category based on the AddressClass [here](https://github.com/lanwin/esphome_samsung_ac/blob/main/components/samsung_ac/nasa.h).

P.S. is there some way how to set priority on the issue? This is very, very, very low priority 😄

lanwin, 2023-10-05:
If you open another 20 issues ... then we should think about a priority system :D

bzumik1, 2023-10-05:
Totally agree 😄. I thought there was at least some upvote mechanism on GitHub, but I do not see it. Maybe that is something optional and needs to be allowed for the repository; I'm not sure.

P.S. Please don't get discouraged by the number of issues; I am just opening them as I find some things that could be improved so they can be fixed later. Unfortunately, I do not know how to code ESPHome components, so I cannot help there. But if there is anything you think I can assist with, please let me know.

bzumik1, 2023-10-05:
By the way, maybe it would be great to have a roadmap in README.md so we could see what the developer (currently just you) is working on.

lanwin, 2023-10-07:
Is now fixed.

closed 2023-10-07 by lanwin

bzumik1, 2023-10-07:
Checked and looks good, thx for implementation.

[image]