omerfaruk-aran/esphome_samsung_hvac_bus #18: And support for flow_control_pin

SebuZet, 2023-10-05:
Hi,

Firrst of all thenks for great work. I was wondering if it is possible to use modbus device instead of uart?
I've got device with MAX485 chip and it woon't work correctly without flow control. Maybe youo can add a flow control pin supprt?

More info: https://electronics.stackexchange.com/questions/153500/correctly-using-re-and-de-with-rs485

Thanks,
Seba

lanwin, 2023-10-06:
Hi Seba, what do you mean by flow_control_pin? For UART?

Adding modbus would make no sens. RS485 is only a transport bus, like Ethernet. This projects adds the software protocol which Samsung uses on RS485.

Modbus has both. An hardware bus and an software protocol. And both is already supported by home assist. All you need to do is to review the documentation and configure the matching registers within home assist.

SebuZet, 2023-10-06:
Thanks for your response. I'm sorry for my ignorance. I thought Samsung HVAC is using modbus for communication between devices.
I've got [Olimex RS485](https://www.olimex.com/Products/Modules/Interface/MOD-RS485/open-source-hardware). To make it working with modbus device I had to configure **flow_control_pin** in modbus controller in ESPHome. My understanding is that MAX485 chip is a half-duplex chip. It has to be switched between read and write modes using this pin.
After connecting this to my AC I've got connection error (E01). I thought it is because of that.

lanwin, 2023-10-07:
Ah ok from what I understood this device allows to use modbus via an RS485 transport. This will not work since the Samsung ACs dont talk modbus as you have written.

Sorry! You need another device to uses this.

closed 2023-10-07 by lanwin