omerfaruk-aran/esphome_samsung_hvac_bus #20: [Common] Wrong description

SebuZet, 2023-10-09:
Him 

I'm trying to connect my device. I've received M5Stack devices. It looks like there is no comunnication detected.
My device is AR12NXCXAWKNEU - I believe it is Non NASA version. Should it work?

Ps. There is an error in Plug and Play Hardware instalation section:

> Connect the components as follows:
> 
> Connect F1 on the AC unit to B on the M5STACK controller.
> Connect F2 on the AC unit to A on the M5STACK controller.
> Connect V1 on the AC unit to DC on the M5STACK controller.
> Connect V2 on the AC unit to G on the M5STACK controller.

On the picture there is F1 <-> A and F2 <-> B connection.

lanwin, 2023-10-10:
Hi @SebuZet, @bzumik1 has added the documentation for the M5STACK controller. I hope he can help you there.

I will add a debug option soon to display the raw messages so its simpler to check if there is any data.

bzumik1, 2023-10-10:
Hi @SebuZet, @lanwin let me check where could be issue.

bzumik1, 2023-10-10:
Hi @SebuZet the picture is correct, I will correct the documentation.

SebuZet, 2023-10-11:
Stilo nothing. I connected cables in good way - I compared description with DIY section and I knew that image was correct.
THere is still nothing detected.

bzumik1, 2023-10-11:
Have you set up correctly the RX and TX pins in configuration?

SebuZet, 2023-10-11:
I dbelive. I'm using this elements:
https://shop.m5stack.com/products/atom-lite-esp32-development-kit
https://shop.m5stack.com/products/rs485-module
Connect with provided cable. According to devkit description I should use GPIO 26 and 32.
I will check this one more time

bzumik1, 2023-10-11:
Try this:

```yaml
tx_pin: GPIO19
rx_pin: GPIO22
```

it is working for me

bzumik1, 2023-10-11:
Your pins are switched 😉

SebuZet, 2023-10-11:
Pins are fine. RX - TX and TX - RX.
After a long period of time it looks like it is working:

> [19:29:39][C][samsung_ac:025]: Configured devices: 20.00.00
> [19:29:39][C][samsung_ac:045]: Discovered devices:
> [19:29:39][C][samsung_ac:046]:   Outdoor: 00
> [19:29:39][C][samsung_ac:047]:   Indoor:  c8

SebuZet, 2023-10-11:
Yes, it is working. Thank you very much and sorry for the problem.
Nów, I will create a branch of you repo and start playing with source.
Do you have some additional docs/data anout Non NASA protocol?

SebuZet, 2023-10-11:
Is it normal that sending power on command takes at least few seconds?

lanwin, 2023-10-12:
> Is it normal that sending power on command takes at least few seconds?

Yes. It takes a moment until the AC executes the command.

escees, 2023-11-03:
Hi guys, don't want to open another issue so I will ask here because my setup is exactly the same as @SebuZet.
https://shop.m5stack.com/products/atom-lite-esp32-development-kit
https://shop.m5stack.com/products/rs485-module

How long did you @SebuZet wait until unit appear in the logs?
I'm using 
```
uart:
tx_pin: GPIO26
rx_pin: GPIO32
baud_rate: 9600
parity: EVEN
```
Log:

[image]

and this is my config:

```
esphome:
  name: acpokoje
  friendly_name: AC Pokoje

esp32:
  board: m5stack-atom
# ... (other configurations)
uart:
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 9600
  parity: EVEN

# Enable logging
logger:
  level: VERBOSE
# Enable Home Assistant API
api:
  encryption:
    key: "uvtkTdu0U5UFsFlJN679qIgMLliqRW65oUHD25svvqg="

ota:
  password: "74a9147800cccb0134e2036e249afe0f"

wifi:
  manual_ip:
    static_ip: 192.168.1.141
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "AcPokoje Fallback Hotspot"
    password: "Hk0EbxG3nRHd"

captive_portal:
```
Can I improve or adjust something here to make it work? Or I need to wait for example 24h until it show in the log? :))

SebuZet, 2023-11-04:
No more than few minutes. I had a feeling it was an eternity ;). Update is called every 30s. Scanning should be done in few minutes. Make sure your cable are connected in correct way.
I'm now modifying this component and I noticed that every few software updates I have to reboot AC devices (power off and power on).

escees, 2023-11-04:
Thanks, now I've see in the logs that the AC Device was recognized after some tries with switching F1 with F2. Now it just need to be configured for HA :)

closed 2023-11-08 by lanwin