omerfaruk-aran/esphome_samsung_hvac_bus #23: How to get the data heat pump (AE160JNYDGH)

Mocky158, 2023-10-12:
Hello
I apologize in advance, I'm using Github for the first time for a question, so I don't know if this is the right way to write for advice.

I created esp according to see attachment. I also attached the settings I have to the file.

The problem is that I don't know how to know that something has been read or how to change it to read other data.

This is a heat pump (AE160JNYDGH)

I'm sorry if I wrote it incomprehensible, I'm using Google translator, I'm from the Czech Republic
[Tepelné čerpadlo.txt](https://github.com/lanwin/esphome_samsung_ac/files/12879743/Tepelne.cerpadlo.txt)

[image]

[image]

lanwin, 2023-10-12:
> I'm sorry if I wrote it incomprehensible, I'm using Google translator, I'm from the Czech Republic

Everything is fine fine.

logger:
  level: DEBUG

Just try to change your yaml file like this:

```yaml
uart:
  tx_pin: TX
  rx_pin: RX
  baud_rate: 9600
  parity: EVEN
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      timeout: 200ms
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ' ');
```

This should dumpy data witch is received into the log. This way you can see if there is any data.

lanwin, 2023-10-12:
There wireing looks fine so far. But please check you cable connections. I had a lot problems with wires that where not really connected when I started planing with it.

Mocky158, 2023-10-12:
I've already checked the cables twice, but I can do it again, but I'm not physically at the heat pump at the moment.
It looks the same on the website after the code change.
[image]

lanwin, 2023-10-12:
What do you mean by "I'm not physically at the heat pump"?

Mocky158, 2023-10-12:
I'm currently at work :D

Mocky158, 2023-10-13:
I checked the wiring and it was fine, so I ended up swapping the rx/tx connectors, even though I had them connected correctly according to the picture. Finally, I started to see these data, I don't know if it's good.
[logs_tepelne-cerpadlo_logs.txt](https://github.com/lanwin/esphome_samsung_ac/files/12889618/logs_tepelne-cerpadlo_logs.txt)

Mocky158, 2023-10-13:
So should I look for an address starting with 20? according to what is written in the note? Or what should I do with specific data?
[image]

lanwin, 2023-10-13:
Can you please switch the uart baud rate to 2400 and try again.

Mocky158, 2023-10-13:
I am sending the log here
[logs_tepelne-cerpadlo_run.txt](https://github.com/lanwin/esphome_samsung_ac/files/12897526/logs_tepelne-cerpadlo_run.txt)

Mocky158, 2023-10-14:
I saw that these errors pop up when I turn it on, is it OK?
[image]

lanwin, 2023-10-14:
Hmm. Can you please swap F1 and F2 again and try with 2400 and 9600 baud. On my system I got also data with F1 and F2 swapped. But the data is not valid. Your log looks like that. 

You can ignore the error for now.

Mocky158, 2023-10-15:
I'll get to the heat pump again tomorrow morning. So what should I test for RS485 speeds? 2400 and 9600? and I will try to transfer F1 to F2.

lanwin, 2023-10-15:
Just try both and see if you can get anything under "Discovered devices"

Mocky158, 2023-10-16:
I tried different combinations and got no result. I will try to order the M5Stack to rule out my device's fault.

Mocky158, 2023-10-17:
Today I ran out of the purchased M5STACK ATOM Lite which is listed in Plug and Play. I would like to ask to which connectors I should connect F1 and F2. They are pins 19 and 22 as in the M5STACK ATOM RS-485

Mocky158, 2023-10-18:
Could I ask for a sample of what I should see in the log? If I should also see the device (number) in the line Discovered devices: ?

lanwin, 2023-10-18:
Sorry I overlooked your edit. https://github.com/lanwin/esphome_samsung_ac/issues/23#issuecomment-1763699258 

That is exactly what we want to see. Your Indoor unit is 20.00.00. When you see that it works and you can control your devices when add a configuration for each unit to your yaml.

Mocky158, 2023-10-18:
You don't need to apologize for clarifying, it's still bothering me
I removed:
   debugging:
     direction: BOTH
     dummy_receiver: true
     after:
       timeout: 200ms
     sequences:
       - lambda: UARTDebug::log_hex(direction, bytes, ' ');

After that they started showing me temperatures etc. :-) I'm glad it works.
I am sending the data that I currently see.
I would like to ask, I am doing all this because I want when there is a surplus of energy from PV so that the utility water starts to heat up more, so I want it to be heated to 42C and I would like to raise the condition to 60C via the automation in NodeRed every time there is a battery etc. as an example.

But at the moment I can't see these data, how should I proceed to add them?
[image]

Mocky158, 2023-10-19:
Can I offer a remote connection to my ESP .Would that help to develop faster.

lanwin, 2023-10-19:
The log data is pretty confusion since its a lot.

I currently thinking about generating something like a table where every variable can appear with its last value. But I am not sure if that is possible with ESPHome.

lanwin, 2023-10-19:
> Can I offer a remote connection to my ESP .Would that help to develop faster.

Thank you. Maybe later. For now that would not help.

Mocky158, 2023-10-20:
So let me know as soon as I can help you with anything related to reading the data.For now I'll wait :-)

Mocky158, 2023-10-25:
I got these PDFs from Samsung support that show the registry numbers and what's under them. I don't know if it could help to identify which value should be changed to change the temperature of the service water.
[IM_MIM-B19N_CS.pdf](https://github.com/lanwin/esphome_samsung_ac/files/13165400/IM_MIM-B19N_CS.pdf)
[IM_ Modbus MIM-B19N_EN.pdf](https://github.com/lanwin/esphome_samsung_ac/files/13165402/IM_.Modbus.MIM-B19N_EN.pdf)

lanwin, 2023-11-19:
> I got these PDFs from Samsung support that show the registry numbers and what's under them. I don't know if it could help to identify which value should be changed to change the temperature of the [...]

Sorry I overlooked your answer. Sadly this did not help. The numbers really only count when you have that Modbus Interface.