omerfaruk-aran/esphome_samsung_hvac_bus #24: [Non-NASA] Most UART messages aren't decoded

deece, 2023-10-28:
I'm trying to get a Non-NASA system up and running, as I'm replacing a (now abandoned!) MIM-H02 Wifi interface.

It looks like it can't parse the incoming messages, except for one brief moment, immediately after power cycling the AC.

I've attached the logs & config. I am on commit c505fdd0fcb9863bc40a4f95c090c9cfc9431f6e

[logs_ac_rear_logs.txt](https://github.com/lanwin/esphome_samsung_ac/files/13195557/logs_ac_rear_logs.txt)
[ac_rear.yaml.txt](https://github.com/lanwin/esphome_samsung_ac/files/13195559/ac_rear.yaml.txt)

lanwin, 2023-11-06:
Hi deece, sorry for the late reply but I was on vacation. 

Did you try to switch F1/F2?

closed 2023-11-30 by lanwin

deece, 2023-11-30:
Ah, sorry, I didn't see y

deece, 2023-12-05:
That did it! I could have sworn I tried swapping the line, but clearly I was mistaken. Thanks for the help.