omerfaruk-aran/esphome_samsung_hvac_bus #28: Nasa protocol notes

sredfern, 2023-11-07:
Hi Lanwin, love your project and have started playing around with the data.

You've written down the bottom of the page "Hopefully I can provide a description of the NASA protocol here soon.". If possible I would love your unrefined notes on it. I'm trying to pick it up and c isn't my first programming language choice (js, python, sql).

Another question is where did you get all the message numbers from in [nasa.cpp](https://github.com/lanwin/esphome_samsung_ac/blob/main/components/samsung_ac/nasa.cpp) out of interest? It seems like you grabbed them from some xml file? Keen to hear the story here.

lanwin, 2023-11-08:
Hi @sredfern, yes my plan is to write the down. But unfortunately my time is pretty limited and so currently I focus on fixing and adding features for now. The best documentation for now is the nasa.cpp as you find yourself.

The messages numbers come from Samsungs SNET application. Its free to download and is build to monitor, update and error discovery for samsung devices. Its written in Microsofts .NET. .NET per default did not compiles in binary, instead it compiles to IL (which is more generic the binary). IL can be compiled back to .NET languages like C#. You can do that yourself with software like ILSpy. 

The NASA protocol itself is pretty generic (other than the non NASA protocol). Its more or less a generic data transfer protocol. So in large parts SNET itself is designed pretty generic for a wide range of heating and cooling devices. 

They designed the software that it uses something like a mapping file for each device type to be able to map the messages to the UI. This are the XML files. When you install SNET you can find one in "C:\Program Files (x86)\S-NET pro2\Projects\RuleScript\NASA.ptc".

Unfortunately not all messages are mapped in that files. And also some messages you can only find in the non generic parts of the SNET codebase and some messages are not defined anywhere.

betaphi, 2023-12-12:
I hope it is ok, when I post this here but I have created the [requested notes](https://wiki.myehs.eu/wiki/NASA_Protocol) at the newly started Samsung EHS Wiki at [wiki.myehs.eu](https://wiki.myehs.eu). There is not a lot of info on the site (yet) other than the said protocol description but I would be happy about any contributions that help us better understand these class of Samsung products.

Foxhill67, 2024-01-04:
> I hope it is ok, when I post this here but I have created the [requested notes](https://wiki.myehs.eu/wiki/NASA_Protocol) at the newly started Samsung EHS Wiki at [...]

I have created attached Excel sheet, based on the NASA.ptc file. Tried to fill some of the gaps with information I got from my SAMSUNG EHS Split heat pump. Feel free to add these details to the Wiki!
[NASA.xlsx](https://github.com/lanwin/esphome_samsung_ac/files/13833286/NASA.xlsx)

betaphi, 2024-01-04:
Thank you @Foxhill67 
That file is very helpful. There are also more variable informations available via the source code of the Samsung WiFi Kit available [here](https://opensource.samsung.com/uploadList?menuItem=home_appliances&classification1=airconditioners).

As I am very busy currently, having released EHSMonitor, a NASA->MQTT Monitor for the EHS Mono HT Quiet (and possibly other heat pump models as well) and as I'm working hard on a EHS control application, it would be super nice of you to add this info to the Wiki yourself.

Foxhill67, 2024-01-04:
No problem, not a Wiki expert, but I will probably manage to add the info myself ;-) 

I am also working on a solution to get the NASA messages via MQTT to HomeAssistant, fork from @lanwin ESPhome solution. Just ran into a small issue with the end byte accidentally also appearing in the data part, thus prematurely terminating the raw data processing. I will log a separate issue for that. Do you have a link to your EHSMonitor, may be you have tackled this issue already?

And thanks for the link to the WiFi Kit, looks promising. I'll see if it helps to fill even more gaps in the list of codes.

betaphi, 2024-01-04:
> No problem, not a Wiki expert, but I will probably manage to add the info myself ;-)

Thank you! :-)

> I am also working on a solution to get the NASA messages via MQTT to HomeAssistant, fork from @lanwin ESPhome solution. Just ran into a small issue with the end byte accidentally also appearing in [...]

Yes, the NASA decoding is working flawlessly in EHSMonitor. You can find it [here](https://github.com/betaphi/EHSMonitor). Please don't get discouraged by the fact it is written in Swift, as it is perfectly able to run under Linux.

> And thanks for the link to the WiFi Kit, looks promising. I'll see if it helps to fill even more gaps in the list of codes.

I'm glad I could help :-)

betaphi, 2024-01-05:
Thank you @Foxhill67
By the way: This is probably not the best way to communicate. You can contact me via the "eMail this User" button at https://wiki.myehs.eu/wiki/User:Betaphi
It would be great if you could also add the additional variable informations from the WiFi Kit source code...

lanwin, 2024-01-05:
Hey guy, thank you for your great work! Can you please keep me updated here when you found something useful? 

There are still a lot open questions. Like we found that WifiKit is adding an offset to the room temperature (like -2°) but it did not seems to be the same value for everyone. 

@betaphi and thanks for finding the WifiKit source. I did not know that its available somewhere. But it seems that its not as easy to read (from a first look). 
I will add a link to your protocol description to the readme soon.

Foxhill67, 2024-01-05:
> Thank you @Foxhill67 By the way: This is probably not the best way to communicate. You can contact me via the "eMail this User" button at https://wiki.myehs.eu/wiki/User:Betaphi It would be great if you could also add the additional variable informations from the WiFi Kit source code...

@betaphi : I agree to move our communication about the Wiki to the Wiki page itself, but I cannot find the 'eMail this user' button on the link you provided. Can you recheck? And: the WifiKit source is quite extensive, can you pinpoint the file(s) where I can find the additional variable information?

betaphi, 2024-01-06:
@Foxhill67 You probably have to be logged in in order to see the respective button. However the eMail is just betaphi@myehs.eu

DerTiger1, 2024-01-06:
> And: the WifiKit source is quite extensive, can you pinpoint the file(s) where I can find the additional variable information?
All NASA variables are defined in **NasaConst.java**

Foxhill67, 2024-01-06:
> > And: the WifiKit source is quite extensive, can you pinpoint the file(s) where I can find the additional variable information? <
> 
> All NASA variables are defined in **NasaConst.java**

Found it, thanks!

lanwin, 2024-02-01:
@Foxhill67 did you made some experience when to send data? I made some experiments since I noticed that it happens often that the message right after sending results in a checksum error. The error can be ignored since the next message will transport the right data but I was not able to prevent it.

I tested.
* Sending them when no data is available
* Directly inline
* Waiting 20-50 ms after the last message appear.

The SNET code did not seem to have anything special regarding that.

Lately I found that NonNASA commands only work reliable when they directly after a specific message.

Foxhill67, 2024-02-01:
no sorry, I only read messages, have commented out sending message part to avoid disturbing the proper functioning of my heating system. I am primarily interested in monitoring the performance of my heatpump system, have no use case yet to change any settings.

lanwin, 2024-02-01:
Ok no problem.

Did you found witch variables are for power consumption of the devices?

Foxhill67, 2024-02-01:
yes, for my heatpump variable 8414 indicates the total consumed power since installation. 8413 shows total actual consumption. 4427 is total produced energy, 4426 is actual produced energy

lanwin, 2024-02-02:
Ok I tested that. 8414 and 8413 work. But I dont get any data for 4427 and 4426.

Is 8413 in Wh?

Foxhill67, 2024-02-02:
8413 and 8414 are in W. I also see value for 8411 which is always bit less than 8413. Seems to be the net power consumption of the outdoor unit. 8413 looks like total power consumption of outdoor and indoor unit (so including water pump, electronics, etc).

lanwin, 2024-02-02:
@Foxhill67 did found anything interesting in the wifikit source?

Foxhill67, 2024-02-02:
Not really. It only helped to get a technical name for some of the messages. I have combined all info I have of the NASA message in a table here: https://wiki.myehs.eu/wiki/NASA_Protocol#

lanwin, 2024-03-11:
@Foxhill67 when I remember correctly do you have build your own parser right? 

https://github.com/lanwin/esphome_samsung_ac/issues/109

Did you also have invalid messages? Samsung SNET seem to ignore them. My current guess is that this happens cause multiple devices try to send at the same time.

Foxhill67, 2024-03-11:
Sorry, did not notice any invalid messages (but I might be overlooking them, I only create MQTT messages if I see valid data). Might also be because I only have 2 devices (indoor and outdoor unit) and do not send any messages myself, just monitoring

brazoayeye, 2024-07-15:
Hello, to simplify the analysis of sent data i made an excel file that summarize logs (collected with debug_log_messages: true)
The file have a table with all possible addresses and vlookup informations from the two tables copy-pasted from https://wiki.myehs.eu/wiki/NASA_Protocol.
When you import a txt file, columns A:C are updated. A is the sending address, B last recived value, C how many packets have been recived. Importing many txt adds data, to flush the file use the Reset button.

In the attached file you can find all read values I collected in an half hour. xlsb can't be uploaded so i zipped it.
[NASA_logAnalyzer_rev1.zip](https://github.com/user-attachments/files/16234780/NASA_logAnalyzer_rev1.zip)

closed 2024-10-29 by omerfaruk-aran

nicolaerosia, 2025-05-31:
@betaphi, This page is no longer accesible, https://wiki.myehs.eu/wiki/NASA_Protocol . Any chance you have a copy of it ?:( Thank you, it was  very helpful