omerfaruk-aran/esphome_samsung_hvac_bus #31: Platform esphome does not generate unique IDs error

khughes5, 2023-11-19:
### Discussed in https://github.com/lanwin/esphome_samsung_ac/discussions/30

<div type='discussions-op-text'>

<sup>Originally posted by **khughes5** November 19, 2023</sup>
Hi Guys, 

Again, sorry if this is another newb question/problem but i have been trying to resolve issues with my installation, and can't for the life of me figure it out.

Control of my AC works through the website instantly
[image]
however control via HA was really hit and miss for me.. for example if i use the climate control panel item with the system in an off state, change mode to "Cool", it thinks about it for a couple of seconds and then resets back to off. During the last 2 weeks i have probably had it work for a total of 1 day.. but that was fleeting..

I started checking logs and found this repetitive error message:

Platform esphome does not generate unique IDs. ID D4:D4:DA:9D:0D:8C-select-homeac_mode is already used by select.esp_samsung_ac_homeac_mode - ignoring select.esp-samsung_ac_homeac_mode

My ESP at the time was called esp_samsung_ac and my climate name was "My Home" with the sensors named HomeAc_temp, HomeAC_target, HomeAC_mode etc.

In the past, i HAVE renamed the ESP device (usually by creating a new device in ESPhome, editing the config then re-uploading via OTA), and these error messages seemed to indicate it remembered the old configuration.. I even tried removing all the old records from .storage\core.entity_registry.. as well as trying to rename the device and sensor names again..

To make things more insulting.. i gave up today and created a whole new HomeAssistant installation from scratch, installed ESP-Home, then created a new device, copied my old yaml contents in (using the newly generated encryption key), created a manual firmware.. then uploaded it through the OTA on the web interface... I also then had to adopt the device into ESP-Home in discovered devices.. Somehow, this still results in the error message:
[image]

now my ESP is called "ESP-samsungac", my climate called "SamsungAC" and sensors as "AC_temp", "AC_target" etc.. My Climate controls still do not function reliably.. i really don't know what else to do.. short of throwing away the M5 stack i have it installed on and trying everything from scratch again.. but this seems stupid :(

here is my config yaml:

```
esphome:
  name: esp-samsungac
  friendly_name: ESP-SamsungAC

substitutions:
  name: esp-samsungac

esp32:
  board: m5stack-atom
  framework:
    type: arduino

web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxx"

ota:
  password: "73571d78e2b33fbbee49a53066f0436b"
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5
  on_error:
    then:
      - logger.log:
          format: "OTA update errorr %d"
          args: ["x"]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Samsungac Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxx"

captive_portal:

uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN
  #debug:
  #  direction: BOTH
  #  dummy_receiver: true
  #  after:
  #      timeout: 200ms
  #  sequence:
  #    - lambda: UARTDebug::log_hex(direction, bytes, ' ');

external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: [ samsung_ac ]

samsung_ac:
  devices:
    - address: "20.00.00" # for non NASA it sould be like "c4"
      # Each property below are optional - choose what you want to use.

      # You could use the ESPHome/Homeassist climate control
      climate:
        name: "SamsungAC"

      # And/or use seperate controls/sensors
      room_temperature:
        name: AC_temp
      target_temperature:
        name: AC_target
      power:
        name: AC_power
      mode:
        name: AC_mode
        
      # Only supported on NASA devices
      room_humidity:
        name: AC_humidity    

# DEVICE INFO
text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${name} IP address"
      icon: mdi:ip-network

sensor:
  - platform: wifi_signal
    name: "${name} signal"
    update_interval: 360s
  
  - platform: uptime
    name: "${name} uptime"
    update_interval: 360s

button:
  # Restart the ESP
  - platform: restart
    name: "Restart ${name}"
```</div>

khughes5, 2023-11-19:
Done and Done!, let me know if there is any more testing or troubleshooting I can help out with!

Btw, The 4D:4D:4A bit is the MAC address of the m5stack-atom.

khughes5, 2023-11-22:
yesterday, i tried commenting out the mode part of my yaml
`      # mode:
      #  name: AC_mode`
and so far, i am yet to get an error in the logs, and control via climate in the dash appears to be way more responsive than i have seen to date.

lanwin, 2023-11-30:
@khughes5 can you please retry. I made an update the should prevent that.

mbo18, 2023-12-05:
I have the same issue:

>Logger: homeassistant.components.select
Source: helpers/entity_platform.py:610
Integration: Select (documentation, issues)
First occurred: 09:52:10 (7 occurrences)
Last logged: 12:45:34
Platform esphome does not generate unique IDs. ID D4:D4:DA:84:F6:E8-select-mode is already used by select.salon_hvac_mode - ignoring select.salon-hvac_mode

lanwin, 2023-12-06:
Can you please reinstall and try again. It can be possible that you need to restart Homeassist before that error goes away.

mbo18, 2023-12-09:
No more issue here! Thank you

khughes5, 2023-12-09:
Brilliant! All working for me now too, really appreciate the support, time and effort you put into this!

closed 2023-12-09 by khughes5