omerfaruk-aran/esphome_samsung_hvac_bus #33: [COMMON] M5STACK Example as Default

bzumik1, 2023-11-27:
Hi @lanwin,

I'm reaching out to suggest making the M5STACK board the default example setup. Given the popularity of this board among the community, it seems like a logical choice, especially since it's labeled as Plug & Play. This adjustment could enhance the experience for beginners, aligning with the board's user-friendly nature. We could then include guidance on modifying the example for various boards in the DIY section.

Thanks for considering this enhancement!

bzumik1, 2023-12-04:
Hi @lanwin,

I'm considering adding this configuration to the Plug & Play YAML to simplify the process for those who are new to ESPHome or haven't worked with it before. This way, they should be able to easily connect it to Home Assistant. I believe Home Assistant now supports discovery over Bluetooth, so even if they don't modify the WiFi name and password, Home Assistant should be able to locate the device. Please review the configuration, and let me know if it looks good to you.

```yaml
# Device name
substitutions:
  name: esp-samsung-ac

# BASIC CONFIGURATION
esphome:
  name: ${name}

# Select board in this case M5STACK
esp32:
  board: m5stack-atom

# Create web interface where you can controll the AC from your web browser
web_server:
  port: 80

# Enable logging
logger:

# Enable native API for communication with Home Assistant, this can be removed if you do not plan to use it with Home Assistant
api:

# Enable OTA firmware update, you will be able to upload the firmware over WiFi without connecting the device by cables
ota:
  password: !secret esp_home_ota_password

# Setup your WiFi password
wifi:
  ssid: "My WiFi Name" #!!!CHANGE ME!!!
  password: "My WiFi Password" #!!!CHANGE ME!!!

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${name}.o"

captive_portal:

# SAMSUNG AC CONFIGURATION

# Specify pins used by the board to comunicate with AC
uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN

# Import custom component from GitHub
external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: [samsung_ac]

# Configuration of AC component
samsung_ac:
  devices:
    # AC UNIT 1 - add first indoor unit
    - address: "20.00.00" # unit address, for non-NASA devices the format will be like "c4", the address can be found in logs on start
    
    # Each property below is optional (climate, room_temperature etc.) - you can delete those which are not needed
      # Creates climate in Home Assistant
      climate:
        name: "AC Unit 1"

      # Create separate controls/sensors, this is not necessary for the ones already included in climate above,
      # feel free to remove them if not needed
      room_temperature:
        name: "AC Unit 1 - room temperature"
      # Only supported on NASA devices for now
      room_humidity:
        name: "AC Unit 1 - humidity"
      target_temperature:
        name: "AC Unit 1 - target temperature"
      power:
        name: "AC Unit 1 - power switch"
      mode:
        name: "AC Unit 1 - mode"
    

    # AC UNIT 2 - add second indoor unit, remove this section if you have only one unit, add similar section if you have multiple units
    - address: "20.00.01" # unit address, for non-NASA devices the format will be like "c4", the address can be found in logs on start
    
    # Each property below is optional (climate, room_temperature etc.) - you can delete those which are not needed
      # Creates climate in HA
      climate:
        name: "AC Unit 2"

      # Create separate controls/sensors, this is not necessary for the ones already included in climate above,
      # feel free to remove them if not needed
      room_temperature:
        name: "AC Unit 2 - room temperature"
      # Only supported on NASA devices for now
      room_humidity:
        name: "AC Unit 2 - humidity"
      target_temperature:
        name: "AC Unit 2 - target temperature"
      power:
        name: "AC Unit 2 - power switch"
      mode:
        name: "AC Unit 2 - mode"

# DEVICE INFO

# Show IP address in Home Assistant
text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${name} IP address"
      icon: mdi:ip-network

sensor:
  # Shows WiFi signal strength in Home Assistant
  - platform: wifi_signal
    name: "${name} signal"
    update_interval: 10s
  
  # Shows run time in Home Assistant
  - platform: uptime
    name: "${name} uptime"
```

lanwin, 2023-12-05:
ok

bzumik1, 2023-12-05:
Thx, I will create the PR, I will keep the DIY example simple because there I expect more experience users to use it.

bzumik1, 2023-12-05:
PR created https://github.com/lanwin/esphome_samsung_ac/pull/38

lanwin, 2024-01-19:
Example was updated to M5STACK

closed 2024-01-19 by lanwin