omerfaruk-aran/esphome_samsung_hvac_bus #35: Temperature differs from the reported indoor temperature

mbo18, 2023-12-04:
Hi!

I have a Samsung Windfree AR12TXCAAWKNEU with is working fine with your code but I have an issue with the temperature which differs from the one reported by my HVAC: 22.5°C using this code, 20°C using Samsung APP or the smartthings integration in HA.

From ESPHome log:

>[23:04:57][W][samsung_nasa:479]: s:20.00.00 d:b0.00.ff VAR_in_temp_room_f 22.500000
[23:04:57][D][sensor:094]: 'Temperature': Sending state 22.50000 °C with 1 decimals of accuracy
[23:04:57][D][climate:380]: 'Salon' - Sending state:
[23:04:57][D][climate:383]:   Mode: HEAT
[23:04:57][D][climate:388]:   Fan Mode: AUTO
[23:04:57][D][climate:403]:   Current Temperature: 22.50°C
[23:04:57][D][climate:409]:   Target Temperature: 19.00°C
[23:05:07][W][samsung_nasa:500]: s:20.00.00 d:b3.00.ff ENUM_in_operation_power on
[23:05:07][D][climate:380]: 'Salon' - Sending state:
[23:05:07][D][climate:383]:   Mode: HEAT
[23:05:07][D][climate:388]:   Fan Mode: AUTO
[23:05:07][D][climate:403]:   Current Temperature: 22.50°C
[23:05:07][D][climate:409]:   Target Temperature: 19.00°C
[23:05:07][W][samsung_nasa:506]: s:20.00.00 d:b3.00.ff ENUM_in_operation_mode 4
[23:05:07][D][select:015]: 'Mode': Sending state Heat (index 4)
[23:05:07][D][climate:380]: 'Salon' - Sending state:
[23:05:07][D][climate:383]:   Mode: HEAT
[23:05:07][D][climate:388]:   Fan Mode: AUTO
[23:05:07][D][climate:403]:   Current Temperature: 22.50°C
[23:05:08][D][climate:409]:   Target Temperature: 19.00°C
[23:05:08][W][samsung_nasa:494]: s:20.00.00 d:b3.00.ff ENUM_in_state_humidity_percent 44
[23:05:08][D][sensor:094]: 'Humidity': Sending state 44.00000 % with 0 decimals of accuracy

lanwin, 2023-12-05:
Hi mbo18 thanks for reporting this. 

Can you please redeploy your version and add this option to your yaml config:

```
samsung_ac:
  debug_log_messages_raw: true
```

This will print the raw messages to console. 

Please let that run 15 seconds and the copy and paste the logs from that 15 seconds into here. 
Please also paste the current temp from ESPHome and from Samsung APP within that time range.

This could help to solve that problem.

bzumik1, 2023-12-05:
Hi, I am also experiencing this issue, for me the difference was not so big though. Will also check it from my side.

mbo18, 2023-12-05:
Please find attached log and temp as requested.

Please note that now that I have enable debug messages, I can't disable it. I've tried removing the line from the conf and set it to false without success.

[image]

[logs_salon-hvac_run.txt](https://github.com/lanwin/esphome_samsung_ac/files/13557659/logs_salon-hvac_run.txt)

lanwin, 2023-12-05:
Can you please check if the Smartthings value inst simply the target temperature? The only 19° value I see in your logs is from the target temperature variable.

mbo18, 2023-12-05:
I don't think so. Target temperature is set to 19 and is fixed. Also, the HVAC displays sometimes 20°, sometimes 19°

From the smartthings integration:
[image]

Temperature measurement is equal to currently from the climate entity.

lanwin, 2023-12-05:
Could you please take a log when temp is 20° or change the target temp in example to 18°?

mbo18, 2023-12-05:
See log attached.

I've changed the target temp to 18 at [12:48:36] and back to 19 at [12:48:57]
[logs_salon-hvac_logs.txt](https://github.com/lanwin/esphome_samsung_ac/files/13559625/logs_salon-hvac_logs.txt)

mbo18, 2023-12-05:
And now with target temp set to 20° and indoor temp reported as 20°C by Smartthings
[logs_salon-hvac_logs(1).txt](https://github.com/lanwin/esphome_samsung_ac/files/13559666/logs_salon-hvac_logs.1.txt)

mbo18, 2023-12-05:
Re-did it, target temp is set to 20, inside temp is reported as 21 by Samsung
[image]

[logs_salon-hvac_logs(4).txt](https://github.com/lanwin/esphome_samsung_ac/files/13560088/logs_salon-hvac_logs.4.txt)

lanwin, 2023-12-06:
Hmm I am not sure where that 2° gap came from. I could imaging that this is some sort of correction since most AC's are attached near the ceiling where its more hot then on the ground. 

@bzumik1 can you please check what the difference is for you? 

@mbo18 did the target temperature match? Did Smartthings have the same value als ESPHome and if you set the target temp in Smarttings did ESPHome also shows the same value?

bzumik1, 2023-12-06:
Hi @lanwin will check it today or tomorrow.

mbo18, 2023-12-06:
Hi lanwin,

Yes the target temperature match. I have the same value from ESPHome, Smartthings and the device's screen. It all updates in a second when changed.

I think that I will create a template sensor in ESPHome with a calibrate_linear filter.

bzumik1, 2023-12-06:
Hi @lanwin, the difference is 0,5°C on one indoor unit and 1,7°C on another (both mounted at same height). I am trying to get the data to Grafana and compare the temperature which I am getting from smarthings and the one from this. The target temperature is same on both.

bzumik1, 2023-12-06:
> Hi lanwin,
> 
> Yes the target temperature match. I have the same value from ESPHome, Smartthings and the device's screen. It all updates in a second when changed.
> 
> I think that I will create a template sensor in ESPHome with a calibrate_linear filter.

For now this seems like reasonable solution, but we should allow the calibration in this component it self. It is quite strange though because per my understanding these data would be shown on wall panel so do not understand why there is the difference. Can't we be reading different values?

mbo18, 2023-12-08:
I think that Samsung is simulating the indoor temperature. The sensor is on the device at 2m+ but this does not represent the real room temperature.
The temperature we get seems to be very precise but the temperature from samsung is vague (adapted to the display and not precise because not the real temperature but an approximation). I may be wrong :)
[image]

mbo18, 2023-12-10:
So I’m unable to do something on the esphome side as the current temperature has no id and I don’t know how to get the temperature. More over it would be best to being able to change it for the climate component. Maybe add a way to modify the temperature using lambda?

For anyone interested, the formula is: 
indoor temp = current temperature|int - 2

lanwin, 2023-12-12:
@bzumik1 has you done more comparison with your data?

bzumik1, 2023-12-12:
Hi, sorry didn't have time for that. I have it in my called though.

mbo18, 2024-01-10:
I've added 2 more m5stack and now my 3 indoor units are locally connected!

Regarding temperature, I have slightly the same result. Sometimes its -2 or -3°. I did not find the real formula used by samsung.

[image]

lanwin, 2024-01-10:
Do you have 3 different outdoor devices with an indoor device each? Or one outdoor device and 3 indoor device?

lanwin, 2024-02-19:
The configuration now has room_temperature_offset to correct the reported value. Please look into the example.yaml

closed 2024-02-19 by lanwin