omerfaruk-aran/esphome_samsung_hvac_bus #40: RecieveEnable\DataEnable output for MAX485 compatibility

failMode, 2023-12-10:
First up, great work! I appreciate the time you've spent putting this together.

I've been working to get this operational on a Wemos D1 Mini and an MAX485 TTL to RS485 breakout board. The hurdle I've encountered is that the MAX485 chip requires a logic signal to it's DE and RE inputs to enable receiving or sending. Luckily, the inputs are inverted, so you can tie them together electrically and just use simple 'Low to receive, High on send'. If you leave the inputs floating the MAX485 just gets hot and doesn't do anything.

I have spent some time going through the code but I honestly can't get my head around how Samsung_AC::send_bus_message works or how we would get some settings from the YAML to nominate an output pin...

It would be amazing if we could set a MAX485 duplex control GPIO in the YAML that gets pulled low on start and is toggled high when sending. Feel free to tell me to git gud and do it myself. xD

Example Arduino code that I've tested with the MAX485:
  //DE,RE Controling pin of RS-485
  pinMode(7, OUTPUT);
  // DE,RE = HIGH, RS485 Transmit Enabled
  digitalWrite(7,HIGH);
   //DE,RE = 0, RS485 Receive Enabled
  digitalWrite(7,LOW);

lanwin, 2023-12-12:
Hi @failMode isnt there a way to use the esphome configuration to trigger that?

failMode, 2023-12-12:
Thank you for the suggestion @lanwin, I hadn't thought of that. I've done some fiddling:
First thought was to try to commandeer the UART debug function, but the earliest it will execute a sequence is _after_ the first byte is sent (which is obviously too late). ChatGPT suggests using a UART on_data: function but I tested that and doesn't exist so the AI is hallucinating. 

In short, I'm all ears if you can suggest another way to do it! :D A flow control pin on this component though, like on the esphome modbus component would be awesome.

lanwin, 2024-01-02:
Can you please first try to get help on ESPHomes discord for that? I am sure that this should be possible without building it into this project.

lanwin, 2024-01-19:
Did you got any answers there?

brazoayeye, 2024-02-19:
Bumping that topic. I'm going to use a ST485EBDR already present in my device, I have two pins connected to the same GPIO:

If GPIO is low: reciver is enabled, transmitter is disabled (high resistance for the bus)
If GPI is hig: reciver is disabled, transmitter is enabled (low resistance)

When I programmed the device using esp-idf (no esperience with esp-home) i simply set =1 the TransmitEnable pin, than i revert to 0 after the command has been sent.

Now, trying to setup uart zone i can't find a settings for this behaviour.

@lanwin is the uart settings standard for espHome so i must go there to have infos?

failMode, 2024-02-19:
@lanwin I'm so sorry, I didn't see your second reply. I did end up on the home assistant discord and spent a lot of time working through using the modbus component and the uart debug flag but didn't get good results. I did end up grabbing the suggested hardware of an M5 Stack and the RS485 hat which works fantastically - basically any RS485 breakout board with auto flow control will do the job, it's just that the 'cheaper' UART->RS485 breakouts do rely on an extra input (high or low on TX) to handle the flow control. I'll take some time to try to code that feature with a pull request unless someone else beats me to it... Thank you for such a great, clean project!

closed 2024-07-08 by lanwin