omerfaruk-aran/esphome_samsung_hvac_bus #41: Feature request: add quiet mode

mbo18, 2023-12-11:
Hi!

Is it possible to add "Quiet" mode?

Debug log attached:
- At 16:55:36 (fan_mode_real 13) = quiet mode activated (from auto)
- At 16:55:47/48 (fan_mode_real 11) = back to auto

Thank you

[quiet mode.txt](https://github.com/lanwin/esphome_samsung_ac/files/13638372/quiet.mode.txt)

lanwin, 2023-12-12:
I think it will be possible. Can you tell me with indoor device you have? I will try if its possible to detect of the device have a quiet mode or not.

But don't expect a fast change. My time is pretty limited.

mbo18, 2023-12-12:
I know what limited time is, no pressure :)
The quiet mode is the only feature I miss in order to not use the smartthings, that’s why I opened this FR. 

I have 3 Samsung HVAC Wind-Free Elite R32. Models are : AR09TXCAAWKNEU (2.5kW) and AR12TXCAAWKNEU (3.5kW). There is an ESP with this code only on the last one currently but I have ordered two m5stack atom for the others :)

PR #80 (2024-02-07): WindFree and other presets
Added support for setting different presets via `NASA_ALTERNATIVE_MODE`.

On my AR09TXFCAWKNEU (3 indoor units), I have three, Quiet (2), Fast (3) and WindFree (9). In #3 another one, Long Reach (6) was mentioned, so there could be more in the wild.
Additional config to enable/disable different modes might be required in the future.

I went with the presets/custom_presets approach, but on the other hand, all three that I tested with are mutually exclusive with fan modes, so they could also be considered as custom_fan_modes instead.

Fixes #3 
Fixes #41
closed 2024-02-07 by lanwin