omerfaruk-aran/esphome_samsung_hvac_bus #45: Error since latest commits

mbo18, 2024-01-03:
Hi,

Just to let you know that since latest commit I have these errors in ESPHome logs:
```
[11:47:02][D][esp-idf:000]: E (173547) esp-tls: couldn't get hostname for :: getaddrinfo() returns 202, addrinfo=0x0

[11:47:02][D][esp-idf:000]: E (173553) TRANSPORT_BASE: Failed to open a new connection: 32769

[11:47:02][D][esp-idf:000]: E (173569) MQTT_CLIENT: Error transport connect
```

I use an ESP32 (m5stack atom lite)

lanwin, 2024-01-03:
Thank you for that report. Did you configured debug_mqtt support?

mbo18, 2024-01-03:
No, this is my Yaml:
```yaml
esphome:
  name: bureau-hvac
  friendly_name: Bureau HVAC
  comment: "Climatisation de gauche"

esp32:
  board: m5stack-atom
  framework:
    type: arduino #esp-idf

uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: [samsung_ac]
    refresh: 0s

samsung_ac:
  debug_log_messages_raw: false
  debug_log_messages: false
  devices:
    - address: "20.00.00"
      climate:
        name: None
      room_temperature:
        name: "Temperature"
      room_humidity:
        name: "Humidity"

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
    - ssid: !secret wifi_ssid_live
      password: !secret wifi_password_live
  ap:
    password: !secret api_ota_pwd

captive_portal:

web_server:
  port: 80
  auth:
    username: admin
    password: !secret api_ota_pwd

logger:

api:
  encryption:
    key: !secret esphome_enc_key

ota:
  password: !secret api_ota_pwd
```

lanwin, 2024-01-04:
Can you please retry. The error should not be gone.

closed 2024-01-04 by lanwin

mbo18, 2024-01-04:
No more issues! Thank you

commit 67744f9f (2024-01-04): Fix #45: Dont try to use mqtt if its not configured