omerfaruk-aran/esphome_samsung_hvac_bus #46: Does not compile with ESP-IDF framework

mbo18, 2024-01-03:
Hi,

Just to let you know that this component does not compile with the ESP-IDF framework on ESP32. 
Here are all errors in logs:

```
Compiling .pioenvs/bureau-hvac/src/esphome/components/samsung_ac/non_nasa.o
In file included from src/esphome/components/samsung_ac/nasa.cpp:3:
src/esphome/components/samsung_ac/nasa.cpp: In function 'void esphome::samsung_ac::process_nasa_message(std::vector<unsigned char>, esphome::samsung_ac::MessageTarget*)':
src/esphome/components/samsung_ac/nasa.cpp:499:35: error: format '%d' expects argument of type 'int', but argument 7 has type 'long int' [-Werror=format=]
                     ESP_LOGW(TAG, "s:%s d:%s ENUM_in_state_humidity_percent %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_w'
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:499:21: note: in expansion of macro 'ESP_LOGW'
                     ESP_LOGW(TAG, "s:%s d:%s ENUM_in_state_humidity_percent %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                     ^~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:511:35: error: format '%d' expects argument of type 'int', but argument 7 has type 'long int' [-Werror=format=]
                     ESP_LOGW(TAG, "s:%s d:%s ENUM_in_operation_mode %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_w'
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:511:21: note: in expansion of macro 'ESP_LOGW'
                     ESP_LOGW(TAG, "s:%s d:%s ENUM_in_operation_mode %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                     ^~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:517:35: error: format '%d' expects argument of type 'int', but argument 7 has type 'long int' [-Werror=format=]
                     ESP_LOGW(TAG, "s:%s d:%s ENUM_in_fan_mode_real %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_w'
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:517:21: note: in expansion of macro 'ESP_LOGW'
                     ESP_LOGW(TAG, "s:%s d:%s ENUM_in_fan_mode_real %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                     ^~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:531:39: error: format '%d' expects argument of type 'int', but argument 7 has type 'long int' [-Werror=format=]
                         ESP_LOGW(TAG, "s:%s d:%s ENUM_IN_OPERATION_VENT_POWER %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_w'
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:531:25: note: in expansion of macro 'ESP_LOGW'
                         ESP_LOGW(TAG, "s:%s d:%s ENUM_IN_OPERATION_VENT_POWER %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                         ^~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:536:39: error: format '%d' expects argument of type 'int', but argument 7 has type 'long int' [-Werror=format=]
                         ESP_LOGW(TAG, "s:%s d:%s ENUM_IN_OPERATION_VENT_MODE %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_w'
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:536:25: note: in expansion of macro 'ESP_LOGW'
                         ESP_LOGW(TAG, "s:%s d:%s ENUM_IN_OPERATION_VENT_MODE %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                         ^~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:541:39: error: format '%d' expects argument of type 'int', but argument 7 has type 'long int' [-Werror=format=]
[... 207 lines cut ...]
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_w'
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:734:25: note: in expansion of macro 'ESP_LOGW'
                         ESP_LOGW(TAG, "s:%s d:%s VAR_OUT_PRODUCT_OPTION_CAPA %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                         ^~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:747:39: error: format '%d' expects argument of type 'int', but argument 7 has type 'long int' [-Werror=format=]
                         ESP_LOGW(TAG, "s:%s d:%s VAR_OUT_PHASE_CURRENT %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_w'
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/samsung_ac/nasa.cpp:747:25: note: in expansion of macro 'ESP_LOGW'
                         ESP_LOGW(TAG, "s:%s d:%s VAR_OUT_PHASE_CURRENT %d", packet_.sa.to_string().c_str(), packet_.da.to_string().c_str(), message.value);
                         ^~~~~~~~
Compiling .pioenvs/bureau-hvac/src/esphome/components/samsung_ac/protocol.o
cc1plus: some warnings being treated as errors
*** [.pioenvs/bureau-hvac/src/esphome/components/samsung_ac/nasa.o] Error 1
========================= [FAILED] Took 37.82 seconds =========================
```

mbo18, 2024-01-03:
ESP-IDF is the new framework from Espressif for ESP32. See ESPHome doc here: https://esphome.io/components/esp32#esp32-espidf-framework

By default, the current framework is 'arduino':

```yaml
esp32:
  board: m5stack-atom
  framework:
    type: arduino
```
but you can change it with esp-idf (which will become the default somewhere in the future):
```yaml
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
```

lanwin, 2024-01-04:
```
See also "/config/.esphome/build/esp32-test-bug/.pioenvs/esp32-test-bug/CMakeFiles/CMakeOutput.log".
See also "/config/.esphome/build/esp32-test-bug/.pioenvs/esp32-test-bug/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /config/.esphome/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /config/.esphome/build/esp32-test-bug/.pioenvs/esp32-test-bug/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/config/.esphome/platformio/packages/tool-ninja/ninja cmTC_8a11e && [1/2] Building C object CMakeFiles/cmTC_8a11e.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_8a11e
    FAILED: cmTC_8a11e 
    : && /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_8a11e.dir/testCCompiler.c.obj  -o cmTC_8a11e   && :
    /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/project.cmake:298 (__project)
  CMakeLists.txt:3 (project)
```

Can you please check that again? I get a different error when comping.

mbo18, 2024-01-04:
Did you clean the build files before changing to esp-idf and click install ?

mbo18, 2024-01-04:
Strange, redid all and still the same log here: [logs_bureau-hvac_compile.txt](https://github.com/lanwin/esphome_samsung_ac/files/13832312/logs_bureau-hvac_compile.txt)

Yaml is:
```
esphome:
  name: bureau-hvac
  friendly_name: Bureau HVAC
  comment: "Climatisation de gauche"

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: [samsung_ac]
    refresh: 0s

samsung_ac:
  debug_log_messages_raw: false
  debug_log_messages: false
  devices:
    - address: "20.00.00"
      climate:
        name: None
      room_temperature:
        name: "Temperature"
      room_humidity:
        name: "Humidity"

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
    - ssid: !secret wifi_ssid_live
      password: !secret wifi_password_live
  ap:
    password: !secret api_ota_pwd

captive_portal:

web_server:
  port: 80
  auth:
    username: admin
    password: !secret api_ota_pwd

logger:

api:
  encryption:
    key: !secret esphome_enc_key

ota:
  password: !secret api_ota_pwd
```

lanwin, 2024-01-04:
Witch ESPHome version you are on? I am on 2023.12.5

mbo18, 2024-01-04:
Same (first line in the log). I’m using HAOS 11.2 with HA 2023.12.3 and the stable release of the ESPHome addon (2023.12.5). 
Maybe something on your side with dependencies?

mbo18, 2024-01-04:
I just found this: "try to delete folder /home/xxxx/.platformio/packages/toolchain-xtensa32 and run your project again."
from https://community.platformio.org/t/cmake-error-while-building/15692/9

lanwin, 2024-01-05:
I think I stumbled across this problem https://github.com/esphome/issues/issues/3076 the suggestions is to switch over to type: arduino ^^

mbo18, 2024-01-05:
Ah… not fun. Seems to be an issue on RPi. I’m running HAOS on a Nuc.

closed 2024-01-09 by mbo18