omerfaruk-aran/esphome_samsung_hvac_bus #47: Raw message reading terminated prematurely when 0x34 (end byte) is part of data

Foxhill67, 2024-01-04:
Hi @lanwin,

thanks again for your work, it has helped me a lot to connect my Samsung EHS heat pump to Home Assistant.

I did accidentally stumble upon this issue: in void Samsung_AC::loop() in samsung_ac.cpp, the raw input data is read until the end byte 0x34 is found. That works fine most of the time, but fails if 0x34 is part of the data and not yet the end byte! Example:

[10:30:48][W][samsung_ac:018]: RAW: 32002f200000b300ffc014bb06440f0000000044230003ae4044240000796044260000000044270034

In this example:
- value for code 4423 = 0003ae40
- value for code 4424 = 00007960
- value for code 4426 is 00000000
-  value for code 4427 is 0034.... but the algorithm thinks it found the end byte and terminates further processing

Can we think of a clever trick to solve this?

lanwin, 2024-01-05:
Thanks for reporting this! And thank you that you attached example data. That is a good start to find a better way.

The current solution is a pretty basic approach. 

A far better approach would it be to fill the buffer and check if we found a valid packed until we found one or the buffer has reached its max size.

Foxhill67, 2024-01-05:
Tried this in 'my' fork in the second part of void Samsung_AC::loop(), and it seems to work fine (declared bytes_ and size_ as uint16_t in samsung_ac.h):
      last_transmission_ = now;
      while (available())
      {
        uint8_t c;
        read_byte(&c);
        if (c == 0x32 && !receiving_) // start-byte found
        {
          receiving_ = true;
          bytes_ = 0;
          size_ = 0;
          data_.clear();
        }
        if (receiving_)
        {
          data_.push_back(c);
          bytes_++;
          switch (bytes_)
          {
            case 1: // start byte found
              break;
            case 2: // first part of size found
            {
              size_ = c;
              break;
            }
            case 3: // second part of size found
            {
              size_ = (int)size_ << 8 | c;
              ESP_LOGV(TAG, "Message size in packet: %d", size_);
              break;  
            }
            default: // subsequent bytes 
            {
              if (bytes_ >= (size_+2)) // end byte found
              {  
                receiving_ = false;
                process_message(data_, this);
              }
              break;
            }
          }
        }
      }

lanwin, 2024-01-19:
The problem with that solution is that it only works for the NASA protocol. I will try to find a more general solution next week.

lanwin, 2024-01-23:
I now have commited a solution to that. May you have a look?

Foxhill67, 2024-01-23:
Hi, checked for NASA protocol on my SAMSUNG heatpump and all seems to work fine, data is read correctly. So regression test wise: looks good. However, I don't think that my current data stream includes the 0x34 end byte as part of the data, so we have no 100% proof that the original issue is definitively solved. And I do not know how to manipulate the data stream to create a test case. So what shall we do with this issue, close it for now?

closed 2024-01-24 by lanwin