omerfaruk-aran/esphome_samsung_hvac_bus #52: hello, I can't do anything

titof2375, 2024-01-20:
hello, here is my code for esphome
`esphome:
  name: clim-samsung
  friendly_name: clim samsung 
  
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "C7eUhABV32R9GpyS5osEiUr/XXGKlzGfzYq+rvd0rrM="

ota:
  password: "79eabcbd632cc456e17f12ea195055a6"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Clim-Samsung Fallback Hotspot"
    password: "vIFWNhqHOBU7"

captive_portal:

web_server:

uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN
`
and here is what I get from the air conditioning
`[08:00:32][C][wifi:573]: WiFi:
[08:00:32][C][wifi:405]:   Local MAC: 64:B7:08:B7:A2:20
[08:00:32][C][wifi:410]:   SSID: [redacted]
[08:00:32][C][wifi:411]:   IP Address: 192.168.1.110
[08:00:32][C][wifi:413]:   BSSID: [redacted]
[08:00:32][C][wifi:414]:   Hostname: 'clim-samsung'
[08:00:32][C][wifi:416]:   Signal strength: -38 dB ▂▄▆█
[08:00:32][C][wifi:420]:   Channel: 6
[08:00:32][C][wifi:421]:   Subnet: 255.255.255.0
[08:00:32][C][wifi:422]:   Gateway: 192.168.1.1
[08:00:32][C][wifi:423]:   DNS1: 192.168.1.14
[08:00:32][C][wifi:424]:   DNS2: 192.168.1.1
[08:00:32][C][logger:439]: Logger:
[08:00:32][C][logger:440]:   Level: DEBUG
[08:00:32][C][logger:441]:   Log Baud Rate: 115200
[08:00:32][C][logger:443]:   Hardware UART: UART0
[08:00:32][C][uart.idf:139]: UART Bus 1:
[08:00:32][C][uart.idf:140]:   TX Pin: GPIO19
[08:00:32][C][uart.idf:141]:   RX Pin: GPIO22
[08:00:32][C][uart.idf:143]:   RX Buffer Size: 256
[08:00:32][C][uart.idf:145]:   Baud Rate: 9600 baud
[08:00:32][C][uart.idf:146]:   Data Bits: 8
[08:00:32][C][uart.idf:147]:   Parity: EVEN
[08:00:32][C][uart.idf:148]:   Stop bits: 1
[08:00:32][C][captive_portal:088]: Captive Portal:
[08:00:32][C][web_server:168]: Web Server:
[08:00:32][C][web_server:169]:   Address: clim-samsung.local:80
[08:00:32][C][mdns:115]: mDNS:
[08:00:32][C][mdns:116]:   Hostname: clim-samsung
[08:00:32][C][ota:097]: Over-The-Air Updates:
[08:00:32][C][ota:098]:   Address: clim-samsung.local:3232
[08:00:32][C][ota:101]:   Using Password.
[08:00:32][C][api:139]: API Server:
[08:00:32][C][api:140]:   Address: clim-samsung.local:6053
[08:00:32][C][api:142]:   Using noise encryption: YES
[08:00:33][D][api:102]: Accepted 192.168.1.106
[08:00:33][W][component:214]: Component api took a long time for an operation (0.06 s).
[08:00:33][W][component:215]: Components should block for at most 20-30ms.
[08:00:33][D][api.connection:1121]: Home Assistant 2024.1.4 (192.168.1.106): Connected successfully
`

titof2375, 2024-01-20:
and it stays like that

bzumik1, 2024-01-20:
> hello, here is my code for esphome
>
> `esphome:
>
>   name: clim-samsung
>
>   friendly_name: clim samsung
>
>
>
> esp32:
>
>   board: m5stack-atom
>
>   framework:
>
>     type: esp-idf
>
>
>
>
>
> # Enable logging [...]

Hello @titof2375, it’s crucial to prioritize security by avoiding the sharing of OTA and encryption keys publicly. I recommend updating them and securely storing them in the secret.yaml file, similar to how you’ve handled your wifi password. This enhances the overall security of your project.

titof2375, 2024-01-20:
hello @bzumik1, I agree with you regarding security, what is displayed on my previous message are not the real codes for my network, we were modifying it before publication.

lanwin, 2024-01-20:
@titof2375 as far as I see everything is fine. You have a basic ESP device running without any feature configured yet. So without additional features you can not do anything. That is true.

Maybe now you should look into our example.yaml and see whats missing.

titof2375, 2024-01-20:
so how to find device addresses?

lanwin, 2024-01-20:
Please have a look into the readme and example.yaml everything you need to
know should in there.

titof2375, 2024-01-21:
thank you, it's good I found it

closed 2024-01-21 by titof2375