omerfaruk-aran/esphome_samsung_hvac_bus #53: Lag when sending commands to AC

lanwin, 2024-01-20:
When any values is changed, it takes a few seconds until the AC executes it. In this time it can happen that the AC sends messages with the old values to the UI witch causes it to switch back to the old states before the AC executes it and the new values get visible. 

We should change that so the UI skips messages for changed values until the new values appears (or a timeout happens).

omerfaruk-aran, 2024-08-26:
> When any values is changed, it takes a few seconds until the AC executes it. In this time it can happen that the AC sends messages with the old values to the UI witch causes it to switch back to the [...]

Hi Lanwin, I found a solution related to this issue and have been testing it for the past two days. I plan to finalize it and submit a Pull Request today. I wanted to check with you first—do you have any updates on your end?

lanwin, 2024-08-26:
No. But I think there was a compensation in NonNASA if I remind me correctly. My wish would be that the solution is not that complicated :D

omerfaruk-aran, 2024-08-26:
No, it's not too complicated. I wrote a simple loop to check the incoming and outgoing data to see if there are any changes. So far, my tests have shown no issues, but I'll be testing more throughout the day. I had submitted another PR earlier; once that gets approved, I'll submit this one as well.

lanwin, 2024-08-26:
My plan to solve this was to within the components to ignore incoming values unti the dest value was reached or until a timeout occured.

omerfaruk-aran, 2024-08-26:
> Bu sorunu çözmek için planım, hedef değere ulaşılıncaya veya zaman aşımı oluşana kadar bileşenler içinde gelen değerleri yok saymaktı.

I seem to have solved the problem with the code below, but I need to test it in detail.

`for (const auto &pair : devices_)
      {
        optional<Mode> current_value = pair.second->_cur_mode;
        std::string address = pair.second->address;
        unsigned long now = millis();

        if (current_value.has_value() && last_values[address] != mode_to_string(current_value.value()))
        {
          // If the value has changed, save the new value and start the process
          last_values[address] = mode_to_string(current_value.value());
          last_update_time[address] = now;

          // Here you can reflect the new value to the UI
          ESP_LOGI(TAG, "Value changed for device: %s", address.c_str());
        }
        else
        {
          // If the value is the same, skip this step
          ESP_LOGD(TAG, "No change in value for device: %s", address.c_str());

          // timeout control
          if (now - last_update_time[address] > TIMEOUT_PERIOD)
          {
            ESP_LOGW(TAG, "Timeout for device: %s", address.c_str());
            last_values[address] = "default_value"; // Örnek varsayılan değer
            last_update_time[address] = now;
          }
          continue;
        }
      }`

omerfaruk-aran, 2024-08-26:
Hi Lanwin,

If you could check and approve my pending PR, I'll merge this new adjustment and create a new PR. It worked flawlessly on my end; I used to have to click the power button 2-3 times occasionally, but now it works perfectly. Similarly, there were issues when switching between modes where the command wouldn’t be received or wouldn’t notify me after being received—this adjustment seems to have fixed that issue as well. I’m eagerly waiting for your response :)

PR #169 (2024-10-09): Handle UI Update Issues - Improved Value Change Handling in Samsung AC Integration
**Description:** This PR introduces enhancements to the Samsung AC integration, specifically addressing the UI update issues related to value changes. The core improvement lies in handling mode changes and ensuring that the interface accurately reflects the current state of the devices.

**Key Changes:**

- **Value Change Detection:** A mechanism has been implemented to detect changes in the device modes. If a change is detected, it triggers an update while also keeping track of pending changes to ensure consist [... cut]
closed 2024-10-09 by lanwin