omerfaruk-aran/esphome_samsung_hvac_bus #54: temperature sensor

titof2375, 2024-01-21:
Hello, can we tell the air conditioning to regulate the temperature in relation to another temperature sensor than the one in the air conditioning?

DerTiger1, 2024-01-21:
@titof2375: Yes it's possible. You can use any homeassistant Sensors for temperature and the thermostat to regulate the temperature as an add-on in esphome. Here is my working config with home assistant sensor reading "sensor.meter_og_temperature":
(Sorry of the formatting of the comments all '#' lead to bold)

`substitutions:
  esp32_name: "esphome-klimaanlage"
  device_description: "M5-Atom Stack mit RS485 Interface"

esphome:
  name: ${esp32_name}
  comment: ${device_description}

external_components:
  - source: github://lanwin/esphome_samsung_ac@main

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: recommended

uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 9600
  parity: EVEN

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: !secret ha_key
  reboot_timeout: 60s
  
ota:
  password: !secret ota_key
  safe_mode: true
  num_attempts: 1

wifi:
  ssid: !secret iot_ssid
  password: !secret iot_password
  domain: .iot.local
  reboot_timeout: 5min
  ap:
    ssid: "Klimaanlage Fallback Hotspot"
    password: !secret wifi_ap_password

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

sensor:
  - platform: homeassistant
    id: TemperaturOG
    accuracy_decimals: 1
    entity_id: sensor.meter_og_temperature

  - platform: homeassistant
    id: SollwertOG2
    accuracy_decimals: 1
    entity_id: input_number.heizung_sollwert_og2

# Home Assistant switches usw.
switch:
  - platform: safe_mode
    internal: true
    name: use_safe_mode

button:
  - platform: restart
    name: Restart $esp32_name

samsung_ac:
  debug_log_messages: false # logs all messages - only for debugging
  debug_log_messages_raw: false # logs the raw message bytes from all messages - only for debugging

  devices:
    - address: "20.00.00" # for non NASA it sould be like "c4"
      # Each property below are optional - choose what you want to use.

      # You could use the ESPHome/Homeassist climate control
      climate:
        name: "ESP32 mit RS485 Verbindung zur Klimaanlage"
        id: klima
        visual:
          min_temperature: 16 °C
          max_temperature: 26 °C
          temperature_step: 0.1 °C
      # And/or use seperate controls/sensors
      room_temperature:
        name: Klima_Ist_Temperatur
      target_temperature:
        name: Klima_Soll_Temperatur
      power:
        name: Klima_power
      mode:
        name: Klima_mode

      # Only supported on NASA devices
      room_humidity:
        name: Klima_Luftfeuchtigkeit

climate: #https://esphome.io/components/climate/climate_ir.html?highlight=ir
# Example climate controller with presets
# https://esphome.io/components/climate/thermostat.html
  - platform: thermostat
    visual:
      min_temperature: 16 °C
      max_temperature: 26 °C
      temperature_step: 0.1 °C
    name: "Thermostat Regler auf ESP32 für Klimaanlage"
    sensor: TemperaturOG
    min_heating_off_time: 600s
    min_heating_run_time: 1h
    min_cooling_off_time: 600s
    min_cooling_run_time: 1h
    min_idle_time: 180s
    heat_deadband: 0.2 °C
    heat_action:
      - climate.control:
          id: klima
          mode: HEAT
          fan_mode: AUTO
          swing_mode: 'Off'
          target_temperature: 26°C
          #target_temperature: !lambda |-
            #return id(SollwertOG2).state;
            #ESP_LOGI("main", "Value of my sensor: %f", id(SollwertOG2).state);
    cool_action:
      - climate.control:
          id: klima
          mode: COOL
          fan_mode: AUTO
          swing_mode: 'Off'
          target_temperature: 20°C 
    idle_action:
      - climate.control:
          id: klima
          mode: 'OFF'
          target_temperature: 16°C`

titof2375, 2024-01-21:
[image]
Are these the sensors that you declare for the Samsung AC air conditioning driver?

DerTiger1, 2024-01-21:
Yes, the first one is a temperature sensor, the second one is meant to be the setpoint value, but not used yet.

DerTiger1, 2024-01-21:
not directly - the thermostat takes the temperature and regulates the samsung hvac. (connected by the id: klima) At least the behavior is the same.