omerfaruk-aran/esphome_samsung_hvac_bus #58: esphome doesn't compile anymore

DerTiger1, 2024-01-24:
@lanwin: Yesterday's changes result in the following error message:
(maybe you should fork the development in dev and stable)

`INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/esphome-klimaanlage.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-klimaanlage (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/esphome-klimaanlage/src/esphome/components/samsung_ac/protocol.o
Compiling .pioenvs/esphome-klimaanlage/src/esphome/components/samsung_ac/protocol_non_nasa.o
src/esphome/components/samsung_ac/protocol.cpp: In function 'esphome::samsung_ac::DataResult esphome::samsung_ac::process_data(std::vector<unsigned char>&, esphome::samsung_ac::MessageTarget*)':
src/esphome/components/samsung_ac/protocol.cpp:42:24: error: enumeration value 'Ok' not handled in switch [-Werror=switch]
                 switch (result)
                        ^
src/esphome/components/samsung_ac/protocol.cpp:42:24: error: enumeration value 'SizeDidNotMatch' not handled in switch [-Werror=switch]
src/esphome/components/samsung_ac/protocol.cpp:42:24: error: enumeration value 'UnexpectedSize' not handled in switch [-Werror=switch]
cc1plus: some warnings being treated as errors
*** [.pioenvs/esphome-klimaanlage/src/esphome/components/samsung_ac/protocol.o] Error 1`

DerTiger1, 2024-01-24:
btw: I'm trying to add a serial to wifi bridge ([https://github.com/oxan/esphome-stream-server](url)) with a virtual com Port ([https://www.hw-group.com/software/hw-vsp3-virtual-serial-port](url)) to my M5Stack with esphome, to enable "sniffing" all Samsung commands directly.

lanwin, 2024-01-25:
Hi @DerTiger1 thanks for reporting this. This is cause you have configured framework: espidf with seems to use a little bit different C++ compiler settings then the default Arduino framework. 

Sadly espidf did not compile on my Raspberry (cause the framework is broken there) , so I can not reproduce it.

But I had made a change witch should fix that case. Please try again.

lanwin, 2024-01-25:
> btw: I'm trying to add a serial to wifi bridge ([https://github.com/oxan/esphome-stream-server](url)) with a virtual com Port ([https://www.hw-group.com/software/hw-vsp3-virtual-serial-port](url)) to my M5Stack with esphome, to enable "sniffing" all Samsung commands directly.

Oh that sounds amazing. I already thought about something like that but did not yet looked for that. I think the problem should be that this will not work together with esphome_samsung_ac since both consume the data coming from Serial. 

But please try it, it can be that I am wrong.

Maybe if it does not work I could integrate something like this directly into the codebase.

mbo18, 2024-01-25:
> Sadly espidf did not compile on my Raspberry (cause the framework is broken there) , so I can not reproduce it.

Too bad because with the idf framework we can use the Bluetooth proxy. Hope someone will solve this someday

DerTiger1, 2024-01-25:
It‘s allready fixed - I‘m able to compile also with bluetooth proxy enabled.

closed 2024-01-26 by lanwin