omerfaruk-aran/esphome_samsung_hvac_bus #60: Power Switch only works sometimes

GammaScorpii, 2024-01-27:
So I managed to get this working but at first I thought I hadn't because I wasn't able to switch the unit on. But everything else works flawlessly, reading house temp, setting target temp, changing the mode. Sometimes the power switch works but other times it sends the signal but it seems like it gets reset a second later. Here's a log:

`[17:41:04][W][samsung_ac:018]: update
[17:41:04][C][samsung_ac:039]: Configured devices: 00
[17:41:04][C][samsung_ac:059]: Discovered devices:
[17:41:04][C][samsung_ac:060]:   Outdoor: -
[17:41:04][C][samsung_ac:061]:   Indoor:  00
[17:41:06][D][switch:012]: 'Power' Turning ON.
[17:41:06][D][switch:055]: 'Power': Sending state ON
[17:41:06][W][samsung_ac:098]: write 32d000b01f041401f4210000bb34
[17:41:08][D][number:012]: 'Target temperature': Sending state 20.000000
[17:41:08][D][sensor:094]: 'House temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[17:41:08][D][switch:055]: 'Power': Sending state OFF
[17:41:08][D][select:015]: 'Mode': Sending state Cool (index 1)
[17:41:15][D][number:012]: 'Target temperature': Sending state 20.000000
[17:41:15][D][sensor:094]: 'House temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[17:41:15][D][select:015]: 'Mode': Sending state Cool (index 1)
[17:41:20][D][switch:012]: 'Power' Turning ON.
[17:41:20][D][switch:055]: 'Power': Sending state ON
[17:41:20][W][samsung_ac:098]: write 32d000b01f041401f4210000bb34
[17:41:21][D][number:012]: 'Target temperature': Sending state 20.000000
[17:41:21][D][sensor:094]: 'House temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[17:41:21][D][select:015]: 'Mode': Sending state Cool (index 1)
[17:41:28][D][number:012]: 'Target temperature': Sending state 20.000000
[17:41:28][D][sensor:094]: 'House temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[17:41:28][D][select:015]: 'Mode': Sending state Cool (index 1)`

[17:41:06][D][switch:055]: 'Power': Sending state ON <--- This was me turning it on with Home Assistant.
[17:41:06][W][samsung_ac:098]: write 32d000b01f041401f4210000bb34
[17:41:08][D][number:012]: 'Target temperature': Sending state 20.000000
[17:41:08][D][sensor:094]: 'House temperature': Sending state 22.00000 °C with 1 decimals of accuracy
[17:41:08][D][switch:055]: 'Power': Sending state OFF <--- This wasn't me

But then later you see I try again and this time it stays on. What's going on here?

GammaScorpii, 2024-01-27:
Not sure if it's related but the first couple of times I tried it I got this in the log:

[18:22:14][W][samsung_ac:065]: Current message exceeds the size limits.

lanwin, 2024-01-27:
Hi @GammaScorpii thanks for your report. 

Yes the NonNASA requests have some problems. I did not figured out what exactly is the problem, but I am on it.

Please follow also https://github.com/lanwin/esphome_samsung_ac/issues/59#issuecomment-1913125651 and https://github.com/lanwin/esphome_samsung_ac/issues/1#issuecomment-1913138208 both are regarding similar problems.

Can you please pull again. The "Current message exceeds the size limits." should now be fixed and the outdoor unit should be listed again.

lanwin, 2024-02-01:
Can you please pull and check if the current version still have this problem?

lanwin, 2024-02-02:
Ok then I close it