omerfaruk-aran/esphome_samsung_hvac_bus #64: How to: Use esphome as RS485 to Wifi bridge

DerTiger1, 2024-01-29:
It is possible to bridge the serial connection over Wifi to a virtual COM-Port and to use the Samsung S-Net software for remote controlling your HVAC Unit:

1.  Add [https://github.com/oxan/esphome-stream-server](url) to your esphome yaml:
    external_components:
      - source: github://oxan/esphome-stream-server
    stream_server:
2. Compile and upload
3. Install a virtual COM-port driver (e.g.: [https://www.hw-group.com/software/hw-vsp3-virtual-serial-port](url))
4. Start the virtual COM-Port and make sure that "NVT enabled" is unchecked under settings.
5. Connect with the Samsung S-Net Software using the option F1/F2 (to change COM-Port you have to click on the "hearbeat" icon in the upper left)

Enjoy

bzumik1, 2024-01-29:
@DerTiger1 that is great to know, thanks a lot!

lanwin, 2024-01-30:
Do you see a way to log the data that is written thru the virtual COM port?