omerfaruk-aran/esphome_samsung_hvac_bus #69: Non-Nasa F3 F4 serial bus

Foggy2, 2024-02-02:
Hello @lanwin 

First of all thanks for putting this together and sharing it on GitHub. It is exactly what I have been looking for.

I am wondering if you have considered adding in support for the non-nasa F3 F4 (wall controller <-> indoor unit) serial bus? I ask as it is much easier for me (I assume others too) to connect to the cabling behind the wall controller than to connect to the outdoor unit or the indoor unit in the roof space.

I realise that this wouldn't be a minor change to your codebase. To this end I have been experimenting with adapting your existing code to function based off of the specifications shared by [DannyDeGaspari](https://github.com/DannyDeGaspari/Samsung-HVAC-buscontrol). I have reached a point now where I am able to reliably send commands and have them accepted.

Temperature and On/Off works. I have a ducted system with a 3rd party ventilation system so am unable to test for fan speed/blade positions.

Mode does not appear to be working but I note that other non-nasa users are reporting the same on the F1F2 serial bus. So this is a work in progress.

I would submit a PR myself but this is my first time using c++ and I only understand how all the code fits together at a basic level. I fear that I would not be able to produce something of good enough quality myself so instead I wish to point out my repo [here](https://github.com/Foggy2/esphome_samsung_ac/tree/F3F4-bus-conversion) where you can see the changes that I have made to reach this point.

I am more than happy to keep working on this on my own repo for myself but in the event that you are happy to explore the F3 F4 serial bus further I would be happy to help where I can.

Thanks!

nicolaerosia, 2024-02-02:
I'm also interested in F3/F4, however, on my AE260RNWSEG the F3/F4 has power, and I think I need a special adapter. The wall controller is getting powered from these lines

lanwin, 2024-02-02:
I am not sure about that. What I currently know is, it seems that the outdoor unit uses different commands then the remote control. 
So I have to think about how I can handle this difference.

In the mean time the best way you can help is to implement the packets that are required for that. Please look at NonNasaDataPacket how to do that. NonNasaCommandC6 is a good example on how to start. It would be good to get a pullrequest per command.

And it would help to get example packets for each command. Something like 3200c8250000000003000000ee34 . You can get them when you enable debug_log_messages_raw https://github.com/lanwin/esphome_samsung_ac?tab=readme-ov-file#development

Foggy2, 2024-02-03:
Hi @lanwin 
I understand. As I learn more and time permits I will work on sending the required commands in PRs.
Thanks!

lanwin, 2024-02-05:
They did not need to be perfect :D C++ is also not my preferred language.

dzungpv, 2024-09-21:
@Foggy2 what hardware you are using to access F3F4 data, it is not normal RS-485?

Foggy2, 2024-09-22:
@dzungpv it is normal RS-485 like the F1 F2 communications bus.

I have checked my notes and F3 connects to pin A, and F4 connects to pin B.

dzungpv, 2024-09-22:
> @dzungpv it is normal RS-485 like the F1 F2 communications bus.
> 
> 
> 
> I have checked my notes and F3 connects to pin A, and F4 connects to pin B.

I ask because some wiki say that F3F4 has power and must using other chip. Likely 10-12v between them.

Foggy2, 2024-09-22:
> I ask because some wiki say that F3F4 has power and must using other chip. Likely 10-12v between them.

I understand. In my instance, I placed the microcontroller in the wall cavity behind the wired wall controller and connected to the wall controller pins.

My wall controller has 4x pins F3, F4, 12v+ &  12v-. To the best of my understanding the F3 F4 pins are standard RS-485 and I used a [M5Stack RS-485 to TTL converter](https://core-electronics.com.au/m5stack-rs485-to-ttl-converter-unit.html) without any issues.

However, I have also seen that some controllers only have the two pins. I do not know how those work so I am unable to assist in that area.

closed 2024-10-29 by omerfaruk-aran