omerfaruk-aran/esphome_samsung_hvac_bus #76: [Non-NASA] Blade Movement (AJ050NCJ2EG)

TCB13, 2024-02-06:
> open an Issue for the blade modes (...) and describe what blade modes do you have. It would also help I you could attach the messages set (0-2s) directly after you changed the blade mode. At its best for each mode.

Here's my split remote for reference. It only has a blade button (middle one) that will make them go up and down continuously or stop at any given point.

[image]

Each time I turn OFF and ON the split the blade position gets reset.

Attached you may find the log for "Fan Only" for split `01`. At first the blades were 50% open (seems to be the default whenever I turn ON the unit) logs `wind_direction:31`. After pressing the remote button blades started moving up and down, logs `wind_direction:26`. Finished the test by pressing again the button and then proceeded to turn OFF the split.

[FAN_MODE-BLADE-ON-OFF.txt](https://github.com/lanwin/esphome_samsung_ac/files/14178930/FAN_MODE-BLADE-ON-OFF.txt)

Tried the same in "Heat Mode" and the log shows the exact same `wind_direction:31` and `wind_direction:26`.

[HEAT_MODE-BLADE-ON-OFF.txt](https://github.com/lanwin/esphome_samsung_ac/files/14178931/HEAT_MODE-BLADE-ON-OFF.txt)

Thank you.

Goaheadz, 2024-02-07:
@TCB13 
I dont have any "yellow" MSG or RAW encoded messages in my log files like you, and my AC behaves differently, since after power on, it recalls the blade position/angle from previous usage.

lanwin, 2024-02-07:
> @TCB13 I dont have any "yellow" MSG or RAW encoded messages in my log files like you, and my AC behaves differently, since after power on, it recalls the blade position/angle from previous usage.

To enable message logging please read here https://github.com/lanwin/esphome_samsung_ac?tab=readme-ov-file#development

Goaheadz, 2024-02-07:
When I get home, I will try to get some logs of the blade messages.

Also seeing some extra information:
Frequency, capacity, current, voltage, power.
Maybe all fields will be fulfilled when splits power on, and this can be used as sensors as well to gather more information.
[image]

matthias882, 2024-02-07:
> @lanwin , side note, are you planning on exposing the inverter data? It would be nice to have it in HA!

Its the f3-command from outdoor unit. I got it to work yesterday, for now only as internal variable and in debug log. No idea how to get it to the esphome-api :-D. The Power-Value seems not to be extremely accurate, but it should be nice to log the consumed Power in home automation.

TCB13, 2024-02-07:
> The Power-Value seems not to be extremely accurate, but it should be nice to log the consumed Power in home automation.

This was what I wanted to read from it indeed. Maybe @lanwin that is more versed in the ESPHome part can eventually take care of it.

matthias882, 2024-02-07:
There are a bunch of more nice values like the outdoor temperature, temperatures of the condenser unit, the fan-status, the valves or the EEVs but i don't know how interesting these values are for somebody

TCB13, 2024-02-07:
> There are a bunch of more nice values like the outdoor temperature, temperatures of the condenser unit, the fan-status, the valves or the EEVs but i don't know how interesting these values are for somebody

Well if you can expose them so @lanwin can then display them in ESPHome then people can decide what they want to see or not on HA.

Goaheadz, 2024-02-08:
Outdoor Temp would be a nice to have in HA: (ou_outdoor_temp[°C])
[W][samsung_ac:493]: MSG: {src:c8;dst:04;cmd:f0;commandF0:{ou_freeze_protection:0; ou_heating_overload:0; ou_defrost_control:0; ou_discharge_protection:0; ou_current_control:0; inverter_order_frequency[Hz]:0; inverter_target_frequency[Hz]:0; inverter_current_frequency[Hz]:0; ou_bldc_fan:0; ou_error_code:00}}
[W][samsung_ac:493]: MSG: {src:c8;dst:01;cmd:c0;commandC0:{ou_operation_mode:88; ou_4way_valve:0; ou_hot_gas_bypass:0; ou_compressor:0; ou_ac_fan:0; ou_outdoor_temp[°C]:12; ou_discharge_temp[°C]:11; ou_condenser_mid_temp[°C]:12}}
[W][samsung_ac:493]: MSG: {src:c8;dst:03;cmd:f1;commandF1:{Electronic Expansion Valves: EEV_A:480; EEV_B:480; EEV_C:480; EEV_D:480}}

3 splitters:
[W][samsung_ac:493]: MSG: {src:02;dst:c8;cmd:20;command20:{target_temp:23; room_temp:17; pipe_in:18; pipe_out:18; power:0; wind_direction:31; fanspeed:0; mode:01}}
[W][samsung_ac:493]: MSG: {src:03;dst:c8;cmd:20;command20:{target_temp:23; room_temp:17; pipe_in:17; pipe_out:17; power:0; wind_direction:31; fanspeed:0; mode:01}}
[W][samsung_ac:493]: MSG: {src:04;dst:c8;cmd:20;command20:{target_temp:22; room_temp:17; pipe_in:17; pipe_out:17; power:0; wind_direction:31; fanspeed:0; mode:01}}

All of them show wind_direction:31 and no changes to that value even, when I press direction from remote.
(or at lest I was not able to notice)

TCB13, 2024-02-08:
> wind_direction:31 and no changes to that value, even when I press direction from remote.

Are you sure you're looking at the `src:` of the correct split? Note that all splits are constantly saying how they're configured even when turned OFF.

In my system (looking at the right `src`) when the blades aren't moving I get `wind_direction:31`. After pressing the remote button blades start moving up and down and it logs `wind_direction:26`.

Goaheadz, 2024-02-08:
> In my system (looking at the right `src`) when the blades aren't moving I get `wind_direction:31`. After pressing the remote button blades start moving up and down and it logs `wind_direction:26`.

I will give it another try today.

Do you have any configuration to write log to text file or is just plain copy-paste from log window to "notepad"?

TCB13, 2024-02-08:
> I will give it another try today.
>
> Do you have any configuration to write log to text file or is just plain copy-paste from log window to "notepad"?

The log window on ESPHome has a "download log" button :)

Goaheadz, 2024-02-08:
> The log window on ESPHome has a "download log" button :)

Dammit ...I feel so stupid!! :)

lanwin, 2024-02-08:
Sorry Non-NASA Blade movement is not yet impl. @atanasenko addid it yesterday for NASA.

Goaheadz, 2024-02-08:
@lanwin I confirm @TCB13 report for non-nasa vertical blade commands.

**Blade moving vertical is (send move command) being reported as "wind_direction:26"**
[19:35:41][W][samsung_ac:493]: MSG: {src:02;dst:c8;cmd:20;command20:{target_temp:23; room_temp:20; pipe_in:44; pipe_out:44; power:1; wind_direction:26; fanspeed:2; mode:01}}
**Blade in stopped position (send stop command)  is being reported as "wind_direction:31"**
[19:35:51][W][samsung_ac:493]: MSG: {src:02;dst:c8;cmd:20;command20:{target_temp:23; room_temp:20; pipe_in:45; pipe_out:45; power:1; wind_direction:31; fanspeed:2; mode:01}}

atanasenko, 2024-02-08:
Yes, Non-NASA has placeholders in the code for alt modes and swing modes. Somebody who can test on a real non-nasa unit will have to implement the glue code.
https://github.com/lanwin/esphome_samsung_ac/blob/main/components/samsung_ac/protocol_non_nasa.cpp

atanasenko, 2024-02-08:
I'll also have a look at adding the harness for configuring supported modes in config file.

TCB13, 2024-10-29:
Is this really completed?

omerfaruk-aran, 2024-10-29:
Hello @TCB13,

As the new maintainer of this project, I’m reviewing past issues and discussions. It looks like this particular topic hasn’t had any updates since February. If this issue persists or if you have additional insights to share, please feel free to open a new issue or discussion using the updated templates we've provided. This will help us better track and address your concerns more efficiently.

Regarding the responses from @atanasenko, it seems there were plans to implement the code placeholders for alt modes and swing modes for Non-NASA devices, as well as adjustments to configure supported modes in the configuration file. Could you please confirm if any progress was made on this, or if additional support or testing is needed from the community?

Thank you for your continued engagement! Let’s work together to improve the project.

Best regards,
Ömer Faruk ARAN

TCB13, 2024-10-29:
@omerfaruk-aran this isn't fixed. Please re-open this issue. There's important information attached related to the commands and logs.