omerfaruk-aran/esphome_samsung_hvac_bus #77: [Non-NASA] Test Beep - Not Important (AJ050NCJ2EG)

TCB13, 2024-02-06:
I've noticed that my remote has the ability to "beep" the unit for testing:

[image]

This isn't an important feature but could be an interesting way to have a test button on ESPHome to quickly identify indoor units. Attaches you may find the log produced by unit `01` while making the beep.

[BEEP.txt](https://github.com/lanwin/esphome_samsung_ac/files/14179140/BEEP.txt)

lanwin, 2024-02-06:
@matthias882 did you have an idea how to find witch message triggers the beep?

TCB13, 2024-02-06:
> ist there something else on the bus than 2 indoor and 1 outdoor units?

No. I assume `CC` is the outdoor unit, but I might be wrong.

matthias882, 2024-02-06:
the outdoor-unit should be the c8
in my system (2xIU and 1xOU) only the c8 talks to the inoor units

TCB13, 2024-02-06:
> there is a fourth device on the bus

I only have the outdoor unit and two indoor units. What about the ESP itself? How does it speak?

matthias882, 2024-02-06:
the esp currently talks as D0 and nobody asked him before ;) D0 should normaly be the SNet PCSoftware.

I think there is some other device in the outdoor unit, c8 is the mainboard and cc some additional thing connected to the mainboard by the bus.

But back to he topic. There is the possibility hat the beep commands don't go over the bus. The infrared-remote says to he indoor-unit that it has to beep. There is no need for the IU to tell it to someone else. Its a thing only between the remote and the IU I think.

closed 2024-10-29 by omerfaruk-aran