omerfaruk-aran/esphome_samsung_hvac_bus #87: [Non-NASA] Unable to control AC since 7 Feb update

Rebem13, 2024-02-08:
I am running an M5StackMini unit on my non-NASA indoor AC (showing as 00 and outdoor as c8) and have been controlling it perfectly for a couple of weeks. Since the code update yesterday (Feb 7th) I seem to have lost control of the AC. I can see the command to turn on/off in the logs like this:
[08:41:09][D][switch:016]: 'Aircon_power' Turning OFF.
[08:41:09][D][switch:055]: 'Aircon_power': Sending state OFF

...but it isn't reflected in the state of the device. I see no errors in the debug log. Where should I look next please?

TCB13, 2024-02-08:
I'm currently running the commit 9b87eadd378717072423c969fe387096efe6a678 from 3 days ago and that's the latest version that I'm sure works fine.

@lanwin maybe you can add a tag to that commit so people can [pull it directly](https://esphome.io/components/external_components.html) in their ESPHome yaml with the `ref` option? Unfortunately doesn't seem to allow us to specific commits by hash. Thanks.

lanwin, 2024-02-08:
@Rebem13 can you please turn on message logging https://github.com/lanwin/esphome_samsung_ac?tab=readme-ov-file#development and check if you see any message with cmd:f8 within the log. And can you check if you see a log line starting with write when you try to control the AC? 

We currently wait for that that f8 message and sending the control message after that, cause that seems to work more reliable for that other NonNASA users.

lanwin, 2024-02-08:
@TCB13 the problem is more that I dont know what an important commit is yet.

Rebem13, 2024-02-08:
> external_components:
>   - source: 
>      type: git
>      url: github://lanwin/esphome_samsung_ac
>      ref: 74
>     components: 
>       - samsung_ac

Thanks, I will give this a go after getting the info for @lanwin but when I tried it quickly, I got all sorts of compile errors:
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^

Rebem13, 2024-02-08:
> @Rebem13 can you please turn on message logging https://github.com/lanwin/esphome_samsung_ac?tab=readme-ov-file#development and check if you see any message with cmd:f8 within the log. And can you [...]

I am seeing a few cmd:f8 messages in the log but not many. Here is an example:
[17:09:59][W][samsung_ac:482]: MSG: {src:c8;dst:00;cmd:f8;raw:016230e60900780ffb}
[17:09:59][W][samsung_ac:032]: RAW: 3200c8402000000060670043ec34

Absolutely no messages with the word write in them despite sending the command to turn off the aircon and change from cool to fan (it's a heatwave here in Perth ~39 degrees celcius currently!!).

I feel like the controls in HA are reflecting the status of the unit because when I flick the switch to turn it off, the switch goes back to on again.

lanwin, 2024-02-08:
You only need `debug_log_messages` line from the Development section. The mqtt stuff is only interesting if you like to develop on the NASA protocol.

Can you check if you see this line `MSG: {src:c8;dst:f0;cmd:f8;}`

Rebem13, 2024-02-08:
> MSG: {src:c8;dst:f0;cmd:f8;}

No, I only see them like this:
MSG: {src:c8;dst:00;cmd:f8;raw:025850f73100e3140b}

Wouldn't the dst be 00, not f0? My indoor unit is 00, outdoor is c8.

lanwin, 2024-02-08:
Do you see any `cmd:c6` ? 

Can you please publish a full log? 

@matthias882 do you have an idea why he did not see that message?

lanwin, 2024-02-08:
Can you try the current main if it works now again?

Rebem13, 2024-02-08:
No change - let me just check one thing though : to try the current main, I just use the install option through ESPHome? So basically recompile?

I did note this in the simple logs:
[18:48:50][W][samsung_ac:171]: NonNASA: invalid crc - got 67 but should be 174: 3200c828004334

TCB13, 2024-02-08:
> I just use the install option through ESPHome? So basically recompile?
> 
> I did note this in the si

Don't forget to add the `refresh: 0s` so it will pull the latest code instead of compiling from your cache:

```
external_components:
  - source: github://lanwin/esphome_samsung_ac@main
    components: 
      - samsung_ac
    refresh: 0s
```

Rebem13, 2024-02-08:
Thanks for the info @TCB13 - that compiled properly.

@lanwin - all fixed!!! What changed?

matthias882, 2024-02-08:
> No, I only see them like this: MSG: {src:c8;dst:00;cmd:f8;raw:025850f73100e3140b}
> 
> Wouldn't the dst be 00, not f0? My indoor unit is 00, outdoor is c8.

The message with cmd f8 to destinaion f0 is only the trigger to send the commands to the bus, because in the currently known systems there is a huge space in the bus communication. The messages with cmd f8 to dest. 00 are answered by the indoor-unit and thatswhy there is no gap for sending our control-commands.

matthias882, 2024-02-08:
I compiled the new version and it works also on my system. Our messages are are send randomly after the c6 or the f8 command. (dont know why there is now a c6 on my system, it wasn't there yeterday). But it works fine.

matthias882, 2024-02-08:
we should find out who ist the destination d0 and f0. If one o these is on the bus, he will answer the messages and our comunication ist affected.
What could this be? What else is connected to the F1/F2? A cable remote? External sensors?

Rebem13, 2024-02-08:
On my aircon system I have 3 units - outdoor, indoor (in ceiling) and the remote panel. Outdoor and Indoor are connected via F1/F2 and the remote is on F3/F4. I can confirm there's nothing else in the system - I've checked the cables physically.

BTW, I even have fan control now! It's all working really well - thank you everyone!

matthias882, 2024-02-08:
> On my aircon system I have 3 units - outdoor, indoor (in ceiling) and the remote panel. Outdoor and Indoor are connected via F1/F2 and the remote is on F3/F4. I can confirm there's nothing else in [...]

could you please try to send a complete log of the communication debug messages? I guess the remote-panes also has message on the f1/f2 bus...

Rebem13, 2024-02-09:
Logs here. I started the logging, changed the mode to fan (which turned the unit on) and then adjusted the temp all so there were a few writes in there:

INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/samsung-aircon.yaml...
INFO Updating https://github.com/lanwin/esphome_samsung_ac.git@main
INFO Starting log output from 10.1.1.134 using esphome API
INFO Successfully connected to samsung-aircon @ 10.1.1.134 in 0.145s
INFO Successful handshake with samsung-aircon @ 10.1.1.134 in 0.113s
[08:31:56][I][app:102]: ESPHome version 2023.12.9 compiled on Feb  9 2024, 08:29:35
[08:31:56][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:31:57][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:31:57][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:31:57][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:31:57][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:c2;raw:60000000000000006a}
[08:31:57][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:2f;raw:0104010b010b010eed}
[08:31:57][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:31:57][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:31:58][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:31:58][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:31:58][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:c3;raw:00000000001004001f}
[08:31:58][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:23;raw:0000000000000000eb}
[08:31:58][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:31:58][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:31:58][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:31:58][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:31:58][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:cf;raw:000000000000000007}
[08:31:59][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:24;raw:0000000000000000ec}
[08:31:59][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:31:59][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:31:59][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:31:59][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:31:59][W][samsung_ac:018]: update
[08:31:59][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:80;raw:00000150050000001c}
[08:31:59][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:25;raw:0000000000000000ed}
[08:32:00][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:32:00][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:32:00][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:32:00][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:32:00][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:82;raw:02011131300005005c}
[08:32:00][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:28;raw:020078008c0e0e0016}
[08:32:00][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:32:00][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:32:01][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:32:01][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:32:01][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:88;raw:000000000000000040}
[08:32:01][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:2d;raw:0000000000000100e4}
[08:32:01][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:32:01][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:32:01][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:32:01][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:32:02][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:89;raw:2f000000000000006e}
[08:32:02][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:20;command20:{target_temp:26; room_temp:26; pipe_in:26; pipe_out:27; power:0; wind_direction:31; fanspeed:0; mode:08}}
[08:32:02][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:32:02][W][samsung_ac:078]: write 32d000b01f1a1a03f4210000a934
[08:32:02][W][samsung_ac:486]: MSG: {src:00;dst:d0;cmd:54;raw:1f1a1a03f42100004d}
[08:32:02][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:32:02][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:32:02][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:32:02][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:8d;raw:08000000000000004d}
[08:32:02][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:40;raw:2000000060670040ef}
[08:32:03][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:32:03][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:32:03][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:32:03][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:32:03][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:f0;commandF0:{ou_freeze_protection:0; ou_heating_overload:0; ou_defrost_control:0; ou_discharge_protection:0; ou_current_control:0; inverter_order_frequency[Hz]:0; inverter_target_frequency[Hz]:0; inverter_current_frequency[Hz]:0; ou_bldc_fan:0; ou_error_co [... cut ...]
[08:32:12][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}
[08:32:12][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:c2;raw:60000000000000006a}
[08:32:12][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:24;raw:0000000000000000ec}
[08:32:12][W][samsung_ac:486]: MSG: {src:c8;dst:d0;cmd:c6;commandC6:{control_status:1}}
[08:32:13][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
[08:32:13][W][samsung_ac:486]: MSG: {src:c8;dst:f0;cmd:86;raw:0800000000000008be}
[08:32:13][W][samsung_ac:486]: MSG: {src:c8;dst:ad;cmd:d1;raw:fe000000000000004a}

lanwin, 2024-02-09:
Ok so writes work. But your device still did not react to the changes?

Rebem13, 2024-02-09:
> Ok so writes work. But your device still did not react to the changes?

Sorry - I reported earlier (about 3 posts up) that it's all working perfectly thanks. The logs were a response to @matthias882 requesting them. Thanks for your help!

lanwin, 2024-02-09:
Ah sorry I overlooked that :D

matthias882, 2024-02-09:
Yes, thanks for the log. There are only f8 commands to dest00. Hmmm, that's different to many other systems.. But ok, the way to solve this is found.
An it seems so that the cable-remote creates no messages on F1/F2 bus. I see only a communication between outdoor and indoor-unit (c8 and 00). That's nice to know.
And the third point: congratulation, you have nice outside temperature with 29°C, wish a had this here also. In German winter it's 5° in this moment :-D

Rebem13, 2024-02-09:
"And the third point: congratulation, you have nice outside temperature with 29°C, wish a had this here also. In German winter it's 5° in this moment :-D"

Well it hit 43°C here today so the outdoor unit must have been in the shade!!! Oh wait, that was 29°C at 8:30AM ;-)

matthias882, 2024-02-09:
Puh, that's really hot! Hope your AC works perfect now ;)

Rebem13, 2024-02-10:
Actually, is there any way to pull the outdoor unit temperature into Home Assistant? I can see it in this line but I'm not sure how to configure it in the YAML to pull it through. Room Temp didn't work:
[08:32:11][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:c0;commandC0:{ou_operation_mode:08; ou_4way_valve:0; ou_hot_gas_bypass:0; ou_compressor:0; ou_ac_fan:0; ou_outdoor_temp[°C]:29; ou_discharge_temp[°C]:29; ou_condenser_mid_temp[°C]:28}}

lanwin, 2024-02-10:
Can you please make a separate issue for that. For now there is no sensor the uses this value.

lanwin, 2024-02-13:
Ok since the issue itself is gone, I will close this now.

closed 2024-02-13 by lanwin