omerfaruk-aran/esphome_samsung_hvac_bus #90: [NASA] [ESP-IDF] room temperatur not working anymore

DerTiger1, 2024-02-08:
@lanwin, i'd like to ask you again to give versioning a chance. I'm heating with this heatpump and would like to have a developer and a stable version. Currently the room temperature data is broken.

lanwin, 2024-02-08:
@DerTiger1 I created a issue especially for that versioning thing https://github.com/lanwin/esphome_samsung_ac/issues/91 .

Can you please attach your build log and a message log?

I am not sure why this did not work for you. For me the current main branch has working temperatures. Maybe there is a ESP-IDF problem here.

lanwin, 2024-02-13:
@DerTiger1 did it still dont work?

DerTiger1, 2024-02-13:
@lanwin, sorry for the late response - the temperature is working again and it's compiling fine with esp-idf without any compiler hints or issues.