omerfaruk-aran/esphome_samsung_hvac_bus #91: How to do versioning?

lanwin, 2024-02-08:
Since the codebase changes a lot recently there also a lot bugs and breaking changes. So we need a versioning strategy. 

The best would be to have a tag an real version but that would be make more sens when there are news versions that are tested and getting developed over a longer time. Maybe later.....

Currently I think about walking with a `main` (dev) branch and a `stable` branch. The `main` has the latest changed but can be unstable. `stable` contains a mostly stable (expect from known bugs) branch witch gets merged then and when from `main`.

I would mention both in our example config.

But I am not sure if this is a good plan. 

What do you think @atanasenko @matthias882 @bzumik1 @mbo18 @Foggy2 ?

atanasenko, 2024-02-08:
Yea I think that's a good plan. Maybe even add tags now and then before introducing larger changes.

bzumik1, 2024-02-08:
Hello @lanwin, I was reflecting on this today and believe it's an excellent idea. I suggest maintaining stability in the main branch and carrying out development activities in the dev branch, which is a common practice. Additionally, it would be beneficial to systematically track the features in both the main and dev branches.

lanwin, 2024-02-09:
I am not sure. But for a project using a dev branch has the drawback of looking less active. Since most people look in the main branch for activity.

lanwin, 2024-02-09:
For the uses this should be irrelevant since most people simply copying the example config with then points to stable.

TCB13, 2024-02-10:
For future reference, this could be a good candidate for a stable release: https://github.com/lanwin/esphome_samsung_ac/issues/87#issuecomment-1933636382

bzumik1, 2024-02-11:
> For the uses this should be irrelevant since most people simply copying the example config with then points to stable. 

I see what you mean, than your approach is fine.

mbo18, 2024-02-11:
Having 2 branches like you described is fine as long as users don’t have to change their configuration to get the latest stable code

Foggy2, 2024-02-12:
I agree it is a good idea.

I think @mbo18 has a valid point. Anyone that has deployed the project right now would be pulling from main and would end up being on dev if they do not change their configuration.

I think that either should be fine though because main is effectively a dev branch right now. So it should not make the experience any worse for existing users.

lanwin, 2024-02-13:
Ok I have created a stable branch and an initial release. I plan to use date based release tag since I made good experiences with that in the past.

closed 2024-02-13 by lanwin