omerfaruk-aran/esphome_samsung_hvac_bus #101: A list of supported devices

hnykda, 2024-02-15:
Hello and thanks for an awesome work!

Could we get a list of supported devices? Or just keep this issue pinned so people can report tested and working devices here. Would be greatly appreciated!

lanwin, 2024-02-16:
> Could we get a list of supported devices? Or just keep this issue pinned so people can report tested and working devices here. Would be greatly appreciated!

Hi @hnykda currently we did not have a list of supported devices. But we did not yet found a device that we where not able to integrate. So chances are good. We also have uses that use this integration with a headpump instead of a AC cause they also using the NASA protocol.

> Is there something I can ask the local vendors to validate if it works (e.g. “if it has NASA protocol” or “it has F1/F2 cable available”)? Is e.g. this type: [AJ040TXJ2KG/EU+AR09+AR12TXFCAWKNEU](https://bterm.cz/produkt/klimatizace-multisplit/set-2x1/samsung-wind-free-comfort-multisplit-2x1-25-35-kw-vcetne-montaze/) supported?

I am not a AC seller or something and I have not insights in that industry, so I can only guess. But you could ask if the device is a NASA based device. Since the seller needs to sell the corresponding accessories I would guess that this must be relevant. Also my device had NASA in the description to differ them from the older Non NASA based ones. So if its not in the description I would ask the seller. 
When you have the chance to choose, I would recommend to buy NASA devices since there is more possible with them. 

> Also, what features are supported with this integration? Really just everything (that would be living a dream!)? Any tradeoffs when using this?

This integration is pretty young. So you should expect bugs. But the basic stuff works. Set temp, set fanspeed, set mode, see room temperature. Support for setting swing modes and advanced modes like quiet or windfree where added a few days ago (but we have not yet tested that much). And we are right before adding more metrics like power consumption and so on. After that I did not see much what could be missing yet.

hnykda, 2024-02-16:
Hey. That is awesome, thank you a lot for the response, that's very helpful.

hnykda, 2024-05-07:
Thanks @atanasenko !

I found this little help: 
[image]

hnykda, 2024-07-16:
I added this to the readme, we can close this.

closed 2024-07-16 by hnykda