omerfaruk-aran/esphome_samsung_hvac_bus #102: [Non-NASA] Control of Unit No Longer Working With Recent Changes

jbekkema, 2024-02-19:
I just updated my ESPHome devices (M5STACK ATOM Lite) to the latest commit, and control of the AC units is no longer working. Sorry for the new issue: this seems similar to issue #87, but it looks like that has been resolved.

It was previously working using a commit from around mid-to-late January (although it did have the issue where it might take 2 or 3 tries as described in Issue #1 ). I have tried the latest commit from both the main and stable branches and see the same behaviour.

The state updates perfectly (for example, when changing the mode, temperature, etc. via the infrared remote it is reflected correctly), however attempting to control the AC unit has no effect. The ESPHome device acknowledges the command, but it has zero impact, for example:

```
[21:00:21][D][switch:012]: 'Living Room Power' Turning ON.
[21:00:21][D][switch:055]: 'Living Room Power': Sending state ON
```

I'd be happy to go back and try particular commits to see where the problem was introduced, but I can't seem to find a way to pull particular commits using ESPHome for external components (I could be wrong). The same behaviour is occurring for two separate Non-NASA Samsung Split ACs: a AQV30UW and a AQV18UW.

Log file attached, covering from turning the ESPHome device on, leaving it sitting there for a little while, and then attempting to turn the AC unit on (with nothing happening).
[esphome_ac_log_main.txt](https://github.com/lanwin/esphome_samsung_ac/files/14337155/esphome_ac_log_main.txt)

lanwin, 2024-02-20:
Can you please remove the UART debug stuff and add only the debug_log_messages options from here https://github.com/lanwin/esphome_samsung_ac?tab=readme-ov-file#development and then please attach a corresponding log.

We changed the code to send messages only after two kinds of messages appear. When that did not work, its probably cause non of these two messages appear in your configuration. We need to investigate why that is the case here. CC @matthias882

matthias882, 2024-02-20:
Yes, there ist no Comand c6 in the log and no Comand f8 with destination f0. Thats the Problem.
But I have no Idea for a solution.
In my own non-NASA system these are the only 2 commands with a large gap in communication after them to send our data.
we need to know the timing in this communication here to solve the problem.
Print the miliseconds since last message in debug_log?

jbekkema, 2024-02-20:
> Can you please remove the UART debug stuff and add only the debug_log_messages options

Updated log attached. The debug_log_messages options were already included, but I've removed the UART debug options.
[esphome_ac_log_main_2.txt](https://github.com/lanwin/esphome_samsung_ac/files/14348352/esphome_ac_log_main_2.txt)

In addition, after some time of being of the unit being "off" I notice the UART commands stop: I'm guessing the outdoor unit goes into some sort of idle/sleep state. I don't think you can wait for any messages before sending a command in this state, as it doesn't look like there are any (at least any that are being logged). Example:
[esphome_ac_log_main_2_idle.txt](https://github.com/lanwin/esphome_samsung_ac/files/14348594/esphome_ac_log_main_2_idle.txt)

Here is a longer log, showing the AC in the idle state, me turning it on via the infrared control and back off, and then waiting until it enters back into an idle state with no further messages being logged.
[esphome_ac_log_main_2_idle_long.txt](https://github.com/lanwin/esphome_samsung_ac/files/14348947/esphome_ac_log_main_2_idle_long.txt)

So I don't think waiting for a command is a viable option, at least with my two AC models.

> Yes, there ist no Comand c6 in the log and no Comand f8 with destination f0.

How does Non-NASA handle collision detection? It seems like it's checksums with resends? Or is there some sort of master/slave polling? Something else?

> Print the miliseconds since last message in debug_log?

If this something I can configure? Or does it require a source change?

matthias882, 2024-02-20:
As far as I understand the communication it is a master/slave system. The outdoor-unit (0xc8) is Master. He sends messages to the other units and waits for an answer.

In your "wake up from idle" log the indoor unit (0x00) sends a message without beeing asked to wake up the outdoor unit.
```
[16:57:01][W][samsung_ac:032]: RAW: 3200c8250000000000000000ed34
[16:57:01][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:25;raw:0000000000000000ed}
[16:57:01][W][samsung_ac:032]: RAW: 32c8dec70101000000000000d134
[16:57:01][W][samsung_ac:486]: MSG: {src:c8;dst:de;cmd:c7;raw:0101000000000000d1}
```
And because it is no master it gets no answer to it. The Outdoor unit starts a new conversation instead and sends a message to adress 0xde.

The indoor unit message is no special wake-up-message, its the same as in normal communication if it ist asked from the outdoor-unit.
```
[16:57:05][W][samsung_ac:032]: RAW: 32c800c300000000001070006b34
[16:57:05][W][samsung_ac:486]: MSG: {src:c8;dst:00;cmd:c3;raw:00000000001070006b}
[16:57:05][W][samsung_ac:032]: RAW: 3200c8250000000000000000ed34
[16:57:05][W][samsung_ac:486]: MSG: {src:00;dst:c8;cmd:25;raw:0000000000000000ed}
```
I would think every message on bus adressed to the outdoor-unit (0xc8) will wake it up and start the bus traffic. but we should test it. Perhaps our controll-message to the indoor-unit will wake up the bus too, this would be the best case.
The "idle-mode" should be easy to detect (no incoming messages for long period of time) and the data-package to wake up the unit could be send before a controll-message. in best case the controll-message wakes up the system and controlls the indoor unit in one time. -> I would try to send the controll-command while "bus idle mode" instantly without waiting for any message. With a bit of luck it should work ;)

The "waiting for a command" was born from the communication to the SNET pc-software. The Outdoor-Unit polls the software-adress (0xD0) with command 0xc6 and waits a short moment for the answer. But there ist no pc connected ant thatswhy is here enough time to send our control-command.

But there was a problem in this way, the master only polls to 0xd0 if there was the SNET-Software previous connected. Something unknown starts this when you connect the pc-software.

The workarround was born to wait for the command 0xf8 with destination 0xf0. this seems to be also in many comuncations and don't know who destination 0xf0 is, he never answers. thats also a free space on bus to send our commands.

The only correct way should be to find out whats the message from SNET-Software that starts the 0xd0-polling initially and send it every time on startup of the esp or every time there ist no polling for 0xd0

> Is this something I can configure? Or does it require a source change?

no, its not implemented yet

jbekkema, 2024-02-20:
Interesting, thanks for the information @matthias882.

> And because it is no master it gets no answer to it. The Outdoor unit starts a new conversation instead and sends a message to adress 0xde.

Unrelated to this ticket, but it makes me wonder whether it may be worth adding a "keepalive" option that polls the outdoor unit every 60 seconds or so to keep state messages flowing: might be handy if someone is running automations based on things like the reported room temperature.

> Something unknown starts this when you connect the pc-software.

It sounds likely there is some sort of registration command or process that registers an address with the outdoor unit so it knows to poll it? That would probably make sense in a multi-split setup, there the outdoor unit has multiple indoor units it would need to give time to.

I don't have necessary hardware to give the SNET software a try, but I do notice some strange messages when power cycling the AC unit (log attached below). Could these be the indoor unit registering itself? Or just a UART mess from the power cycle?
[esphome_ac_log_main_power_cycle.txt](https://github.com/lanwin/esphome_samsung_ac/files/14350885/esphome_ac_log_main_power_cycle.txt)

It kind of looks like the outdoor unit may be polling for devices (with "32c8ebf900da34"), and then the indoor unit registering (maybe "3200c8fcf0c434"), but that's pure speculation on my part.

lanwin, 2024-02-26:
@jbekkema did you make some tests?

jbekkema, 2024-02-27:
> @jbekkema did you make some tests?

Sorry @lanwin I haven't had a chance yet. I'm hoping to take a closer look once I have some free time. :)

matthias882, 2024-03-05:
I have a new idea to fix the problem but I don't really know how to include it to the code...
What if we generate a table with the information:
"src" ; "dst" ; "cmd" ; "time to next message"
and update it in background or maybe on every "update".
Then the code could cyclic evaluate this table to search for the longest gap in communication and use this src/dst/cmd combination to trigger sending our message.
The communication in every system is a littlebit different, but in all systems it repeats its behaviour in a short cycle. We dont need to send our commands after one special message, we only need a free space on databus for it.

ik-del, 2024-03-12:
I tried out the project and run into an issue similar to what was outlined above: I can see the state of the unit but controlling it doesn't work. I used 2 different hardware setups to eliminate any possible hardware issues and experienced the same behaviour. I will attach a log file showing what happens after startup. After about 2 mins each time after startup no RS485 messages are received or transmitted.
[logs_test.txt](https://github.com/lanwin/esphome_samsung_ac/files/14578687/logs_test.txt)

jbekkema, 2024-03-13:
Ok, so I've started to have a bit of a play with S-NET pro.

It appears when S-NET pro is launched (just launched, not told to Connect yet) the following bytes (hex) are sent. Upon sending the outdoor unit immediately wakes up (if it was sleeping/idle) and starts sending some traffic. S-NET pro doesn't reply to any of the commands from the outdoor unit at this stage.

```
>>> FF:FA:2C:01:00:00:00:00:FF:F0
>>> FF:FA:2C:02:00:FF:F0:FF:FA:2C:03:00:FF:F0:FF:FA:2C:04:00:FF:F0:FF:FA:2C:05:00:FF:F0
```

Breaking that up for readability we see:

```
FF:FA:2C:01:00:00:00:00:FF:F0
FF:FA:2C:02:00:FF:F0
FF:FA:2C:03:00:FF:F0
FF:FA:2C:04:00:FF:F0
FF:FA:2C:05:00:FF:F0
```

To me this looks like some sort of broadcast/scan for outdoor units. I can't rule out it's something the virtual com interface is doing (and it's not triggered by S-NET pro), but the behaviour is consistent and happens each time the software is launched. My guess at this stage is this is how the software finds any outdoor units available to connect to: by looking at the addresses of the replying units.

It seems like this would be a good way for ESPHome to wake the outdoor unit if it's asleep, and possibility identify the outdoor unit (likely address "C8") without a user needing to configure it in the yaml file. It could also be that the outdoor unit will wake up no matter what is sent, in which case simply sending the desired command twice may be enough (if nothing has been received for a period of time).

Next up we have telling the S-NET pro software to connect. Interestingly this process seems similar to above. The S-NET pro software appears to adopt the address "D0" for itself. This is what is sent when clicking "connect":

```
>>> 32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34
>>> FF:FA:2C:01:00:00:09:60:FF:F0:FF:FA:2C:02:08:FF:F0:FF:FA:2C:03:03:FF:F0:FF:FA:2C:04:01:FF:F0:FF:FA:2C:05:00:FF:F0:32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34:32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34:32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34
>>> 32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34
```

Breaking that up for readability we see:

```
32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34
FF:FA:2C:01:00:00:09:60:FF:F0
FF:FA:2C:02:08:FF:F0
FF:FA:2C:03:03:FF:F0
FF:FA:2C:04:01:FF:F0
FF:FA:2C:05:00:FF:F0
32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34
32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34
32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34
32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34
```

To me "32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34" looks like a data request to the outdoor unit. The S-NET pro software is continuously sending it while running. The outdoor unit always replies to it. I don't think the fact it is the first command is relevant. The "FF:FA:2C:" commands having a byte that's different to the "scan" commands (where the byte is always null) is more interesting. Maybe it's simply a checksum, or maybe it's some form of registration (but I don't see how as it doesn't list the "D0" address).

Upon disconnecting S-NET pro there is nothing special sent: it simply stops sending the "32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34" command multiple times a second, and so the outdoor unit stops replying.

Full log attached. The S-NET pro software is opened at 00:54:25, and instructed to connect at 00:56:11. Please note that the outdoor unit already knows about S-NET pro software (I did not power cycle the AC prior to generating the log), which does affect things slightly.
[snet_pro_non_nasa.txt](https://github.com/lanwin/esphome_samsung_ac/files/14580570/snet_pro_non_nasa.txt)

jbekkema, 2024-03-13:
Just a quick follow-up: my suspicions are growing that the "FF:FA...FF:F0" style commands are a result of the virtual terminal, and not necessarily something S-NET pro is explicitly sending. If this is correct it means any UART command will wake the outdoor unit. ESPHome could send a dummy command to wake the unit, before sending the desired command.

If that is also the case, then it's likely ESPHome could also just send the D1 command ("32:D0:C8:D1:D2:00:00:00:00:00:00:00:1B:34") to "register" with the outdoor unit, and then adopt matthias882's approach of waiting for an appropriate command before sending our own. As the outdoor unit always seems to respond to the D1 command, it could likely also be used to test whether the outdoor unit is awake, or wake it up.

Rebem13, 2024-03-13:
Issue #87 was the one I reported and it looks like I have the same issue again - similar to this thread. I can read status of the units but none of my sent commands work. Here is a copy of my logs if that helps:
[logs_samsung-aircon_run (3).txt](https://github.com/lanwin/esphome_samsung_ac/files/14581923/logs_samsung-aircon_run.3.txt)

jbekkema, 2024-03-15:
I've spent some time digging into both the behaviour of the indoor units, as well as S-NET pro's behaviour, and put together a fork that should work for Non-NASA users. You can give it a shot by changing the external_components section in your YAML file to the following:

```
external_components:
  - source: github://jbekkema/esphome_samsung_ac@non_nasa_support
    components: [samsung_ac]
```

The big changes are that we now register with the outdoor unit as a "controller" at start, only send control messages when we receive a request_control message (@matthias882's approach here is correct), and detect and wake the outdoor and indoor units if they're sleeping (which appears to happen after around 5 minutes of idle time). Feedback whether it works for you would be appreciated, as AC unit behaviour may differ slightly.

This implementation is not yet bulletproof in my testing, but it's close. I find the occasional request may not make it through when the UART bus is busy, which I intend on addressing soon by detecting these failures and resending as required.

The other change I intend on making is a YAML configurable option to prevent the outdoor/indoor units from sleeping. This is S-NET pro's behaviour, and it is handy that it allows you to continuously track temperature and other values, however I'm reluctant to have it as the default behaviour in case it adds added wear to the units. I can't see it being a problem, but as the indoor units themselves don't have this behaviour I think it's better to do the same. It would be interesting to know what the behaviour is on a proper wired controller is though.

If you have previously used S-NET pro with your unit, I recommend fully power cycling it prior to trying the fork. It should still work, but as S-NET pro already performs registration behaviour may differ slightly. After starting your ESPHome device, you should also wait 30s prior to trying to control your AC unit (as it appears to wait for an "update" cycle to fire before proper processing starts).

Once I've addressed the above issues/changes I'll send through a pull request (assuming it all works for other Non-NASA users as well).

Rebem13, 2024-03-15:
Wow, clever people in this community never cease to amaze me! I've tried your fork @jbekkema and it works perfectly - control of my AC has been restored. Thank you so much!

ik-del, 2024-03-15:
Amazing, the fork works perfectly for me too!! Thank you!

jbekkema, 2024-03-16:
That's great to hear @Rebem13 and @ik-del! Thanks for testing!

I've also just added some additional updates:

- Command Acknowledgement and Resending: In the rare event the indoor unit doesn't receive/process a control command, it will be automatically resent.
- Keepalive Option: A new "non_nasa_keepalive" config option has been added that can set to prevent the AC device from sleeping. Enabling this option will allow you to track values like indoor and outdoor temperature even when the AC isn't in use (handy if you have automations based on these). It is off by default. https://github.com/jbekkema/esphome_samsung_ac/blob/non_nasa_support/example.yaml#L117
- Fixes Packet Parsing Issue: There was a problem with the packet parsing logic when a packet error occurs (such as a failed CRC check). Corrupted packets do appear to happen every now and then, particularly then the bus is busy. The commit message goes into more detail, but basically it was causing a chunk of valid packets to be lost after a packet error.

With these final changes, controlling both the AC in the office and the AC at home now works perfectly for me every time. Both setups only have single indoor units, so I'm unsure how well it'll hold up just as well in an environment with multiple indoor units or existing controllers, so that's definitely worth testing (however it should work). It's also worth testing against a NASA setup to make sure the packet parsing changes haven't inadvertently broken anything there.

I'll send through a PR so hopefully these changes can be incorporated upstream into the repo.

closed 2024-04-10 by jbekkema

wifi75, 2025-04-03:
> external_components:
>   - source: github://jbekkema/esphome_samsung_ac@non_nasa_support
>     components: [samsung_ac]

hello please can you post your configuration for non nasa?

external_components:
  - source: github://jbekkema/esphome_samsung_ac@non_nasa_support

wifi75, 2025-05-14:
No news?