omerfaruk-aran/esphome_samsung_hvac_bus #113: Did I fry my 485 unit?

Nanganator, 2024-03-25:
I cannot for the life of me get this working. My heat pump is a MH068FXEA4. I have 2X F1 and F2 terminals on my outdoor unit with 2X indoor units connected to each F1 and F2 (total 4 indoor units). I am getting nothing but “-“ for indoor and outdoor.

The UART TTL to RS485 unit I am using is one of these [HW-0519](https://www.aliexpress.com/item/1005005866372508.html) from AliExpress with an ESP-WROOM-32 dev board on UART2. After trying all the troubleshooting I noticed the TX and RX lights are constantly on, even if only power is supplied and A/B/TX/RX are not connected to anything.
Can anyone with the same board confirm if yours does the same or have I maybe fried my boards?

I have tried 9600 and 2400 baud rate.
I have tried both of the F1/F2 terminals on my outdoor unit.
I have tried swapping F1 and F2.
I tried disabling logging but then get no output so cannot see if it is working.

Any ideas what I might have done wrong?
(apologies if this is not the correct place for this)

Nanganator, 2024-03-25:
My ESPHome config is:
`esphome:
  name: accontroller
  friendly_name: accontroller

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password:

wifi:
  ssid: !secret wifi_ssid_2
  password: !secret wifi_password_2

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Accontroller Fallback Hotspot"
    password: "kGVu1zv0xCC7"

captive_portal:
    
# Specify pins used by the board to comunicate with RS485 board - in this case M5STACK
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  # For older devices it could also be 2400. But you should start with 9600.
  baud_rate: 9600
  parity: EVEN

web_server:
  port: 80

## SAMSUNG AC CONFIGURATION

# Import custom component from GitHub
external_components:
  - source: github://lanwin/esphome_samsung_ac@stable # use @main if you want the latest development (possibly unstable?)
    components: [samsung_ac]

# Configuration of AC component
samsung_ac:
  devices:`

brazoayeye, 2024-03-26:
> The UART TTL to RS485 unit I am using is one of these [HW-519](https://www.aliexpress.com/item/1005005866372508.html) from AliExpress with an ESP-WROOM-32 dev board on UART2. After trying all the troubleshooting I noticed the TX and RX lights are constantly on, even if only power is supplied and A/B/TX/RX are not connected to anything. Can anyone with the same board confirm if yours does the same or have I maybe fried my boards?
The real question is: are internal - external units still working when you remove component? If it's so, nothing has been fried. 
I used the same component, but I connected to pins   tx_pin: GPIO4,  rx_pin: GPIO34 (my custom board exposed only few pins)
> I have tried 9600 and 2400 baud rate. I have tried both of the F1/F2 terminals on my outdoor unit. I have tried swapping F1 and F2. I tried disabling logging but then get no output so cannot see if it is working.
Double check connections from esp, and pin setup. you may have inverted TX/RX channels?
In that component I only used A/B in the RS485 side, I let the third unconnected.

Nanganator, 2024-03-27:
I guess it just does not work with the MH068FXEA4 outdoor and MH026FPEA indoor unit combo. What a shame. I'll have to try with IR instead.

closed 2024-03-27 by Nanganator

lanwin, 2024-04-02:
@Nanganator sorry I was off for a while.

"TX and RX lights are constantly" seems to indicate a problem with wiring. I had that mostly when F1 and F2 connection where swapped. 

I guess your AC would work with this component. It must be a problem with your HW components here. The software will only work when the connection indicator blinks only from time to time.

Nanganator, 2024-04-05:
@lanwin Thanks so much for the update! Your comment gave me the motivation to get my hands on another RS485 unit, connect to my FTDI adapter first to ensure it was sending & receiving properly. Then hooked it up. Little bit of enabling/disabling logging, 9600 then 2400, and swapping the F1/F2 later and I’m now pulling data from the 4 indoor units!! Sending not yet working but I’ll keep fiddling over the weekend. Possibly not good wiring contact like you mention.

Guess my 485 unit was for sure fried. Even only supplying power and not connecting A or B to anything, RX and TX lights came on solid.

Thanks again for the help!!

TCB13, 2024-04-21:
> Guess my 485 unit was for sure fried. Even only supplying power and not connecting A or B to anything, RX and TX lights came on solid.

I've been using a [module like yours since ever](https://github.com/lanwin/esphome_samsung_ac/issues/59#issue-2101026311) and at some point I had that behavior as well. But after a few restarts it was okay.