omerfaruk-aran/esphome_samsung_hvac_bus #116: file example.yaml is all in errors when you try to get the firmware

AAfetisov, 2024-04-14:
INFO ESPHome 2024.3.2
INFO Reading configuration /config/esphome/samsungac.yaml...
WARNING 'samsung_ac': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
Failed config

samsung_ac: [source /config/esphome/s[image]
    vertical_swing: True
    horizontal_swing: True
    presets: 
      quiet: 
        name: Makes no sound
        enabled: True
  devices: 
    - address: 20.00.00
      
      [capabilities] is an invalid option for [0]. Please check the indentation.
      capabilities: 
        horizontal_swing: False
      climate: 
        name: Kitchen climate
      room_temperature: 
        name: Kitchen temperature
      target_temperature: 
        name: Kitchen target temperature
      power: 
        name: Kitchen power
      mode: 
        name: Kitchen mode
      
      [room_temperature_offset] is an invalid option for [0]. Did you mean [room_temperature], [target_temperature]?
      room_temperature_offset: -1.4
      room_humidity: 
        name: Kitchen humidity
      
      [water_temperature] is an invalid option for [0]. Did you mean [target_temperature], [room_temperature]?
      water_temperature: 
        name: Warm water
      
      [outdoor_temperature] is an invalid option for [0]. Did you mean [room_temperature], [target_temperature]?
      outdoor_temperature: 
        name: Outdoor temperature

brazoayeye, 2024-04-14:
Maybe are you using stable instead of main?

`github://lanwin/esphome_samsung_ac@stable # use @main if you want the latest development (possibly unstable?)`

AAfetisov, 2024-04-14:
https://github.com/lanwin/esphome_samsung_ac/blob/main/example.yaml
got it from the guide in main. The link from stable leads to the same file btw)

closed 2024-04-14 by AAfetisov