omerfaruk-aran/esphome_samsung_hvac_bus #130: [NASA] Override Temperature Readings Of Indoor Unit

bzumik1, 2024-06-20:
I believe this should be relatively straightforward as the AC system supports central control. It would be highly beneficial to have the option to use Home Assistant (HA) sensors to provide temperature information to the air conditioning (AC) system. This would enable more accurate temperature control and enhance overall system performance.

Has anyone attempted to check if this can be implemented?

blka, 2024-07-10:
@bzumik1 did u tried setup climate by using „better thermostat” maybe? There is an option to add external temperature sensor.

bzumik1, 2024-07-10:
Hi @blka, yes I did, but unless I am doing something wrong the better thermostat just turns on and off the AC and that is not what I want. I want to have the AC on all the time and let it regulate the wind power etc.

closed 2024-10-29 by omerfaruk-aran

vniehues, 2025-06-30:
@omerfaruk-aran this was closed as completed but as far as I can see this is not actually done. 
To my knowledge we are able to adjust a temperature offset for the temperature that homeassistant shows but that does not actually change the temperature that the indoor unit uses. 
The problem with this is that when the indoor unit has the correct values, it will automatically lower it's cooling power in all modes (not just auto) when the current temperature is close to the target temperature. 
This of course does not happen when we just adjust the reported temp in homeassistant. 

Since external temperature sensors (standalone and with a controller) exist this should be possible. 
As far as I have researched, the external sensor uses F3/F4 to communicate with the indoor unit and allows sending it's own temperature reading to the indoor unit which then uses those. 
The indoor unit needs to be configured correctly to actually use these external values but the functionality is there. 

My proposal would be to allow the user to select a sensor from homeassistant and send those values to the indoor unit. 
Of course we need to do some research to check how the data has to be sent in order to do this.

vniehues, 2025-06-30:
[This document](https://res.cloudinary.com/govimg/image/upload/v1622831152/5b294f9467c0d0489028b276/Controls%20product%20overvierw%20course_6_4_2021_JDLP_V2.pdf) describes the "Wired Remote Controllers" (over F3/F4 12VDC) as well as the "External Room Sensor" which directly plugs into the indoor unit's PCB - I will try to get my hands on an external sensor either from Samsung or just a generic two-wire-sensor (which should work if I interpret this correctly) and see if this makes the AC behave more in line with my expectations. 

I still think it would be great if we can find a way to report external sensor data to the indoor unit via ESPHome.

nicolaerosia, 2025-06-30:
Take a look here https://github.com/70p4z/samsung-nasa-mqtt

vniehues, 2025-07-01:
Oh nice, thank you @nicolaerosia!
From the link above, [this](https://github.com/70p4z/samsung-nasa-mqtt/blob/399a461b1b478a5655a6fa8bed566b2bd19d70b2/nasa_messages.py#L327) looks like it's what we want to set. 
We still would need to figure out how to set the indoor unit to actually use that value (Could be [this](https://github.com/70p4z/samsung-nasa-mqtt/blob/399a461b1b478a5655a6fa8bed566b2bd19d70b2/nasa_messages.py#L209). I think a 0 indicates no external sensor while a 1 indicates to use the external sensor - I saw that somewhere yesterday but I can't find it right now).

I will open an issue over there and ask for clarification on that as well as a better hardware overview so that I can hook up my spare Raspberry to test if this works.

nicolaerosia, 2025-07-01:
@vniehues maybe you can just publish this packet over F1/F2

nicolaerosia, 2025-07-01:
@vniehues if it works, please let me know - I had this on my TODO list for a while. I want to report "my own temperature" to get water law working with that instead of what the controller reports.
I even bought a THVD8000EVM to connect to F3/F4 but did not have time.

70p4z, 2025-07-06:
Hello there!
I can confirm the only way I found to skip wired controller from samsung and replace them with wireless custom sensors (+ adding few scripting from home assistant to lower the hysterisis to 0.2C for my floor heating) was to use f3/f4 bus and forge notification (neither write nor request) over 423a and 42da to pusg my custom temps into the samsung cobtroller.
Also you can keep your wired controller plugged in but you only need to disconnect its internal sensor (hence having it complain error 653 at boot) but manual override for on/off and temps is therefore possible.
Cheers

nicolaerosia, 2025-07-08:
@vniehues where is your remote connected to? not F3/F4 ?

70p4z, 2025-07-08:
I'm not sure about the sensor displaying the ambient temperature, but it updates its display with the flow target and ambient target temperature I send over F3/F4.
Also concerning sending ambient temps over F1/F2, you could probably try the same sequence as I did:

ZONE 1: send a notification packet 
- write the ZONE1 ambient temperature in 0x423A
- write the zone ambietn sensor enabled: 0x4076 = 1
example:
510000b0ff20c0148902423a00f6407601
src:510000 dst:b0ff20 type:normal ins:notification nonce:0x89
   0x423a (VAR_IN_TEMP_ROOM_ZONE1): 00f6
   0x4076 (ENUM_IN_ROOM_TEMP_SENSOR ): 01

ZONE2: same as zone 1 but with changed NASA FSV
- write the ZONE2 ambient temperature in 0x42DA
- write the zone ambient sensor enabled: 0x4118 = 1
510000b0ff20c0148a0241180142da00fb
src:510000 dst:b0ff20 type:normal ins:notification nonce:0x8a
   0x42da (VAR_IN_TEMP_ROOM_ZONE2): 00fb
   0x4118 (ENUM_IN_ROOM_TEMP_SENSOR_ZONE2): 01

Warning: binary encoded packet are not NASA armored (missing start 0x32 byte, crc 2 bytes and end 0x34 byte)

Warning: Updating the 0x4203 (current ambient temp zone1) and 0x4234 (current ambient temp zone2) does not work, they are somewhat read only (at least for MIM E03CN indoor controller).

Cheers