omerfaruk-aran/esphome_samsung_hvac_bus #144: [NASA] Control Beep Sound

hnykda, 2024-07-16:
Hi,

my ACs have an option to turn on the beeping sound when controlling the AC. I can control this via a remote or via Smarthings app. 

I am trying to figure out how to add this, so I found that there are two promising codes in NASA spec. I added custom sensors like this:

```
  devices:
    - address: "20.00.00"
      climate:
        name: "Haven room AC"
      custom_sensor:
        # NASA_USE_SILENCE
        - name: "Haven Silence Mode"
          message: 0x4046
        # NASA_CONTROL_SILENCE
        - name: "Haven Silence Mode Control"
          message: 0x4045
```

but they are always set to 0 and 1, respectively (I can see them being logged), even when I change the settings. So I suspect it's not the correct code (and these are just some hardcoded values). 

Any idea how this can be added? 

Here is a full log when I was trying to toggle these:
[logs_ac-control-beta_logs.txt](https://github.com/user-attachments/files/16245391/logs_ac-control-beta_logs.txt)

lanwin, 2024-07-16:
Hi @hnykda as I understand, the silence mode is a mode where the AC itself is more silent. I dont think this is related to the BEEP.

What you can to is to turnoff the BEEP multiple times and look if the log contains the an equal message all the times 0-3sec after you turned it off.

hnykda, 2024-07-16:
Yeah, I suspected something like that. So here I turned on the AC, and triggered the beep sound with the remote, and started checked logs for up to approx 10-30 seconds from that time. But it seems there is nothing useful? I asked chatgpt to find any inconsistencies, didn't find any, nor I can eyeball anything (it was on a unit with 
address 20.00.00). 

[b.log](https://github.com/user-attachments/files/16250956/b.log)
[a.log](https://github.com/user-attachments/files/16250961/a.log)

lanwin, 2024-07-18:
When you do that via remote I would expect that. Then the Indoor unit receives the remote command and execute it directly without sending any status via BUS.

Can you check the same via Smartthing. There must be a message since there is no other way the command could be traval to to Indoor unit.

hnykda, 2024-07-18:
[logs_ac-control-beta_run.txt](https://github.com/user-attachments/files/16285002/logs_ac-control-beta_run.txt)

here is a similar try with raw messages

lanwin, 2024-07-19:
What do you mean by skipped? Debug Log should log everything witch could be decoded. But we dont use everything since we dont know the meaning of most of the messages.

Can you please make 3 examples where you change the volume from 100 to 0 with smartthings and three logs where that start exactly at the second where you made the change.

mchiriciuc, 2024-07-20:
Hi,
I have tried to find the same command but with a different approach: I used the mqtt logging and looked for a topic which changed it's value form 0 to 1 with the same pattern as my enable / disable BEEP from smartthings.
Found no message with that pattern.
I think that the indoor unit does not publish this (and the dispaly on/off) to the F1/F2 buss since it is for that internal unit only.

I found message 0x42e5 which is filter clean interval in hours (if it is usefull to anybody), but I'm unable to set up a custom sensor for it, simply it stays "unavailable", even if updates are present on the debug, like this:

[21:29:32][W][samsung_ac:687]: MSG: #Packet Src:20.00.00 Dst:b3.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 159}
 > Variable 42e5 = 300

I also found that message 0x8420 is providing energy consumption since last power on, and a custom sensor is working fine for it.

Let me know if I can test things to improve the project.

mchiriciuc, 2024-07-20:
Sorry, my bad with the custom sensor, not placing int on the indoor section :))

lanwin, 2024-07-22:
Oh nice findings. Yes I think to only option would be to do it from Smartthings. Since there must be a message witch goes from Smartthings hub to the indoor unit via F1/F2.

hnykda, 2024-07-22:
> What do you mean by skipped? 

Well, the packet numbers are having gaps in them. That's what I meant. But I expected that what you said is happening - there are more packets flying around but only some are those we parse and show.

When you say debug log, do you mean the UART debug log or the raw messages? Because with raw messages I didn't seem to get any more of them. Does MQTT shows all, even the one it can't parse?

btw. I don't have a hub when I tried that, this is all remotely from cloud

lanwin, 2024-07-22:
As far as I know, when you have Smart Thins connected to your AC, then you must have a device like "SAMSUNG | WiFi Interface MIM-H04EN | NASA" connected to your AC. That device connects your AC system to WIFI and makes a bridge or a hub to smartthings.

Can you post your device list that gets logged? I assume there must be 3 devices listed.

hnykda, 2024-07-22:
> As far as I know, when you have Smart Thins connected to your AC, then you must have a device like "SAMSUNG | WiFi Interface MIM-H04EN | NASA" connected to your AC. That device connects your AC system to WIFI and makes a bridge or a hub to smartthings.

Then it has to be built in inside the AC. There was simply a guide in the manual to hold some button on remote for 5 seconds, the AC display showed "AP", and then I could add the (indoor) AC unit to my Smartthings.

> Can you post your device list that gets logged? I assume there must be 3 devices listed.

Those are devices from here https://github.com/lanwin/esphome_samsung_ac/issues/144#issuecomment-2236770883 - two indoor units. And then one outdoor unit. So that's the three you guessed?

hnykda, 2024-07-23:
Wow. This is beyond my understanding, but sounds awesome! Thanks for doing this. Let me know if there is something I could help with.

hnykda, 2024-08-25:
I tried again with this commit https://github.com/lanwin/esphome_samsung_ac/commit/0e3463e7001da8c03a03cfbefc20ce3e56520870 and 

```
  debug_log_undefined_messages: true
```

but when I try turning the beep via API, I can see it does work - it beeps - but I don't see anything logged in the output.

hnykda, 2024-08-25:
I tried again with this commit https://github.com/lanwin/esphome_samsung_ac/commit/0e3463e7001da8c03a03cfbefc20ce3e56520870 and 

```
  debug_log_undefined_messages: true
```

here is a log when I try turning on the beep via API (I turned it on somewhere after 22:08:05):

<details>

```
[22:07:58][I][app:100]: ESPHome version 2024.8.0 compiled on Aug 25 2024, 22:06:50
[22:08:04][W][samsung_ac:796]: MSG: #Packet Src:10.00.00 Dst:b0.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 36}
 > Variable 202 = 65535
 > Variable 207 = 2
 > LongVariable 410 = 0
 > LongVariable 41b = 2162687
 > Variable 22f7 = 13
 > Variable 22f9 = 0
 > Variable 22fb = 89
 > Variable 22fd = 0
 > Enum 8000 = 0
 > Enum 8002 = 0
 > Enum 8003 = 1
 > Enum 800d = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 202 = 65535
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 207 = 2
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff LongVariable 410 = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff LongVariable 41b = 2162687
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 22f7 = 13
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 22f9 = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 22fb = 89
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 22fd = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8000 = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8002 = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8003 = 1
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 800d = 0
[22:08:04][W][samsung_ac:796]: MSG: #Packet Src:10.00.00 Dst:b0.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 37}
 > Structure #608 = 10
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Structure #608 = 10
[22:08:04][W][samsung_ac:796]: MSG: #Packet Src:10.00.00 Dst:b0.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 38}
 > Structure #60c = 7
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Structure #60c = 7
[22:08:04][W][samsung_ac:796]: MSG: #Packet Src:10.00.00 Dst:b0.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 39}
 > Enum 8010 = 0
 > Enum 801a = 0
 > Enum 8020 = 255
 > Enum 8030 = 255
 > Enum 8032 = 255
 > Enum 8033 = 0
 > Enum 8043 = 1
 > Enum 8045 = 1
 > Enum 8046 = 1
 > Enum 8047 = 255
 > Enum 8048 = 255
 > Enum 8061 = 0
 > Enum 8066 = 0
 > Enum 8077 = 0
 > Enum 807c = 0
 > Enum 807d = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8010 = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 801a = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8020 = 255
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8030 = 255
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8032 = 255
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8033 = 0
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8043 = 1
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8045 = 1
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8046 = 1
[22:08:04][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8047 = 255
[... 799 lines cut ...]
 > Variable 8248 = 65535
 > Enum 801a = 0
 > Variable 82d4 = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 823f = 200
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8001 = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8003 = 1
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 808d = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 803f = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 8243 = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 8235 = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 805e = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 8031 = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 8248 = 65535
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Enum 801a = 0
[22:08:30][W][samsung_ac:773]: Undefined s:10.00.00 d:b0.00.ff Variable 82d4 = 0
[22:08:30][W][samsung_ac:796]: MSG: #Packet Src:20.00.00 Dst:b0.00.ff {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 4;PacketNumber: 20}
 > Variable 4202 = 190
 > Variable 4203 = 240
 > Variable 4205 = 228
 > Variable 4206 = 226
[22:08:30][W][samsung_ac:773]: Undefined s:20.00.00 d:b0.00.ff Variable 4202 = 190

```

</details>

omerfaruk-aran, 2024-08-25:
Hello @hnykda,

I reviewed the log records again, but I couldn't find any logs related to the beep sound. When you try to control the beep sound, pay special attention to the log entries that start with "Undefined s:20.00.00." These logs should be triggered when you control the beep sound via the API or SmartThings.

hnykda, 2024-08-26:
Hey. Yeah, I get that I am looking for those for `20.00.00`, but similarly to you, I don't see anything "unusual". Don't really know what else I should do now (but thanks for the hint!).

omerfaruk-aran, 2024-08-26:
> Hey. Yeah, I get that I am looking for those for `20.00.00`, but similarly to you, I don't see anything "unusual". Don't really know what else I should do now (but thanks for the hint!).

You’re welcome. I would like you to try one more thing. Please set the variables debug_log_messages and debug_log_undefined_messages to false, and while monitoring the log screen, activate/deactivate the beep sound of the air conditioner through SmartThings. Try this frequently, and if any logs appear on the screen, please send them to me.

omerfaruk-aran, 2024-08-27:
Hello @hnykda 
Have you had a chance to try my last article?

hnykda, 2024-08-30:
Hi @omerfaruk-aran , thanks for trying to help, really appreciated!

So I have this config, where the defaults for `debug_log_*` is `false` I believe:

<details>

```
samsung_ac:
  capabilities:
    vertical_swing: true
    horizontal_swing: true
    presets:
      quiet: true
      fast: true
      windfree: true
      sleep: true
  
  # /opt/homebrew/opt/mosquitto/sbin/mosquitto -c /opt/homebrew/etc/mosquitto/mosquitto.conf
  # debug_mqtt_host: 192.168.0.203
  # debug_mqtt_port: 1883
  # debug_mqtt_username: alpha
  # debug_mqtt_password: alpha
  
  ### DEBUGGING
  # debug_log_messages: true
  # debug_log_messages_raw: true
  # debug_log_undefined_messages: true
  ### DEBUGGING

  non_nasa_keepalive: false

  devices:
    - address: "20.00.00"
      climate:
        name: "Haven room AC"
      room_temperature:
        name: "Haven room AC temperature"
      target_temperature:
        name: "Haven room AC target temperature"
      power:
        name: "Haven room AC power"
      mode:
        name: "Haven room AC mode"
      room_humidity:
        name: "Haven room AC humidity"
      room_temperature_offset: -1
      # https://github.com/lanwin/esphome_samsung_ac/issues/144
      # custom_sensor:
      #   - name: "Haven Silence Mode"
      #     message: 0x4046
      #   - name: "Haven Silence Mode 1"
      #     message: 0x4045
      automatic_cleaning:
        name: "Haven room AC automatic clean"
    - address: "20.00.01"
      climate:
        name: "Kitchen AC"
      room_temperature:
        name: "Kitchen AC temperature"
      target_temperature:
        name: "Kitchen AC target temperature"
      room_humidity:
        name: "Kitchen AC humidity"
      power:
        name: "Kitchen AC power"
      mode:
        name: "Kitchen AC mode"
      automatic_cleaning:
        name: "Kitchen AC automatic clean"
      room_temperature_offset: -2
    - address: "10.00.00"      
      outdoor_temperature:
        name: "Eastern AC Outdoor temperature"
      custom_sensor:
        - name: "Eastern AC Power"
          message: 0x8411      
          unit_of_measurement: W
          icon: "mdi:flash"
          device_class: power
        - name: "Eastern AC Energy total"
          message: 0x8414
          unit_of_measurement: Wh
          icon: "mdi:counter"
          device_class: energy

```

</details>

But when toggling the beep sound, I see nothing in the logs (the first yellow log appears when turning on the AC):

[image]

hnykda, 2024-08-30:
Hi. I understand what you are saying, but I think it's simply not true. Here is a screenshot from the Smartthings android app: https://github.com/lanwin/esphome_samsung_ac/issues/121#issuecomment-2230985118 , specifically this:

[image]

This is what I was toggling when sending you the logs above. I was not doing it via remote, but via smartthings app

omerfaruk-aran, 2024-08-30:
@hnykda This is great, when you toggle Sound Effect it should actually show a package. Can you try this at least 10 times, waiting 1 second between turning on/off again and again, and observe the result?

hnykda, 2024-08-30:
Eh, I can try again, but this is literally what I did above. I waited like 5 seconds at least, and tried that maybe for 5 times? Do you want me to try it again regardless? 😁

omerfaruk-aran, 2024-08-30:
> Eh, tekrar deneyebilirim ama tam olarak yukarıda yaptığım şey bu. En azından 5 saniye bekledim ve bunu belki 5 kez denedim? Tekrar denememi ister misin? 😁

So, the packages are supposed to arrive, but I don't understand why they haven't arrived yet. Do you have access to the SmartThings API? You can use the following endpoint: `https://api.smartthings.com/v1/devices/`. After querying the devices using Postman or the Chrome plug-in TalendApi to obtain the `deviceId`, you can `GET` the device information from the `https://api.smartthings.com/v1/devices/{deviceId}/status` service. Could you please send it to me?

hnykda, 2024-08-31:
Sure, here it is:

<details>

```
{
  "components": {
    "main": {
      "relativeHumidityMeasurement": {
        "humidity": {
          "value": 65,
          "unit": "%",
          "timestamp": "2024-08-31T05:03:43.956Z"
        }
      },
      "custom.airConditionerOdorController": {
        "airConditionerOdorControllerProgress": {
          "value": null
        },
        "airConditionerOdorControllerState": {
          "value": null
        }
      },
      "custom.thermostatSetpointControl": {
        "minimumSetpoint": {
          "value": 16,
          "unit": "C",
          "timestamp": "2024-08-31T04:45:04.498Z"
        },
        "maximumSetpoint": {
          "value": 30,
          "unit": "C",
          "timestamp": "2024-08-31T04:45:04.498Z"
        }
      },
      "airConditionerMode": {
        "availableAcModes": {
          "value": null
        },
        "supportedAcModes": {
          "value": [
            "auto",
            "cool",
            "dry",
            "wind",
            "heat"
          ],
          "timestamp": "2024-08-31T04:45:04.498Z"
        },
        "airConditionerMode": {
          "value": "cool",
          "timestamp": "2024-08-31T04:16:13.870Z"
        }
      },
      "custom.spiMode": {
        "spiMode": {
          "value": null
        }
      },
      "samsungce.deviceIdentification": {
        "micomAssayCode": {
          "value": null
        },
        "modelName": {
          "value": null
        },
        "serialNumber": {
          "value": null
        },
        "serialNumberExtra": {
[... 506 lines cut ...]
        },
        "veryFineDustFilterCapacity": {
          "value": null
        }
      },
      "custom.doNotDisturbMode": {
        "doNotDisturb": {
          "value": null
        },
        "startTime": {
          "value": null
        },
        "endTime": {
          "value": null
        }
      }
    }
  }
}
```

</details>

I can make API calls and also control it from the mysmarththings. Check it here, where I also record the ESPhome screen:
[recording](https://github.com/user-attachments/assets/1eaa4868-d641-4036-9891-e8c8aef0f16f)

omerfaruk-aran, 2024-08-31:
It's really annoying. We'll keep looking until we find it. If we can't solve it, at worst we can solve it with an API plugin. :)

hnykda, 2024-08-31:
Good idea. Thanks for your help and let me know if I could help any further.

titof2375, 2024-09-05:
Hello, in the same vein, there is also the light (lighting) which is below the bit

techyporcupine, 2024-10-11:
Any update on this, thanks!

techyporcupine, 2024-10-11:
I don't know if this helps, but there are addresses here for "buzzer" control on intesis modbus devices. https://www.idx.co.za/wp-content/uploads/2021/01/samsung-nasa-vrf-modbus-user-manual.pdf

htht2001, 2024-11-02:
I don't know if it helps
The beeping can be controlled by S-NET in the programming codes
[image]

omerfaruk-aran, 2024-11-04:
@techyporcupine , @htht2001 ;
Thank you for sharing the information regarding the buzzer control settings from both the Intesis modbus documentation and the S-NET Pro screenshot.

### Feedback on Intesis Modbus Device Information
Unfortunately, despite the helpful details in the shared modbus manual for buzzer control addresses, our current integration cannot use these to pinpoint or catch specific control packets for the buzzer in the Samsung system. This is mainly because the Modbus protocol differs from the packet structure we work with in either NASA or non-NASA protocols for Samsung VRF systems.

### Comments on S-NET Pro Programming Codes
The S-NET Pro configuration settings for the buzzer (highlighted in your screenshots) are intriguing. However, these options only adjust the indoor unit’s configuration via proprietary programming commands. Without additional packet capture or a deeper understanding of the S-NET interface behavior, we cannot utilize these options to intercept or control the buzzer through the communication packets our system processes.

In summary, while these are great leads, they unfortunately do not provide us with enough information to identify or generate the control packet for the buzzer.

Thank you for your contributions, and if you have further insights or if we manage to intercept packets from S-NET, we could continue this investigation.

htht2001, 2024-11-04:
This is the logs I sent programming commands to via S-NET

```
[14:00:56][W][samsung_ac:867]: Write #Packet Src:80.ff.00 Dst:20.01.14 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 2;PacketNumber: 196}
 > Enum 03 = 9
 > Enum 05 = 1
[14:00:56][W][samsung_ac:862]: Response #Packet Src:20.01.14 Dst:80.ff.00 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 5;PacketNumber: 196}
 > Enum 03 = 255
 > Enum 05 = 255
[14:00:56][W][samsung_ac:867]: Write #Packet Src:80.ff.00 Dst:20.01.14 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 2;PacketNumber: 198}
 > Structure #616 = 5
[14:00:56][W][samsung_ac:862]: Response #Packet Src:20.01.14 Dst:80.ff.00 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 5;PacketNumber: 198}
 > Structure #616 = 5
[14:00:56][W][samsung_ac:867]: Write #Packet Src:80.ff.00 Dst:20.01.14 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 2;PacketNumber: 200}
 > Structure #615 = 35
[14:00:56][W][samsung_ac:862]: Response #Packet Src:20.01.14 Dst:80.ff.00 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 5;PacketNumber: 200}
 > Structure #615 = 13
[14:00:57][W][samsung_ac:877]: Read #Packet Src:80.ff.00 Dst:20.01.14 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 1;PacketNumber: 206}
 > Structure #616 = 5
[14:00:57][W][samsung_ac:862]: Response #Packet Src:20.01.14 Dst:80.ff.00 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 5;PacketNumber: 206}
 > Structure #616 = 5
[14:00:57][D][climate:396]: 'fl' - Sending state:
[14:00:57][D][climate:399]:   Mode: OFF
[14:00:57][D][climate:404]:   Fan Mode: AUTO
[14:00:57][D][climate:416]:   Swing Mode: VERTICAL
[14:00:57][D][climate:419]:   Current Temperature: 20.20°C
[14:00:57][D][climate:425]:   Target Temperature: 20.00°C
[14:00:57][D][sensor:094]: 'error_code_00': Sending state 0.00000  with 0 decimals of accuracy
[14:00:58][W][samsung_ac:867]: Write #Packet Src:80.ff.00 Dst:20.01.14 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 1;PacketType: 1;DataType: 2;PacketNumber: 211}
 > Enum 05 = 2
[14:00:58][W][samsung_ac:862]: Response #Packet Src:20.01.14 Dst:80.ff.00 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 5;PacketNumber: 211}
 > Enum 05 = 255
[14:00:58][W][samsung_ac:867]: Write #Packet Src:80.ff.00 Dst:20.01.14 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 2;PacketNumber: 235}
 > Enum 03 = 10
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.00
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.01
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.02
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.03
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.04
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.05
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.06
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.07
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.08
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.09
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.0a
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.0b
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.0c
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.0d
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.0e
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.0f
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.10
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.11
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.12
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.13
[14:00:58][D][device_state_tracker:046]: No change in value for device: 20.01.14
[14:00:58][C][samsung_ac:053]: Configured devices: 10.01.00, 10.01.01, 20.01.00, 20.01.01, 20.01.02, 20.01.03, 20.01.04, 20.01.05, 20.01.06, 20.01.07, 20.01.08, 20.01.09, 20.01.0a, 20.01.0b, 20.01.0c, 20.01.0d, 20.01.0e, 20.01.0f, 20.01.10, 20.01.11, 20.01.12, 20.01.13, 20.01.14
[14:00:58][C][samsung_ac:065]: Discovered devices:
[14:00:58][C][samsung_ac:066]:   Outdoor: 10.01.00, 10.01.01
[14:00:58][C][samsung_ac:067]:   Indoor:  20.01.00, 20.01.01, 20.01.02, 20.01.03, 20.01.04, 20.01.05, 20.01.06, 20.01.07, 20.01.08, 20.01.09, 20.01.0a, 20.01.0b, 20.01.0c, 20.01.0d, 20.01.0e, 20.01.0f, 20.01.10, 20.01.11, 20.01.12, 20.01.13, 20.01.14, 20.ff.ff
[14:00:58][C][samsung_ac:070]:   Other:   65.01.00, 80.ff.00
[14:01:00][W][samsung_ac:867]: Write #Packet Src:10.01.00 Dst:20.aa.c6 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 2;PacketNumber: 123}
 > Enum 2004 = 3
 > LongVariable 418 = 2140870
 > Variable 217 = 33389
 > LongVariable 417 = 2097428
 > LongVariable 419 = 2097663
 > Enum 2012 = 22
[14:01:00][W][samsung_ac:862]: Response #Packet Src:20.01.14 Dst:10.01.00 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 5;PacketNumber: 123}
 > Enum 2004 = 4
 > LongVariable 418 = 2140870
 > Variable 217 = 33389
 > LongVariable 417 = 2097428
 > LongVariable 419 = 2097663
 > Enum 2012 = 4
```

hnykda, 2024-11-04:
(That would be pretty cool if we could sniff the packet structure from S-NET if S-NET actually uses that modbus interface as well)

omerfaruk-aran, 2024-11-04:
That’s a great idea, and I think it could provide valuable insights if we manage to sniff the packet structure from S-NET. However, I’m currently very busy working on an SAP Business One add-on related to inflation accounting in Turkey. I’ll definitely look into this further when I have some free time. Thanks for your patience and understanding!

schejbity, 2025-05-21:
Do you think the commands sent by the remote control are NASA packets?
If so, I can try sniffing with Flipper Zero or with the IR-to-WiFi bridge I use with the TV.
What do you think?

hnykda, 2025-05-21:
Eh, no idea, if you could try, it would be great.

titof2375, 2025-07-03:
Hello, I am sure that this person managed to turn off the beep in their integration on Jeedom. We can even turn off the light.
https://github.com/Flobul

jayrage887, 2026-08-27:
I think this should be possible though, the wifikit 2.0 is attached to F1 F2 as well.