omerfaruk-aran/esphome_samsung_hvac_bus #153: Different device but with known wire interface

goncalorijo, 2024-08-01:
Hello,

Following a suggestion from another user on the climate_ip integration, i've decided to try this one.
Got a few hours ago a M5Stack and a Tail485.

My ACs have the following WiFi module:
[image]

This is the internal connection:
[image]

This is the wiring diagram:
[image]

I've measured 12V on pin 5 (+) and pin 4 (-):
[image]

So, with this configuration, i think i can replace my wifi module with esphome_samsung_ac module, connection 12V and G.

Now i need to figure our if H_UART_SIN should be connected to A or B on the RS485 module.
Looking into the diagram on the Tail485 component i see that TXD has the same color as the A. This means that I should connect the A to pin 2?

I'm not sure if this will be compatible and if possible would like some guidance if i'm thinking correctly.

Thank you!

goncalorijo, 2024-08-01:
I've notices that i was trying to connect the M5Stack to the wrong connector!
I've disassembled some parts and found out that I also have the tradicional plug:
[image]

This is my AC interior unit:
[image]

jszkiela72, 2024-08-01:
Pink +12V, white 0V. F1, F2 are under the metal cap. Let me know if you managed to establish communication with the unit. I can't handle it. I always have:

[22:45:39][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34
[22:45:40][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34
[22:45:41][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34

goncalorijo, 2024-08-01:
Thank you for your message.

Do you know if there is a connection diagram for this esphome_samsung_ac?
I can't find it anywhere... just some pictures of users...

jszkiela72, 2024-08-01:
do you have M5STACK ATOM Lite + M5STACK ATOM RS-485 Base?

goncalorijo, 2024-08-01:
Kind of:
[image]

Was not able to find the base, but got this that is supposed to be similar in every way.

jszkiela72, 2024-08-01:
Your hardware is ok. Connect it:
[image]

goncalorijo, 2024-08-01:
Yay! Thank you!

But can you share a picture of the location of V1, V2, F1 and F2 on the AC?
That is what i'm looking for.

jszkiela72, 2024-08-01:
[image]

jszkiela72, 2024-08-01:
I did it!!! I have one outdoor unit and 5 indoor ACs from 2015 (non-NASA). If you have any problems, write to me.
[image]

goncalorijo, 2024-08-03:
It is Saturday and kids are at grandparents! I have one full day to just make it work :P

Everything is wired up and connected:
[image]

But i'm facing the constant "update" message without any information:
[image]

I think i've messed up the esp home configuration file. I had to comment a bunch of things for it to work and i'm not sure if i correctly configured everything... Here are the problems i've faced:

1. I have no API value.
2. I had nothing unde OTA and had to ad "  platform: esphome" to it
3. Since i'm using the tail rs485 version, i've changed the pings aligned with the comments to GPIO026 and GPIO032
4. Commented everything bellow " # Skip everything below on the first run!"

Is it possible for you to share with me the initial configuration yaml file to detect devices?

Thank you

jszkiela72, 2024-08-03:
Try my code. Change only GPIO.
```
# This example provides a basic configuration and description based on the M5STACK components
# For further configuration you should visit the ESPHome docs https://esphome.io
# Every line starting with a # can be removed.

## BASIC CONFIGURATION
esphome:
  name: samsung_ac
  friendly_name: Climate

# Select chipset and board configuration - in this case M5STACK
esp32:
  board: m5stack-atom

# [Optional] - Create web interface where you can control the AC from your web browser (without Home Assistant)
web_server:
  port: 80

# Enable logging
logger:
#  level: INFO
  logs:
    component: ERROR # Remove the "Your component takes to long to respond warning"

# Enable native API for communication with Home Assistant, this can be removed if you do not plan to use it with Home Assistant
api:

# Enable OTA firmware update, you will be able to upload the firmware over WiFi without connecting the device by cables
ota:
  platform: esphome

# Setup your WiFi password
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

# Specify pins used by the board to comunicate with RS485 board - in this case M5STACK
uart:
  tx_pin: GPIO19   # Use GPIO26 for ATOM Tail485 
  rx_pin: GPIO22   # Use GPIO32 for ATOM Tail485
  # For older devices it could also be 2400. But you should start with 9600.
  baud_rate: 9600
  parity: EVEN

#  debug:
#    direction: BOTH
#    dummy_receiver: false
#    after:
#      delimiter: "\n"
#    sequence:
#        - lambda: UARTDebug::log_hex(direction, bytes, ' ');  

## SAMSUNG AC CONFIGURATION

# Import custom component from GitHub
external_components:
  - source: github://lanwin/esphome_samsung_ac@main # use @main if you want the latest development (possibly unstable?)
    components: [samsung_ac]

# Configuration of AC component
samsung_ac:

  # For NonNASA devices the following option can be enabled to prevent the device from sleeping when idle. This allows
  # values like internal and external temperature to continue to be tracked when the device isn't in use.
  non_nasa_keepalive: true

  # Capabilities configure the features alle devices of your AC system have (all parts of this section are optional). 
  # All capabilities are off by default, you need to enable only those your devices have.
  # You can override or configure them also on a per devices basis (look below for that).
  capabilities: 
    vertical_swing: true
    horizontal_swing: true
    # Presets define special AC modes like Windfree and so on. 
    # The following modes are available: sleep, quiet, fast, longreach, windfree
    presets: 
      # short version
      #quiet: true
      # long version - allows to locilize names
      quiet:
        name: "Makes no sound"
        enabled: true

  devices:

    # Skip everything below on the first run! Wait a minute, watch your ESPHome logs until you see the "Discovered devices:" section and you see some addresses:
    # For NASA devices it looks like this (with 4 indoor devices): 
    #   Indoor:  20.00.00, 20.00.01, 20.00.02, 20.00.03
    # For NonNASA devices it looks like this (with 2 indoor devices): 
    #   Indoor:  00, 01
    # You need this addresses for the the sections below. If you see only a "-" then you should visit the troubleshooting section in the readme.     

    - address: "00" # Indoor device address 
      # Each property below is optional (climate, room_temperature etc.) - you can delete those which you dont need.
      # For the names we suggest to choose a combination of room name and the thing it controls. 

      # Configures/overrides the capabilites for this devices. 
      # Look above for all options.
      capabilities:
        horizontal_swing: false # This device have no h swing. 
        
      # Creates climate control in Home Assistant. A climate control combines multiple of the controls below (like temperature, mode etc.)
      climate:
        name: "Sypialnia climate"

      # The controls directly below are all included in the climate control. Its adviced to only add the climate control and skip the extra controls.
      room_temperature:
        name: "Sypialnia temperature"
      target_temperature:
        name: "Sypialnia target temperature"
      power:
        name: "Sypialnia power"
      mode:
        name: "Sypialnia mode"

      # If your AC sits near or inside the ceiling, the reported room temperature is often a little bit heigher then whats 
      # measured below. This property can be used to correct that value.
      room_temperature_offset: -1.0

    - address: "01" # Indoor device address 
```

goncalorijo, 2024-08-03:
Thank you!

I think it helped me with my problem.
But I've also changed to VERBOSE the message level.
[image]

Will keep progressing and keep this thread updated.

goncalorijo, 2024-08-03:
And everything is working!

Thank you very much!

closed 2024-08-03 by goncalorijo

jszkiela72, 2024-08-03:
314 / 5 000
Glad I could help you. Please check what works for you.
Mine works fine:
Mode selection.
Off.
Temperature change.
Not working fine:
Fan speed.
Swinging.

I want to know if this is just my problem or all Non-NASA ACs
I'm going to ask @Lanwin to fix these problems.
I am still using the https://github.com/atxbyea/samsungrac integration which may help in getting the data and solving problems.

goncalorijo, 2024-08-03:
I've just configured the desired temperature, actual temperature, fan speed and mode (warm or cold).
This is enough for me at the moment.

In the future, if needed, i will try to add those things.

Thank you again!