omerfaruk-aran/esphome_samsung_hvac_bus #154: Warnings

goncalorijo, 2024-08-03:
I'm having a lot of warnings on my logs:
[image]

Any idea of what could it be?

goncalorijo, 2024-08-06:
Adding some verbose log:
```
[19:53:23][D][samsung_ac:502]: Sending controller registration request...
[19:53:23][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34
[19:53:23][D][samsung_ac:502]: Sending controller registration request...
[19:53:24][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34
[19:53:24][D][number:012]: 'Temperatura Desejada': Sending state 23.000000
[19:53:24][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:24][D][climate:399]:   Mode: OFF
[19:53:24][D][climate:404]:   Fan Mode: AUTO
[19:53:24][D][climate:419]:   Current Temperature: 24.00°C
[19:53:24][D][climate:425]:   Target Temperature: 23.00°C
[19:53:24][V][sensor:043]: 'Temperatura Actual': Received new state 24.000000
[19:53:24][D][sensor:094]: 'Temperatura Actual': Sending state 24.00000 °C with 1 decimals of accuracy
[19:53:24][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:24][D][climate:399]:   Mode: OFF
[19:53:24][D][climate:404]:   Fan Mode: AUTO
[19:53:24][D][climate:419]:   Current Temperature: 24.00°C
[19:53:24][D][climate:425]:   Target Temperature: 23.00°C
[19:53:24][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:24][D][climate:399]:   Mode: OFF
[19:53:24][D][climate:404]:   Fan Mode: AUTO
[19:53:24][D][climate:419]:   Current Temperature: 24.00°C
[19:53:24][D][climate:425]:   Target Temperature: 23.00°C
[19:53:24][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:24][D][climate:399]:   Mode: OFF
[19:53:24][D][climate:404]:   Fan Mode: AUTO
[19:53:24][D][climate:419]:   Current Temperature: 24.00°C
[19:53:24][D][climate:425]:   Target Temperature: 23.00°C
[19:53:24][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:24][D][climate:399]:   Mode: OFF
[19:53:24][D][climate:404]:   Fan Mode: AUTO
[19:53:24][D][climate:419]:   Current Temperature: 24.00°C
[19:53:24][D][climate:425]:   Target Temperature: 23.00°C
[19:53:24][W][samsung_ac:244]: Unsupported alt_mode 0
[19:53:24][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:24][D][climate:399]:   Mode: OFF
[19:53:24][D][climate:404]:   Fan Mode: AUTO
[19:53:24][D][climate:419]:   Current Temperature: 24.00°C
[19:53:24][D][climate:425]:   Target Temperature: 23.00°C
[19:53:24][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:24][D][climate:399]:   Mode: OFF
[19:53:24][D][climate:404]:   Fan Mode: AUTO
[19:53:24][D][climate:419]:   Current Temperature: 24.00°C
[19:53:24][D][climate:425]:   Target Temperature: 23.00°C
[19:53:24][D][samsung_ac:502]: Sending controller registration request...
[19:53:24][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34
[19:53:25][D][samsung_ac:502]: Sending controller registration request...
[19:53:25][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34
[19:53:25][V][samsung_ac:087]: invalid end byte: 32c800f8be4369e9ffffffff32c8
[19:53:25][D][number:012]: 'Temperatura Desejada': Sending state 23.000000
[19:53:25][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:25][D][climate:399]:   Mode: OFF
[19:53:25][D][climate:404]:   Fan Mode: AUTO
[19:53:25][D][climate:419]:   Current Temperature: 24.00°C
[19:53:25][D][climate:425]:   Target Temperature: 23.00°C
[19:53:25][V][sensor:043]: 'Temperatura Actual': Received new state 24.000000
[19:53:25][D][sensor:094]: 'Temperatura Actual': Sending state 24.00000 °C with 1 decimals of accuracy
[19:53:25][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:25][D][climate:399]:   Mode: OFF
[19:53:25][D][climate:404]:   Fan Mode: AUTO
[19:53:25][D][climate:419]:   Current Temperature: 24.00°C
[19:53:25][D][climate:425]:   Target Temperature: 23.00°C
[19:53:25][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:25][D][climate:399]:   Mode: OFF
[19:53:25][D][climate:404]:   Fan Mode: AUTO
[19:53:25][D][climate:419]:   Current Temperature: 24.00°C
[19:53:25][D][climate:425]:   Target Temperature: 23.00°C
[19:53:25][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:25][D][climate:399]:   Mode: OFF
[... 172 lines cut ...]
[19:53:29][D][climate:404]:   Fan Mode: AUTO
[19:53:29][D][climate:419]:   Current Temperature: 24.00°C
[19:53:29][D][climate:425]:   Target Temperature: 23.00°C
[19:53:29][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:29][D][climate:399]:   Mode: OFF
[19:53:29][D][climate:404]:   Fan Mode: AUTO
[19:53:29][D][climate:419]:   Current Temperature: 24.00°C
[19:53:29][D][climate:425]:   Target Temperature: 23.00°C
[19:53:29][W][samsung_ac:244]: Unsupported alt_mode 0
[19:53:29][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:29][D][climate:399]:   Mode: OFF
[19:53:29][D][climate:404]:   Fan Mode: AUTO
[19:53:29][D][climate:419]:   Current Temperature: 24.00°C
[19:53:29][D][climate:425]:   Target Temperature: 23.00°C
[19:53:29][D][climate:396]: 'Piso 0 - Sala' - Sending state:
[19:53:29][D][climate:399]:   Mode: OFF
[19:53:29][D][climate:404]:   Fan Mode: AUTO
[19:53:29][D][climate:419]:   Current Temperature: 24.00°C
[19:53:29][D][climate:425]:   Target Temperature: 23.00°C
[19:53:30][D][samsung_ac:502]: Sending controller registration request...
[19:53:30][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34
[19:53:30][D][samsung_ac:502]: Sending controller registration request...
[19:53:30][W][samsung_ac:080]: write 32d0c8d1d2000000000000001b34
[19:53:30][V][samsung_ac:087]: invalid end byte: 32c800c068e6e1e85fffffffff32
```

goncalorijo, 2024-08-06:
This is my configuration file:
`
esphome:
  name: ac-m5stack-simuladores
  friendly_name: Controlador AC M5Stack Simulador

esp32:
  board: m5stack-atom

web_server:
  port: 80

logger:
  level: INFO
  logs:
    component: ERROR # Remove the "Your component takes to long to respond warning"

api:

ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  tx_pin: GPIO26   # Use GPIO26 for ATOM Tail485 
  rx_pin: GPIO32   # Use GPIO32 for ATOM Tail485
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://lanwin/esphome_samsung_ac@main # use @main if you want the latest development (possibly unstable?)
    components: [samsung_ac]

samsung_ac:

  non_nasa_keepalive: true

  capabilities: 
    vertical_swing: false
  
  devices:
    - address: "00" # Indoor device address 
      climate:
        name: "Piso 1 - Quarto Traseiras"

    - address: "01" # Indoor device address 
      climate:
        name: "Piso 2 - Projector"

    - address: "02" # Indoor device address 
      climate:
        name: "Piso 2 - Simuladores"
`

goncalorijo, 2024-08-09:
Understood and thank you for your reply. Crossing fingers!

One additional question, is there any configuration that i can add to stop receiving this warning? Or is it better to just leave it like this?

Thank you!

lanwin, 2024-08-09:
Yes you can disable the capabilities until the message disappears.

goncalorijo, 2024-08-09:
Ok, thank you.

I've been looking for the list of capabilities, but haven't found it.
Is it available somewhere? Can you please point me to it?

Thank you

lanwin, 2024-08-13:
Just look above the presets section in the example yaml.

NdR91, 2024-08-30:
Hi @lanwin,

First of all thank you for your work. You saved me from having a totally non-smart AC in my house.
My Samsung is NASA (or maybe, i think it's NASA) and i have the very same error:

`[09:44:44][W][samsung_ac:358]: Unsupported alt_mode 0`

Also (I don't know if it's related to the same discussion) i notice that sometimes commands aren't received from the AC.

lucapepe97, 2024-09-01:
Hello @lanwin, 
I want to thank you very much for your work.

The controller works very good even with an old cheap NonNasa single unit (begin of 2019) (Both compressor and internal unit has code: ar09nxfpewqneu.).

The only problem I also recive the warning Unsupported alt_mode 0.

I disabled every capabilities on the config, but the warning is still persistent.
Attached my config, thanks again for any help.

[log config.txt](https://github.com/user-attachments/files/16829426/log.config.txt)

lanwin, 2024-09-02:
You can just ignore that message for now.

closed 2024-10-29 by omerfaruk-aran