omerfaruk-aran/esphome_samsung_hvac_bus #165: [NASA] - compiler error on ESP-IDF

DerTiger1, 2024-08-24:
In file included from src/esphome/components/samsung_ac/protocol_nasa.cpp:4:
src/esphome/components/samsung_ac/protocol_nasa.cpp: In function 'void esphome::samsung_ac::process_messageset_debug(std::__cxx11::string, std::__cxx11::string, esphome::samsung_ac::MessageSet&, esphome::samsung_ac::MessageTarget*)':
src/esphome/components/samsung_ac/protocol_nasa.cpp:27:23: error: format '%f' expects argument of type 'double', but argument 7 has type 'long int' [-Werror=format=]
         ESP_LOGW(TAG, "s:%s d:%s " #message_name " %f", source.c_str(), dest.c_str(), temp); \
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:72:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
 #define ESPHOME_LOG_FORMAT(format) format
                                    ^~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_w'
 #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
                            ^~~~~~~~~~
src/esphome/components/samsung_ac/protocol_nasa.cpp:27:9: note: in expansion of macro 'ESP_LOGW'
         ESP_LOGW(TAG, "s:%s d:%s " #message_name " %f", source.c_str(), dest.c_str(), temp); \
         ^~~~~~~~
src/esphome/components/samsung_ac/protocol_nasa.cpp:861:17: note: in expansion of macro 'LOG_MESSAGE'
                 LOG_MESSAGE(ENUM_IN_OPERATION_VENT_POWER, message.value, source, dest);

omerfaruk-aran, 2024-08-25:
> In file included from src/esphome/components/samsung_ac/protocol_nasa.cpp:4: src/esphome/components/samsung_ac/protocol_nasa.cpp: In function 'void [...]

Hi there!

Could you please let us know which branch you are using to load ESPHome?

The error you are encountering seems to be caused by attempting to log a long int value using the %f format specifier in the ESP_LOGW function. The %f format is intended for double or float types, so trying to print a long int with it leads to this issue. You should use %ld or %li to correctly format a long int.

We may need to review the type of message.value and adjust the format specifier accordingly to resolve the issue.

Let us know if you need further assistance!

hnykda, 2024-08-25:
It happened to me on main. I was able to get it compiled with this: https://github.com/lanwin/esphome_samsung_ac/pull/166

lanwin, 2024-08-26:
Should be fixed now

closed 2024-08-26 by lanwin

DerTiger1, 2024-08-26:
wow - vielen Dank @lanwin !!!
works fine :-)