omerfaruk-aran/esphome_samsung_hvac_bus #176: How to set temperature of all AC's at once?

barisahmet, 2024-09-16:
Hello,

Thanks for the great integration. I have set up the integration and works flawlessly. However, when I update temperature of all AC's at once, it fails to set all (I have 8 ceiling AC's). I have to spam the change few times to set it all. It's working well when I set temperature for just one AC.

Does anyone have same problem?

I am using latest stable Home Assistant, latest stable esphome, M5Stack. My AC units are mixed, I can provide model numbers if you want but I don't think it's related.

lanwin, 2024-09-17:
Hi @barisahmet this is a known bug and its discussed here https://github.com/lanwin/esphome_samsung_ac/issues/57

closed 2024-09-17 by lanwin

htht2001, 2024-10-07:
In an attempt to understand where the problem is, this is the log in which I send a command in S-NET pro 2 to set the temperature of 4 units to 22

```
[23:44:13][W][samsung_ac:828]: Request #Packet Src:80.ff.00 Dst:b2.01.20 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 3;PacketNumber: 252}
 > LongVariable 4405 = 3112
 > LongVariable 4406 = 0
 > LongVariable 4407 = 0
 > Variable 4201 = 220
[23:44:13][W][samsung_ac:822]: Ack #Packet Src:20.01.0a Dst:80.ff.00 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 6;PacketNumber: 252}
 s 17
```

And this is the log that I execute the same command through esphome

```
[23:48:12][W][samsung_ac:468]: publish packet #Packet Src:80.ff.00 Dst:20.01.03 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 3;PacketNumber: 23}
 > Variable 4201 = 220
[23:48:12][W][samsung_ac:086]: write 32001280ff00200103c0131701420100dc88f234
[23:48:12][W][samsung_ac:468]: publish packet #Packet Src:80.ff.00 Dst:20.01.0b {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 3;PacketNumber: 24}
 > Variable 4201 = 220
[23:48:12][W][samsung_ac:086]: write 32001280ff0020010bc0131801420100dc640e34
[23:48:12][W][samsung_ac:468]: publish packet #Packet Src:80.ff.00 Dst:20.01.05 {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 3;PacketNumber: 25}
 > Variable 4201 = 220
[23:48:12][W][samsung_ac:086]: write 32001280ff00200105c0131901420100dc52d934
[23:48:12][W][samsung_ac:468]: publish packet #Packet Src:80.ff.00 Dst:20.01.0a {PacketInformation: 1;ProtocolVersion: 2;RetryCount: 0;PacketType: 1;DataType: 3;PacketNumber: 26}
 > Variable 4201 = 220
[23:48:12][W][samsung_ac:086]: write 32001280ff0020010ac0131a01420100dc046d34
```

It seems that the form of sending to several units at the same time in S-NET is completely different
And of course in S-NET a proper Ack is received in response

@lanwin 
If you need further tests
I am here for you
Thank you very much

barisahmet, 2024-10-07:
For the record, I am currently using @atanasenko's fork and it's working well for now.