omerfaruk-aran/esphome_samsung_hvac_bus #178: [NASA] can read but not control | Wind-Free Elite AR09TXCAAWKNEU/X

bulior, 2024-09-18:
Hey,

I have an issue that i can read but can´t controll my AR09TXCAAWKNEU/X.
The ESP is reading the status correctly and will update if i change the mode e.g. 

Do you have an idea how to handle this model with esp ?

[image]

<details>
<summary>ESPHome Config:</summary>

esphome:
  name: samsung_ac
  friendly_name: Climate

esp32:
  board: m5stack-atom

web_server:
  port: 80

logger:
  level: INFO
  logs:
    component: ERROR # Remove the "Your component takes to long to respond warning"

Assistant
api:
  

ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  tx_pin: GPIO19   # Use GPIO26 for ATOM Tail485 
  rx_pin: GPIO22   # Use GPIO32 for ATOM Tail485
  baud_rate: 9600
  parity: EVEN

external_components:
  - source: github://lanwin/esphome_samsung_ac@main # use @main if you want the latest development (possibly unstable?)
    components: [samsung_ac]

samsung_ac:
  capabilities:
    vertical_swing: true
    horizontal_swing: true
    presets:
      quiet: true
      fast: true
      windfree: true
      sleep: true
  

  debug_log_messages: true

  debug_log_messages_raw: true

  non_nasa_keepalive: false

  devices:
    - address: "20.00.00"
      climate:
        name: "Haven room AC"
      room_temperature:
        name: "Haven room AC temperature"
      target_temperature:
        name: "Haven room AC target temperature"
      power:
        name: "Haven room AC power"
      mode:
        name: "Haven room AC mode"
      room_humidity:
        name: "Haven room AC humidity"
      room_temperature_offset: -1

</details>

lanwin, 2024-09-19:
Hi, that is a known problem and its tracked here https://github.com/lanwin/esphome_samsung_ac/issues/127

closed 2024-09-19 by lanwin

lanwin, 2024-09-19:
What you can do to help is to try https://github.com/lanwin/esphome_samsung_ac/discussions/66 and check if you can control your AC via Samsung SNET software.

lanwin, 2024-09-19:
But please report your findings to the other issue.