omerfaruk-aran/esphome_samsung_hvac_bus #188: [NASA] Short form of presets not working

laszloh, 2024-10-11:
Since the PR https://github.com/lanwin/esphome_samsung_ac/pull/169 the following yaml generates an error in the code generation phase:

```yaml
  capabilities: 
    vertical_swing: true
    horizontal_swing: true
    presets: 
      quiet: true
      eco: true
      fast: true
```

output:

```
INFO ESPHome 2024.9.2
INFO Reading configuration /config/esphome/esphome-web-d95bf4.yaml...
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Vienna'
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1018, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1005, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 485, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 194, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 206, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 684, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 186, in wrapped
    await coro(conf)
  File "/data/external_components/11b80f5a/components/samsung_ac/__init__.py", line 304, in to_code
    preset_conf.get(CONF_PRESET_NAME, preset_info["displayName"]),
    ^^^^^^^^^^^^^^^
AttributeError: 'bool' object has no attribute 'get'
```

Code generation works with the long form tho.

Pixelschleuder, 2024-10-16:
How can I use the long form instead, e.g. for "eco: true"?

omerfaruk-aran, 2024-10-19:
Thank you for reporting this issue. The problem arises because the `quiet`, `eco`, and `fast` preset values in your YAML configuration are defined as `bool`, which causes an error when trying to use the `.get()` method. The solution is to adjust the YAML file so that presets are defined as objects rather than simple boolean values.

Please try the following updated YAML configuration:

```
capabilities:
  vertical_swing: true
  horizontal_swing: true
  presets:
    - name: quiet
      value: true
    - name: eco
      value: true
    - name: fast
      value: true
```
This structure ensures that each preset has both a `name` and a `value` field, which resolves the issue during code generation. Let us know if this works for you, and feel free to reach out with any further issues!

Pixelschleuder, 2024-10-21:
@omerfaruk-aran this config throws the error "expected a dictionary.". 

In the example of blob/main/example.yaml the long and the short version is shown.
The short version does not work anymore, while the long version still works:

presets: 
      # short version
      quiet: true
      # long version - allows to locilize names
      quiet:
        name: "Makes no sound"
        enabled: true

omerfaruk-aran, 2024-10-21:
Hello,

Regarding your issue, it is similar to what was reported in #192. I have already created a pull request (#193) that addresses this specific problem. Once it's merged, it should resolve the issue you're experiencing.

In the meantime, please review issue #188 and PR #193 for more details and updates.

Thank you for your patience!

closed 2024-10-21 by lanwin